morrownr/88x2bu-20210702 #6: Build fails

YacineSahli, 2021-11-07:
Build fails on Fedora 35 kernel 5.14.16.
Any idea ?

Full error log:
```
DKMS make.log for rtl88x2bu-5.13.1 for kernel 5.14.16-301.fc35.x86_64 (x86_64)
Sun Nov  7 05:31:22 PM CET 2021
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.16-301.fc35.x86_64/build M=/var/lib/dkms/rtl88x2bu/5.13.1/build  modules
make[1]: Entering directory '/usr/src/kernels/5.14.16-301.fc35.x86_64'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_chplan.o
/var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_mlme_ext.c:299:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  299 |                 if (rfctl->regd_name)
      |                    ^
/var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_mlme_ext.c:302:9: note: here
  302 |         default:
      |         ^~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_mlme_ext.c:1670:38: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1670 |                         ptable->func = &OnAuthClient;
      |                         ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_mlme_ext.c:1672:9: note: here
 1672 |         case WIFI_ASSOCREQ:
      |         ^~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/monitor/rtw_radiotap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/wds/rtw_wds.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_ft.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_wnm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_mbo.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_roch.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/crypto/aes-internal.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/crypto/aes-internal-enc.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/crypto/aes-gcm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/crypto/aes-ccm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/crypto/aes-omac1.o
/var/lib/dkms/rtl88x2bu/5.13.1/build/core/efuse/rtw_efuse.c: In function ‘rtw_efuse_analyze’:
/var/lib/dkms/rtl88x2bu/5.13.1/build/core/efuse/rtw_efuse.c:971:17: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  971 |                 if (i % 16 == 0)
      |                 ^~
In file included from /var/lib/dkms/rtl88x2bu/5.13.1/build/include/drv_types.h:60,
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/core/efuse/rtw_efuse.c:17:
/var/lib/dkms/rtl88x2bu/5.13.1/build/include/rtw_debug.h:261:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  261 |         do {\
      |         ^~
/var/lib/dkms/rtl88x2bu/5.13.1/build/core/efuse/rtw_efuse.c:973:25: note: in expansion of macro ‘_RTW_PRINT_SEL’
  973 |                         _RTW_PRINT_SEL(RTW_DBGDUMP, "%02X%s"
      |                         ^~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/crypto/ccmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/crypto/gcmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/crypto/aes-siv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/crypto/aes-ctr.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/crypto/sha256-internal.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/crypto/sha256.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/crypto/sha256-prf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/crypto/rtw_crypto_wrap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_swcrypto.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/nlrtw.o
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/recv_linux.c: In function ‘napi_recv’:
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/recv_linux.c:408:76: error: ‘GRO_DROP’ undeclared (first use in this function)
  408 |                         if (rtw_napi_gro_receive(&padapter->napi, pskb) != GRO_DROP)
      |                                                                            ^~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/recv_linux.c:408:76: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/recv_linux.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_get_chbwoff_from_cfg80211_chan_def’:
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:6796:21: warning: this statement may fall through [-Wimplicit-fallthrough=]
 6796 |                 *ht = 0;
      |                 ~~~~^~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:6798:9: note: here
 6798 |         case NL80211_CHAN_WIDTH_20:
      |         ^~~~
make[1]: *** [Makefile:1858: /var/lib/dkms/rtl88x2bu/5.13.1/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.16-301.fc35.x86_64'
make: *** [Makefile:2498: modules] Error 2
```

morrownr, 2021-11-07:
After seening your message I did some checking. Somehow I forgot to apply the patches for kernels 5.12 and 5.15 during the process of testing and getting this driver ready. I have applied said patches now.

Recommend that you run ./remove-driver.sh and the delete the current driver directory. Perform a reboot and run the instalation again. Let me know what happens.

Regards

YacineSahli, 2021-11-08:
It's building and running well now, thanks a lot !

closed 2021-11-10 by morrownr