morrownr/88x2bu-20210702 #7: NetGear A6150 does not work on Ubuntu kernel 5.11

loocy3, 2021-11-21:
The driver does not work on ubuntu kernel: 5.11.0-38-generic(Realtek) with NetGear A6150. 
I have successful used 88x2bu driver on ubuntu kernel: 5.8(Realtek) but it does not work after I upgrade ubuntu to 5.11(Realtek).
So I reinstalled 88x2bu-20210702, it still does not work.
No wireless interface show in 'lshw -C network'.
Could you have a check? Thank you.

morrownr, 2021-11-21:
Hi @loocy3 

I need you to post the results of the following command:

$ sudo uname -a && mokutil --sb-state && lsusb && rfkill list all && dkms status && iw dev

I did just a little research on that adapter. Is it a nano adapter? It appears to be based on a 8812bu chipset.

When you say that you "upgraded", can you be specific? Did you install Ubuntu 21.04 over 20.04 or something similar to that?/

The reason I am asking is that these Realtek drivers really need to be uninstall prior to distro version upgrades. You can read about it in README in the `Removal of the driver` section.

Regards

loocy3, 2021-11-22:
Thank you!
Here are results of the command:
--------------------------------------------------------
Linux shan-desktop 5.11.0-38-generic #42~20.04.1-Ubuntu SMP Tue Sep 28 20:41:07 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
SecureBoot disabled
Platform is in Setup Mode
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0424:2412 Microchip Technology, Inc. (formerly SMSC) 
Bus 001 Device 004: ID 26ce:01a2 ASRock LED Controller
Bus 001 Device 003: ID 05ac:12a8 Apple, Inc. iPhone5/5C/5S/6
Bus 001 Device 002: ID 0846:9055 NetGear, Inc. A6150
Bus 001 Device 007: ID 413c:301a Dell Computer Corp. Dell MS116 USB Optical Mouse
Bus 001 Device 006: ID 413c:2003 Dell Computer Corp. Keyboard
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl88x2bu, 5.13.1, 5.11.0-38-generic, x86_64: installed
virtualbox, 6.1.26, 5.11.0-38-generic, x86_64: installed
virtualbox, 6.1.26, 5.8.0-34-generic, x86_64: installed
nl80211 not found.
----------------------------------------------------------
Yes, I use a nano adapter and I think it is based on a 8812bu chipset.
I installed Ubuntu 20.04
I have move every thing form /lib/modules/5.11.0-38-generic/kernel/drivers/net/wireless/, it is still does not work.

morrownr, 2021-11-22:
Hi @loocy3 (Shan)

> virtualbox, 6.1.26, 5.11.0-38-generic, x86_64: installed

It is highly likely that the above is the problem.

See if the follow article can help you:

https://null-byte.wonderhowto.com/forum/wifi-hacking-attach-usb-wireless-adapter-with-virtual-box-0324433/

If it does help, let me know and I will see about putting a short guide up or a link to this article in case others have problems.

>  I installed Ubuntu 20.04 I have move every thing form /lib/modules/5.11.0-38-generic/kernel/drivers/net/wireless/, it is still does not work.

Be careful about moving files around when the problem has not been determined. You should run the `remove-driver.sh` script to clean things up and then run the installation again.

Regards

loocy3, 2021-11-24:
Hi Morrownr,

Thank you very much.
I am using linux Ubuntu without virtual machine, I have removed virtual-box and still could not  find network.
Maybe there are some thing wrong with the hardware of wifi adapter.
I will have a check.

Thank you 
Kind regards

morrownr, 2021-11-25:
Hi Shan

> I have removed virtual-box and still could not find network. Maybe there are some thing wrong with the hardware of wifi adapter. I will have a check.

The adapter is being recognized because lsusb is telling us what it is. That doesn't mean all is well with the hardware of usb port but we can continue.

Next time you stop by, bring an easy problem with you... heh heh.

How about you run the following and post the results:

$ sudo rfkill unblock wlan
$ lsmod

So far what we know is that your driver is installed and we have the right driver, The driver seems to be waiting to start the adapter but something is blocking it.

Do you have airplane mode on? If you system is a laptop, there can be a lot of different ways that airplane mode can be turned on and off. Some laptops have a physical button and some use a key combination. If you tell what computer you have, maybe we can figure it out.

Nick

loocy3, 2021-11-26:
Thank you very much!

Here are the results of the commands:
---------------------------------------------------------------
shan@shan-desktop:~/projects/depth/RNerf$ sudo rfkill unblock wlan
shan@shan-desktop:~/projects/depth/RNerf$ lsmod
Module                  Size  Used by
ib_core               360448  0
uas                    28672  0
usb_storage            73728  1 uas
iptable_filter         16384  0
bpfilter               16384  0
binfmt_misc            24576  1
nls_iso8859_1          16384  1
nvidia_uvm           1028096  2
nvidia_drm             61440  13
nvidia_modeset       1150976  24 nvidia_drm
intel_rapl_msr         20480  0
intel_rapl_common      24576  1 intel_rapl_msr
snd_sof_pci            24576  0
snd_sof_intel_hda_common    94208  1 snd_sof_pci
snd_hda_codec_realtek   139264  1
snd_hda_codec_generic    81920  1 snd_hda_codec_realtek
snd_sof_intel_hda      20480  1 snd_sof_intel_hda_common
snd_sof_intel_byt      28672  1 snd_sof_pci
snd_sof_intel_ipc      20480  1 snd_sof_intel_byt
snd_sof               126976  4 snd_sof_pci,snd_sof_intel_hda_common,snd_sof_intel_byt,snd_sof_intel_ipc
snd_sof_xtensa_dsp     16384  2 snd_sof_intel_hda_common,snd_sof_intel_byt
snd_soc_hdac_hda       24576  1 snd_sof_intel_hda_common
snd_hda_codec_hdmi     61440  1
snd_hda_ext_core       32768  3 snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_sof_intel_hda
snd_soc_acpi_intel_match    49152  2 snd_sof_pci,snd_sof_intel_hda_common
snd_soc_acpi           16384  3 snd_soc_acpi_intel_match,snd_sof_intel_hda_common,snd_sof_intel_byt
ledtrig_audio          16384  2 snd_hda_codec_generic,snd_sof
x86_pkg_temp_thermal    20480  0
intel_powerclamp       20480  0
snd_hda_intel          53248  4
snd_intel_dspcfg       28672  3 snd_hda_intel,snd_sof_pci,snd_sof_intel_hda_common
soundwire_intel        40960  2 snd_sof_intel_hda_common,snd_intel_dspcfg
soundwire_generic_allocation    16384  1 soundwire_intel
soundwire_cadence      32768  1 soundwire_intel
coretemp               20480  0
nvidia              36904960  1266 nvidia_uvm,nvidia_modeset
snd_hda_codec         147456  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek,snd_soc_hdac_hda
snd_hda_core           94208  9 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_hda_codec_realtek,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_sof_intel_hda
kvm_intel             294912  0
snd_hwdep              16384  1 snd_hda_codec
soundwire_bus          77824  3 soundwire_intel,soundwire_generic_allocation,soundwire_cadence
snd_soc_core          286720  4 soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_soc_hdac_hda
kvm                   823296  1 kvm_intel
snd_compress           28672  1 snd_soc_core
ac97_bus               16384  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm               114688  11 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_sof_intel_ipc,snd_compress,snd_soc_core,snd_hda_core,snd_pcm_dmaengine
crct10dif_pclmul       16384  1
ghash_clmulni_intel    16384  0
aesni_intel           372736  0
snd_seq_midi           20480  0
snd_seq_midi_event     16384  1 snd_seq_midi
crypto_simd            16384  1 aesni_intel
cryptd                 24576  2 crypto_simd,ghash_clmulni_intel
snd_rawmidi            36864  1 snd_seq_midi
glue_helper            16384  1 aesni_intel
drm_kms_helper        237568  1 nvidia_drm
snd_seq                73728  2 snd_seq_midi,snd_seq_midi_event
mei_hdcp               24576  0
cec                    53248  1 drm_kms_helper
rapl                   20480  0
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
rc_core                61440  1 cec
joydev                 24576  0
input_leds             16384  0
intel_cstate           20480  0
ucsi_ccg               20480  0
snd_timer              40960  2 snd_seq,snd_pcm
typec_ucsi             40960  1 ucsi_ccg
fb_sys_fops            16384  1 drm_kms_helper
mei_me                 40960  1
syscopyarea            16384  1 drm_kms_helper
snd                    94208  21 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_rawmidi
sysfillrect            16384  1 drm_kms_helper
typec                  57344  1 typec_ucsi
wmi_bmof               16384  0
intel_wmi_thunderbolt    20480  0
sysimgblt              16384  1 drm_kms_helper
efi_pstore             16384  0
soundcore              16384  1 snd
mei                   122880  3 mei_hdcp,mei_me
ee1004                 20480  0
intel_pch_thermal      20480  0
acpi_tad               16384  0
mac_hid                16384  0
acpi_pad              184320  0
sch_fq_codel           20480  2
msr                    16384  0
parport_pc             45056  0
ppdev                  24576  0
lp                     20480  0
parport                65536  3 parport_pc,lp,ppdev
drm                   548864  17 drm_kms_helper,nvidia,nvidia_drm
ip_tables              32768  1 iptable_filter
x_tables               49152  2 iptable_filter,ip_tables
autofs4                45056  2
hid_generic            16384  0
usbhid                 57344  0
hid                   135168  2 usbhid,hid_generic
crc32_pclmul           16384  0
nvme                   45056  2
e1000e                258048  0
nvme_core             122880  3 nvme
i2c_i801               32768  0
ahci                   40960  1
xhci_pci               20480  0
i2c_smbus              20480  1 i2c_i801
libahci                36864  1 ahci
xhci_pci_renesas       20480  1 xhci_pci
i2c_nvidia_gpu         16384  0
wmi                    32768  2 intel_wmi_thunderbolt,wmi_bmof
video                  53248  0
----------------------------------------------------------------

Best,
Shan

morrownr, 2021-11-26:
Let's see what this says:

$ dkms status

FYI: It is best with these Realtek drivers to uninstall them before upgrades that include a kernel version upgrade such as v5.8 to 5.11. I know that is a pain but then there is a reason I have this site available:

https://github.com/morrownr/USB-WiFi

Users have a lot fewer problems with adapters that use in-kernel drivers.

loocy3, 2021-11-29:
Nick,

Thank you very much for your help.
I have successful re-installed this driver after this command. 
$ sudo apt full-upgrade

Best,
Shan

morrownr, 2021-11-29:
Shan,

That is good news. Enjoy.

Nick

closed 2021-11-29 by morrownr