morrownr/88x2bu-20210702 #9: dkms build error = 10

snortingnails, 2021-11-22:
[make.log](https://github.com/morrownr/88x2bu-20210702/files/7577434/make.log)
im on the latest kernel and i got this error
no idea what it means but it doesn't work so far

morrownr, 2021-11-22:
We need people like you living on the bleeding edge to find all the problems so they are fixed by the time we get there.

I see you are on kernel 5.15. My main dev box is on kernel 5.13 but I do have a test box with a 5.15 kernel and I've seen some reports from users that 5.15 is working well.

In looking at the log, I'm wondering what version of gcc you are running:

$ gcc -version

Also, tell me what distro you are using.

morrownr, 2021-11-29:
Closed due to lack of reply.

closed 2021-11-29 by morrownr