morrownr/88x2bu-20210702 #10: cannot compile on rPi0w

alexl83, 2021-12-07:
Hi, I cannot get this driver to compile on rPi Zero W (it compiles correctly on rPi Zero 2 W by the way)
I'm using re4son kernel from Kali - version 5.4.83-20210516 on raspi os buster (ditribution [github.com/evilsocket/pwnagotchi](url) 1.5.5)

kernel headers for re4son kernel are installed: can you help me please? I'm lost
Thank you!

ale

```
pi@foresight:~/88x2bu-20210702 $ ./raspiOS-32.sh 
I386 support was turned off in Makefile as planned.
Raspberry Pi OS (32 bit) support was turned on in Makefile as planned.
```
`sudo ./install-driver.sh `
```
Running install-driver.sh version 20211204
Starting installation...
Copying source files to: /usr/src/rtl88x2bu-5.13.1
Copying 88x2bu.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source ->
                 /usr/src/rtl88x2bu-5.13.1

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.4.83-Re4son+ cannot be found at
/lib/modules/5.4.83-Re4son+/build or /lib/modules/5.4.83-Re4son+/source.
An error occurred. dkms build error = 1
Please report this error.
```

morrownr, 2021-12-08:
Hi @alexl83 

You are working with things that I am not familiar with so this will be nothing more than an educated guess.

It appears you kernel headers and not installed and given that they are vitial to this operation, we see immediate failure.

The first thing you want to do is run the following to clean things up:

./remove-driver.sh

The next thing I would suggest is that you run the following for Step 3:

Option for Debian, Kali and Linux Mint Debian Edition (LMDE)

sudo apt install -y linux-headers-$(uname -r) build-essential dkms git libelf-dev

My bet is that you ran the one for the Raspberry Pi OS and that may not be right in this situation.

Continue with the rest of the Installation Steps.

alexl83, 2021-12-11:
Dear Nick, thank you for your support! Indeed kernel headers were missing: I solved installing "armel" version of the re4son kernel/headers package. I was misled by "armhf" package installed (but armhf covers only armv7+ onwards, not armv6l as required by rPi Zero W)

Thank you!
Ale

closed 2021-12-11 by alexl83