morrownr/88x2bu-20210702 #11: An error occurred. dkms build error = 10

lwang046, 2021-12-08:
Hi Nick,

I encounter this issue when trying to install the driver at step `sudo ./install-driver.sh`, any help please?

Here is the output error log:
```
Running install-driver.sh version 20211204
Starting installation...
Copying source files to: /usr/src/rtl88x2bu-5.13.1
Copying 88x2bu.conf to: /etc/modprobe.d
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...(bad exit status: 2)
'make' -j2 KVER=5.14.0-kali4-amd64 KSRC=/lib/modules/5.14.0-kali4-amd64/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.14.0-kali4-amd64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
```
Here is my system information:
````
┌──(guest㉿KOS)-[~/Downloads/88x2bu-20210702]
└─$ sudo uname -a && lsusb && rfkill list all && dkms status && iw dev                        10 ⨯
Linux KOS 5.14.0-kali4-amd64 #1 SMP Debian 5.14.16-1kali1 (2021-11-05) x86_64 GNU/Linux
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 003 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 003 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0e0f:000b VMware, Inc. VMware Virtual USB Video Device
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0008 VMware, Inc. Virtual Bluetooth Adapter
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
0: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
rtl88x2bu/5.13.1: added
```
and the gcc version I'm using: `gcc version 11.2.0 (Debian 11.2.0-10)`.

make.log shows some folder that are non-exist:
```
┌──(guest㉿KOS)-[~/Downloads/88x2bu-20210702]
└─$ cat /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log                                        130 ⨯
DKMS make.log for rtl88x2bu-5.13.1 for kernel 5.14.0-kali4-amd64 (x86_64)
Wed Dec  8 08:07:34 AM EST 2021
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.0-kali4-amd64/build M=/var/lib/dkms/rtl88x2bu/5.13.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.14.0-kali4-amd64'
Makefile:1: /usr/src/linux-headers-5.14.0-kali4-common/Makefile: No such file or directory
make[1]: *** No rule to make target '/usr/src/linux-headers-5.14.0-kali4-common/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.14.0-kali4-amd64'
make: *** [Makefile:2508: modules] Error 2
```

morrownr, 2021-12-08:
Hi @lwang046 

Let's see what we have.

>Linux KOS 5.14.0-kali4-amd64 #1 SMP Debian 5.14.16-1kali1 (2021-11-05) x86_64 GNU/Linux

I am familiar with Kali and I am familiar with Debian but what is this `Linux KOS`?

The output of `lsusb` did not show anything that looks like a USB WiFi adapter. Can you clarify what adapter you are using?
Opps... just noticed VMWare. No wonder your adapter is not showing. I have a link in the FAQ to an article about Virtualbox that might give you an idea what you are up against.

>and the gcc version I'm using: gcc version 11.2.0 (Debian 11.2.0-10).

I'm using gcc v 9.3 to 10.2 on various systems. Most users won't be at that level of gcc yet so this needs to be noted.

My suggestion for continued trouble shooting is for you to run `./remove-driver.sh` and the then simply run `make` to see what we get. Keep in mind that none of this driver compiling matters if you don't solve that VMWare issue where the adapter is not showing.

Overall thoughts: If I were to attempt to write a scenario to annoy Linux sysadmins, I don't think I could have done a better job than what you have come up with... a Frankenstein OS in a VM.  My recommendation is for you to back up, write down what you are trying to accomplish and the hardware you have available to do it. I can then make some suggestions for a setup that will work and is supportable.

Regards,

Nick

lwang046, 2021-12-08:
FYI issue was solved by manually installing the common dependency. Neat work btw, now I can finally use wifi adapter with kali in VM.

morrownr, 2021-12-09:
That's great. If you want to write it up so I can include it in the FAQ, got for it.

closed 2022-05-23 by morrownr