morrownr/88x2bu-20210702 #12: New compatible device

No0ob1e, 2021-12-11:
Thank you for this drivers

Tested on TP-LINK Archer T4U v3 (Realtek RTL8812BU) works perfectly

OS: Manjaro Kernel: 5.10.83-1
Aircrack 1.6

morrownr, 2021-12-11:
Hi @No0ob1e 

Thanks for the information. There will be others that appreciate the info.

If you have some spare time, maybe you can do some things that would benefit many:

1. Check over the README.md to see if all information, especially the installation instructions, are good for Manjaro and Arch users. I have a Manjaro installation available for testing but given the amount of work I have to do, I really need folks like you to keep an eye on things. Any info you provide will be a big benefit as it will be used on additional repos for other drivers such as 8812au, 8821au (8811au chipset), 8814au, 8821cu (8811cu chipset) and 8852au (8832au chipset)(hopefully coming soon).

2. Provide a short report regarding how well this driver works with Aircrack-ng. Historically stable drivers with injection support have been in short supply for the rtl8812bu chipset but my testing shows that this driver does a good job in monitor mode but I have only limited time to test so I need feedback from others.

3. Test a couple of things related to Aircrack and monitor mode that I have been working on that I will make available all. One item is a script that helps test monitor mode use with this driver. I am working on this script to help speed my ability to test how well monitor mode is working as it takes a lot of time if not automated. It might also be a script that Aircrack users could modify and make use of. There is also a test document that shows how to do some things in monitor mode since the Realtek monitor mode is a little different in a couple of places from how you see things done with adapters that use in-kernel drivers. I can paste both here if you are interested in taking a look.

Regards

aki-k, 2021-12-12:
This kernel module also works with ID 2357:012d TP-Link Archer T3U [Realtek RTL8812BU] on Fedora 34 (kernel 5.15.6)

TheJosh, 2022-01-28:
Also works with:
- 2357:0115
- TP-Link Archer T4U ver.3 (AC1300)
- RTL8822BU
- Debian 11
- Kernel 5.10.0-11
- Arch: amd64

TheJosh, 2022-01-30:
Also same device but Debian 10, kernel 4.19.0-18, amd64

closed 2023-09-21 by LogicBypass