morrownr/88x2bu-20210702 #13: Monitor interface doesn't see any APs

skver0, 2021-12-17:
Tried doing "iw phy 0 interface add mon0 type monitor" then "ifconfig mon0 up". 

pi@shiro:~ $ sudo aireplay-ng --test mon0
20:55:42  Trying broadcast probe requests...
20:55:44  No Answer...
20:55:44  Found 0 APs
pi@shiro:~ $ uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
Linux shiro 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux
EFI variables are not supported on this system
Bus 001 Device 004: ID 2357:0138 TP-Link
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
1: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
rtl88x2bu, 5.13.1: added
phy#0
        Interface mon0
                ifindex 7
                wdev 0x4
                addr 00:00:00:00:00:00
                type monitor
                txpower 42949572.96 dBm
        Interface wlan0
                ifindex 4
                wdev 0x1
                addr 10:27:f5:1b:5b:81
                type managed
                txpower 42949572.96 dBm

morrownr, 2021-12-17:
Hi @skver0 

> Tried doing "iw phy0 interface add mon0 type monitor" then "ifconfig mon0 up".

You can't do it like that with these Realtek drivers. When you run `$ iw list` you will see this statement:
```
interface combinations are not supported
```

What you can do is go into monitor while keeping the interface name:

Check the wifi interface name and mode
```
iw dev
```

Take the interface down
```
sudo ip link set <wlan0> down
```

Set monitor mode
```
sudo iw <wlan0> set monitor control
```

Bring the interface up
```
sudo ip link set <wlan0> up
```

Verify the mode has changed
```
iw dev
```

You can also rename the interface and then take it into monitor mode. I have been working on a couple of things that might be of interest and since you just volunteered to test them, I'll post them.

[test-mon.sh.tar.gz](https://github.com/morrownr/88x2bu-20210702/files/7737903/test-mon.sh.tar.gz)

[Monitor_Mode.md.tar.gz](https://github.com/morrownr/88x2bu-20210702/files/7737905/Monitor_Mode.md.tar.gz)

The test-mon.sh script is used like this:

$ sudo ./test-mon.sh <wlan0>

Let me know what you think.

P.S. what you were doing works on wifi adapters that use in-kernel drivers as those drivers are standards compliant:

https://github.com/morrownr/USB-WiFi

skver0, 2021-12-17:
Oh I'm dumb, Thanks for the fast reply! I'll try it out!

morrownr, 2021-12-18:
I have added a document called `Monitor_Mode.md` in the `/docs` folder. See if it helps.

closed 2021-12-19 by skver0