morrownr/88x2bu-20210702 #14: 2 new devices after install

JanB97, 2021-12-18:
Hi,

i am using a RaspberryPi 4B and a USB-Wifi Adapter with RTL8822bu chipset.
After installing the driver, there are two new devices showing up (wlan1 and wlan2).

How can I remove that wifi adapter?

[image]

JanB97, 2021-12-18:
It has two antennas, maybe that is the problem?

[image]

morrownr, 2021-12-18:
Hi @JanB97 

> i am using a RaspberryPi 4B and a USB-Wifi Adapter with RTL8822bu chipset.

I have a RasPi4B and adapters based on the rtl8812bu chipset myself.

> After installing the driver, there are two new devices showing up (wlan1 and wlan2).

That is normal if you ran the `cmode-on.sh` script during the installation. You turned Concurrent Mode on.

> How can I remove that wifi adapter?

There is only one wifi adapter. However, Concurrent Mode allows one adapter to have two interfaces.

Recommend you read `Concurrent_Mode.md` in the `/docs` folder.

There is no problem with not using the second interface. In fact, you may find you want to experiment with it at some point.

However, the easiest way to uninstall it is to run `./remove-driver.sh` and then run the installation instructions again without running `cmode-on.sh`.

Regards,

Nick

JanB97, 2021-12-19:
Ah okay, thank you I didn't realize that!.
I also got another problem where the USB-Adapter get's disconnected at random intervals from the router.
I don't think that has anything to do with your driver, but do you have an Idea how to search for the Problem?

morrownr, 2021-12-19:
> Ah okay, thank you I didn't realize that!. I also got another problem where the USB-Adapter get's disconnected at random intervals from the router. I don't think that has anything to do with your driver, but do you have an Idea how to search for the Problem?

There are two sections in the README (main page) that could be of help...

- Recommended WiFi Router/ Access Point Settings

- Recommendations regarding USB

If I had to offer a most probable guess at this point, it would be that your dual band wifi router has both networks set to the same name. Many wifi router makers do that these days and it is a really bad idea as it can and does cause exactly what you are describing. To fix it, go into your router and change the name of either the 2.4 GHz network or the 5 GHz network and reboot the router.

JanB97, 2021-12-20:
Thank you for the advice, I changed the names many years ago, as I had problems with it :)
I will take a look at the README, although I already tried changing USB to USB2 in the driver.

I found how to get the LOGS of the driver, maybe I can find something in there or ill try another USB-Port.

morrownr, 2021-12-20:
Good morning @JanB97 

If you open a terminal and go to the driver directory and run:

sudo ./edit-options.sh

...it will open the 88x2bu.conf file so that you can edit the driver options. The documentation for many of the options is also in that file, you just have to scroll down.

The default for logging is for logging to be off so you have to set a level. Quite frankly, the log is very rarely helpful in my experience but if you want to give it a try, go for it. Once it is on and you see the problem, there is another script in the driver directory called save-log.sh that will save a file called rtw.log that will only have the driver log entries so that you are not digging through a lot of junk.

sudo ./save-log.sh

I'm curious about this. If you want to post the `options` line from 88x2bu.conf so that I can see what options are active, that might help.

JanB97, 2021-12-20:
Hello Nick,

sure I am happy for everyone that wants to help.
Here are the options:

options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_beamform_cap=10 rtw_dfs_region_domain=0 rtw_sel_p2p_iface=1 rtw_switch_usb_mode=0

I tried changing the "rtw_switch_usb_mode" to "2" but that didn't change anything unfortunately.

I can tell you that this is not a problem with your driver, I had the same exact problem with the driver from MrEngman (http://downloads.fars-robotics.net/) but thought there was a problem with the kernel as I had to update that to a unstable version to get it to work.
I have a short (30mm) usb3 extension between the pi and the adapter, as it would not fit otherwise, maybe that is a problem or the specific USB-Port, I will have to check that.

morrownr, 2021-12-20:
Hi @JanB97 

I have 2 Pi's myself. Let me ask some more questions so that I can wrap my mind around this issue:

Which model of Pi do you have?

Is the usb3 extension just a cable or is it a powered hub?

Is the usb3 extension plugged into a usb 2 or usb3 port?

What else is plugged into your usb ports?

Confirm that your adapter is based on a 8812bu chipset?

I know that many years ago the fix for all things wifi was to turn power saving off but I have done a lot of testing on this driver and the rtw_power_mgnt=1 setting is very stable and I recommend it.

There are other recommendations I could make but I need some answers from some of the above questions first.

Have you tried the adapter directly in a usb port without the usb3 extension?

Regards

JanB97, 2021-12-20:
Its a Pi4 2GB.
I had it running without the cable for a couple of days, but than it also runs for days with the cable sometimes.
The Problem is, I can't really test long term as the Pi is in my 3d printer and there is not enough space without the cable.
It is a normal USB3 extension, about 20cm long and I have it in the upper USB3 Port on the Pi.
I otherwise have the printer connected to the USB3-Port below, I will try switching them around.
The Adapter has a 8822bu as far as I know and the Pi reports.

morrownr, 2021-12-20:
> Its a Pi4 2GB. I had it running without the cable for a couple of days, but than it also runs for days with the cable sometimes. The Problem is, I can't really test long term as the Pi is in my 3d [...]

Okay. I'm getting a better handle on things. 

USB3 is not mankind's greatest invention and the implementation on the Pi4B sure seems to have caused a lot of problem reports... thou I am sure the foundation engineers with argue that all is well. I suspect the problem is somewhere in the USB subsystem. I have fought with it myself and talked to others who have as well.

One concern is that the USB subsystem on the Pi4B can only supply up to 1200 mA of power. Realtek's wifi adapters can be power hungry. That is why I usually recommend Mediatek based adapters to Pi users as the Mediatek adapters use a lot less power. Here is some reading if you have a slow day:

https://github.com/morrownr/USB-WiFi

It might be good if you look up the power requirements for everything that is plugged in if they are getting power from the Pi.

Back to topic: Switching the printer and adapter cables is worth a try.

I have seen a lot of problems with USB3 cables. There are a lot of problems with powered hubs also but that is another issue but getting a little low cost powered hub might, big might, solve this problem. One of the ones that seems to work well is this one:

https://www.amazon.com/gp/product/B005D69QD8

Not that this will solve the problem but if you want to make sure the adapter is in USB3 mode, you can use:

rtw_switch_usb_mode=1

Side note that is probably not related: I had 4 guys helping me test and work on this driver when we were getting it ready to go public. The only problem that came up was only with Pi4B's and it showed as the adapter would stop pumping data. Nothing. It only showed in AP mode when using 80 MHz channel width and only with Pi4B's. We never could find a problem with the driver and this problem never would show on any other hardware. Somewhere in the docs i think we recommended that folks pull back to USB2 if this is a problem as that should still be good enough for most people. We had to just scratch our heads and blame the Pi USB subsystem.

Maybe this gives you some ideas.

Regards

PieGuy314, 2021-12-21:
How far away is the router to the Pi? Is there clear line of site? 5GHz has about a third of the coverage than 2.4GHz. Is the router using a fixed channel or is it configured to select what it thinks is best?

JanB97, 2021-12-21:
So I didn't have time to pull the Pi out of my printer, it may take some days still - because of christmas :)
I will definitly try to change the Ports and also maybe a new cable or powered hub if I can get one.
The pi only powers a small noctua fan and the Wifi Adapter, don't know the power requirements of the Wifi Adapter at the moment, i will have to check that as well.

The Router is about 8 meters away but through a wall. 
The channel is fixed to 36,40,44,48 because it would otherwise select a band that is not supported by my firetv stick, at least in germany.
The signal quality is around 90 if it is connected, so I don't think that should be a problem?

morrownr, 2021-12-22:
When you come up for air after the holidays and if something else doesn't fix this, remind me and I will show you how to totally shut down power saving. That module parameter you had set to 0 actually can't totally shut it down. There are actually about 7 total parameters involved in power saving and it is only something an EE could love. We could give it a shot.

Have a good holiday... you too PieGuy.

JanB97, 2022-05-06:
Sorry for not beeing active for a while.
I did add power to the adapter (with a y cable to the psu) and it did work for the last almost 2 weeks.
I will close the ticket now, thank you!

closed 2022-05-06 by JanB97