morrownr/88x2bu-20210702 #15: (information) Throughput on jjPlus WMU6202 and Edimax EW-7822ULC wifi cards

cristian-ciobanu, 2021-12-25:
Can you please tell me what are the average or maximum throughput for 2.4 and 5 GHz on these two AC WiFi cards jjPlus WMU6202 and Edimax EW-7822ULC ?

I plan to buy one of these and put it on my mini PC and configure them to work as an access point.

Thank you.

morrownr, 2021-12-26:
Hi @cristian-ciobanu 

I'll be glad to help as much as I can.

I use USB WiFi adapters as access points. I have a lot of experience doing it. It is fun and can be very useful. Before I try to answer your question, let me throw some information out.

One of the platforms I have used to make AP's and WiFi Routers is the RasPi but other little systems can work fine as well. I have tested a lot of usb adapters and drivers. Here is a short list of the adapter chipsets that I have found work best if we are talking about a dual band adapter working as an AP:

mt7612u
mt7610u
rtl8812au
rtl8811au

You'll notice that I did not mention the rtl8812bu or rtl8822bu (the chipsets support by the driver in this repo.) Will this driver and the adapters you are asking about support AP mode? Yes. Many people do it successfully but I am aware that this driver, in some situations, can be problematic in AP mode. Since you are investigating what to buy, you might as well buy what works best. I can provide suggestions for specific adapters if I know more details about what you are trying to do.

In the mean time, here is a link that gives a lot of information and links to adapters based on the chipsets I pointed out above.

https://github.com/morrownr/USB-WiFi

Regards

cristian-ciobanu, 2021-12-26:
Thanks for the provided information.

I have a miniPC which I use it as a router/firewall.

This has an miniPCIE/USB slot on it and I bought this WiFi AC card a while ago but did not tested: http://en.techinfodepot.shoutwiki.com/wiki/Foxconn_WFUR6 which is based on Realtek RTL8812AU chipset.

I will give it a try to see how it works when configured in AP mode.

morrownr, 2021-12-26:
The 8812au is a good chipset. I have a very modern version of the 8812au driver available:

https://github.com/morrownr/8812au-20210629

All of the instructions are for using it with a usb wifi adapter but it may work with a PCIe adapter. If you decide to go that direction, file an issue in the 8812au repo and we will see what we can do.

FYI: To see everything that is available here at my site:

https://github.com/morrownr?tab=repositories

One last thing. I communicate with a lot folks using Linux and using small systems to make access points and wifi routers. I want to make sure that you understand that Realtek's USB wireless team does not follow Linux standards which call for wireless drivers to be based on mac80211 technology and the driver should be maintained in the kernel which makes adapters plug and play as well as far less problematic. The companies that follow Linux wireless standards are Intel, Mediatek and Atheros. Mediatek does produce USB capable chipsets and adapter based on those chipsets are available. I realize it is a long document but it provides a LOT of information for someone doing what you are doing:

https://github.com/morrownr/USB-WiFi

Scroll down when you get to the above site.

Regards

cristian-ciobanu, 2021-12-26:
> All of the instructions are for using it with a usb wifi adapter but it may work with a PCIe adapter. If you decide to go that direction, file an issue in the 8812au repo and we will see what we can do.
The Foxconn_WFUR6 is not a standard miniPCIE card but a card which runs using USB signals in an miniPCIE slot. In fact this is a WWAN slot which is very similar with a standard miniPCIE slot but support only USB signal based cards.
So the instructions should apply for this card also.

morrownr, 2021-12-26:
I learning something today. I am very interested to know if my 8812au driver works with this setup.

If it does work, you may have solved your problem without spending any money.

FYI: That driver supports DFS channels if that is useful to you.