morrownr/88x2bu-20210702 #16: ACS is not working with hostapd on raspberry pi 4 (Event SCAN_STARTED (47) received)

dypsilon, 2021-12-26:
Hi Nick & Team,

first of all, thank you for all the hard work maintaining those drivers and guides! They are very well written!

I had a "TP-Link Archer T4U ver.3" lying around and wanted to try using it on my RPi 4 as an access point. First of all: it worked! (maybe you want to add it to the list of supported devices) It is delivering very decent speeds:

~~~
[ ID]   Interval         Transfer      Bitrate          Retr
[  5]   0.00-90.00  sec  2.89 GBytes   276 Mbits/sec    0             sender
[  5]   0.00-90.01  sec  2.89 GBytes   276 Mbits/sec                  receiver
~~~

The only issue is, that it works only on channel 36 for me. If I set `channel=0` to initiate ACS, unfortunately hostapd exits with the following log:

~~~
wlan1: interface state COUNTRY_UPDATE->ACS
wlan1: ACS-STARTED 
Interface initialization will be completed in a callback (ACS)
ctrl_iface not configured!
nl80211: Drv Event 33 (NL80211_CMD_TRIGGER_SCAN) received for wlan1
wlan1: nl80211: Scan trigger
wlan1: Event SCAN_STARTED (47) received
Unknown event 47
nl80211: Drv Event 34 (NL80211_CMD_NEW_SCAN_RESULTS) received for wlan1
~~~

Please feel free to close this issue if this is rather a hostapd issue. If anything, I hope this ticket provides some testing results for you.

modprobe config:

~~~
options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_beamform_cap=10 rtw_dfs_region_domain=3 rtw_sel_p2p_iface=1 rtw_switch_usb_mode=0
~~~ 

hostapd.conf

~~~
ssid=[REDACTED]
wpa_passphrase=[REDACTED]

country_code=DE

interface=wlan1
driver=nl80211

wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP

macaddr_acl=0

logger_syslog=0
logger_syslog_level=4
logger_stdout=-1
logger_stdout_level=0

hw_mode=a
wmm_enabled=1

# N
ieee80211n=1
require_ht=1
ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]

# AC
ieee80211ac=1
require_vht=1
ieee80211d=1
ieee80211h=1
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
vht_oper_chwidth=1
channel=36
vht_oper_centr_freq_seg0_idx=42
~~~

By the way, thank you for providing the `ht_capab` and `vht_capab` lines. They gave me additional 50 mbits.

Here is the output of `sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev`

~~~
Linux raspberrypi 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux
-bash: mokutil: command not found
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 2357:0115 TP-Link Archer T4U ver.3
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
1: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
rtl88x2bu, 5.13.1, 5.10.63-v7l+, armv7l: installed
phy#0
        Interface wlan1
                ifindex 4
                wdev 0x2
                addr [REDACTED]
                ssid [REDACTED]
                type AP
                txpower 25.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr [REDACTED]
                type managed
                txpower -100.00 dBm
~~~

morrownr, 2021-12-27:
> Hi Nick & Team,

Hi @dypsilon 
 
> first of all, thank you for all the hard work maintaining those drivers and guides! They are very well written!

Thanks. There is always room for improvement so if you see something that could be better, send us the fix.
 
> I had a "TP-Link Archer T4U ver.3" lying around and wanted to try using it on my RPi 4 as an access point. First of all: it worked! (maybe you want to add it to the list of supported devices)

I don't want you to take this personally and I hope your TP-Link adapter does not take it personally because it is not intended that way. I do not list products by TP-Link or D-Link. Some folks will use the lists as a shopping list and that is not the intention. The primary reason for not listing products by TP-Link and D-Link is that their Linux support is terrible and both companies use Linux in their products to make profit but yet they do not return the support to Linux. That is all I will say on the issue and I hope you understand. There are many products by companies that do a better job of supporting Linux:

https://github.com/morrownr/USB-WiFi

> It is delivering very decent speeds:
> 
> ```
> [ ID]   Interval         Transfer      Bitrate          Retr
> [  5]   0.00-90.00  sec  2.89 GBytes   276 Mbits/sec    0             sender
> [  5]   0.00-90.01  sec  2.89 GBytes   276 Mbits/sec                  receiver
> ```

This is good. I see you left the adapter in USB2 mode which is recommended for this driver if using a RasPi4B.
 
> The only issue is, that it works only on channel 36 for me. If I set `channel=0` to initiate ACS, unfortunately hostapd exits with the following log:

This is a long standing issue and it isn't just an issue with this driver. I'm not aware of this working with any usb adapters. I researched the issue a long time ago but don't remember why is does not work. You may want to use other methods to check the best channel and just set it manually. That is what I do.

> ```
> options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_beamform_cap=10 rtw_dfs_region_domain=3 rtw_sel_p2p_iface=1 rtw_switch_usb_mode=0
> ```

I see you are ready to use a DFS channel.
 
> ```
> ssid=[REDACTED]
> wpa_passphrase=[REDACTED]
>
> country_code=DE
>
> interface=wlan1
> driver=nl80211
>
> wpa=2
> wpa_key_mgmt=WPA-PSK
> rsn_pairwise=CCMP
>
> macaddr_acl=0
>
> logger_syslog=0
> logger_syslog_level=4 [...]

Be careful with `require_ht=1` and  ` require_vht=1`. I do not use them and I have seen people report issues that were corrected by removing these lines. Make sure you understand what they do.
 
> By the way, thank you for providing the `ht_capab` and `vht_capab` lines. They gave me additional 50 mbits.

You are welcome. There is not that much documentation on this topic so I thought I would help a little.

> Here is the output of `sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev`

I don't see any problems. Set your channel manually, stay in USB2 mode with that RasPi4B and you should have a solid AP.

Next you'll be loading OpenWRT on that RasPi4B and your won't need a router anymore.

Regards

dypsilon, 2021-12-27:
> I don't want you to take this personally and I hope your TP-Link adapter does not take it personally because it is not intended that way. I do not list products by TP-Link or D-Link...

Very good point. Unfortunately, I bought the adapter long before I discovered your guide. Next one will probably be an Alfa AWUS036ACM. By the way, do you have an information whether it is capable of providing better performance in terms of mbits?

> This is a long standing issue and it isn't just an issue with this driver. I'm not aware of this working with any usb adapters.

Thank you! This is already some very important information. In this case there is no point in looking for a replacement adapter to fix this particular issue.

> I researched the issue a long time ago but don't remember why is does not work.

The only potential reason I was able to find is that ACS is only supported for some Atheros devices. I am not sure though, whether [this information](https://wireless.wiki.kernel.org/en/users/Documentation/acs) is still relevant. I was hoping that it is not and I was just holding it wrong.

> I don't see any problems. Set your channel manually, stay in USB2 mode with that RasPi4B and you should have a solid AP. Next you'll be loading OpenWRT on that RasPi4B and your won't need a router anymore.

Thank you for all your advice. RP4 really seems to be a very capable router. I am already thinking about pairing it with a managed switch to have a complete router replacement.

I will close this issue as won't fix...

Kind Regards

closed 2021-12-27 by dypsilon

morrownr, 2021-12-27:
> Next one will probably be an Alfa AWUS036ACM. By the way, do you know whether it is capable of providing better performance in terms of mbits?

The Alfa ACM is my favorite for 5 GHz AP mode on a RasPi4B. I have used iperf3 many times to test the throughput. On an clear channel, I see 380-400 Mb/s which is about what we should expect from an AC1200 adapter.