morrownr/88x2bu-20210702 #17: TP-Link rtl88x2bu computer crashes

Appleuy, 2021-12-30:
computer crashes when start monitor mode on TP-Link t4u v3

morrownr, 2021-12-31:
There is guidance early in the README about what information to include with reports here.

morrownr, 2021-12-31:
Problem duplicated. Cause found and reverted. Please run...

$ sudo ./remove-driver.sh

Then delete the driver folder and start over.

I appreciate the report. I've been working on monitor mode and failed to catch this problem.

If I could ask something in return, since you use monitor mode, would you mind given me a report about how well monitor mode is working for you.

Regards,

Nick

Appleuy, 2022-01-01:
[image]
How to wlan1 support vif to evil twin attack

SSBhaumik, 2022-01-01:
Same problem I'm having with airgeddon

morrownr, 2022-01-01:
Good day @ajitwarrier @Appleuy 

> How to wlan1 support vif to evil twin attack

Realtek drivers don't support virtual interface. From the README:

-----

My recommendation for those looking to buy an adapater for Monitor mode is to buy adapters based on the following chipsets: mt7612u, mt7610u, rtl8812au and rtl8811au. My specific recommendations for adapters in order of preference are:

ALFA AWUS036ACHM - long range - in-kernel driver

ALFA AWUS036ACM - in-kernel driver

ALFA AWUS036ACH - long range - [driver](https://github.com/morrownr/8812au-20210629)

ALFA AWUS036ACS - [driver](https://github.com/morrownr/8821au-20210708)

To ask questions, go to [USB-WiFi](https://github.com/morrownr/USB-WiFi)
and post in `Discussions` or `Issues`.

-----

Since both of you are looking for virtual interface support, you need to eliminate the 3rd and 4th options above since those are Realtek drivers. Adapters that are supported with in-kernel driver will support virtual interfaces. The  [USB-WiFi](https://github.com/morrownr/USB-WiFi) has links and reviews to many usb wifi adapters that use in-kernel drivers. Also, there are several discussions in `issues` where folks are discussing which adapter to buy and what the results are.

Regards

morrownr, 2022-01-06:
You are welcome...