morrownr/88x2bu-20210702 #18: computer freezes

SSBhaumik, 2021-12-31:
31 dec 2021 tested with 1. dlink dwa 182 rev d1 - on entering monitor mode computer freezes and even cursor doesnt move. if any audio playing in the background goes into a short audio loop. Cannot shut down as process doesnt terminate at all. have to disconnect power instead.
2. TP LINK Archer t3u plus v1.0- terminal freezes on monitor mode, cursor still works but cannot shut down as process doesnt terminate at all. have to disconnect power instead.
kali 2021.4 kernel 5.14 
pavilion p001tx laptop with 8GB ram.
Rincat release seems to work fine. Had to uninstall yours, if you update this repo please let me know.

morrownr, 2021-12-31:
Problem duplicated. Cause found and reverted. Please run...

$ sudo ./remove-driver.sh

Then delete the driver folder and start over.

I appreciate the report. I've been working on monitor mode and failed to catch this problem.

If I could ask something in return, since you use monitor mode, would you mind given me a report about how well monitor mode is working for you.

Regards,

Nick

SSBhaumik, 2022-01-01:
Monitor mode failed to enable as soon as i pressed enter my computer gave up.

SSBhaumik, 2022-01-01:
Monitor mode and packet injection both working with kali tools like wifipumpkin3, other applications like hostapd,  etc after installation of rincat, there are some problems with aireplay deauths maybe due to wifi6 protocols and channel hopping , appears to work sometimes with mdk4 -c h -B (bssid) but it's painfully slow most of the time.
One peculiar occurance with your driver, often times t3u archer plus v1 would give two wlan interfaces in iwconfig.

morrownr, 2022-01-01:
Hi @SSBhaumik 

Monitor mode in many of Realtek's out-of-kernel drivers can support a lot, but certainly not all, of the features that folks are looking for but there are some things you need to know. 

Many of the guides and instructions that comes with the various utilities and apps are written for Linux Wireless Standards compliant drivers. Realtek does not make Linux Wireless Standards compliant drivers. Some things work differently on Realtek drivers. I have places a document called `Monitor_Mode.md` in the `docs` folder to help with this problem. I have also included a script called `test-mon.sh` to help as well.

To summarize: The monitor mode here and in other Realtek drivers is good enough for many uses and will be good enough for many users. For those looking to get involved with security analysis/pen testing, you probably want to investigate better options. The following site has a LOT of information and links:

https://github.com/morrownr/USB-WiFi

Hope this helps