morrownr/88x2bu-20210702 #19: How to build with Clang / LLVM?

tinyreminder, 2022-01-02:
Hello,

I am using a kernel built with Clang / LLVM.

Wi-Fi driver installation failed because of compiler option for Clang
so I would like to build the driver with Clang / LLVM.
However I couldn't find how to.

Could I use Clang / LLVM?

Thank you in advance.

morrownr, 2022-01-02:
Hi @tinyreminder 

I have no experience with Clang but maybe we sort this out.

Are you following the installation instructions here? If so, maybe we should try a non-dkms install so that we can see what is going on. Run `sudo ./remove-driver.sh` and then the following from the driver directory:

```
$ make clean
$ make
```

Tell me what happens.

Regards

tinyreminder, 2022-01-02:
Thank you for a reply

This is an output when "make clean && make". 

  The kernel was built by: clang version 14.0.0
  You are using:           gcc (Debian 10.2.1-6) 10.2.1 20210110
  CC [M]  /root/src/clang/88x2bu-20210702-main/core/rtw_cmd.o
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-mno-global-merge’
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’

I was set CC=clang but GCC was used to make.
I couldn't find where GCC called.

Kind regards,

morrownr, 2022-01-03:
I suspect this is going to take some time and I don't have a lot of extra time right so let me throw a suggestion out. Go to:

https://github.com/morrownr/USB-WiFi

That repo has a lot of information and links to usb wifi adapters that work with Linux in-kernel drivers. That could be the best solution for now.

Regards

QueenPew, 2022-05-19:
do `EXPORT LLVM=1 && EXPORT LLVM_IS=1` and it is going to compile :P

morrownr, 2022-10-08:
@QueenPew 

Do you have time for a follow on question?

Running gcc here. Now seeing this:

gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’

The backend I am using is based on Ubuntu 22.04 (amd64) and is using gcc 11.2, I installed kernel 6.0 for testing. It is compiled with gcc 12.2.

A little research tells me there has been a discussion and maybe even the compiler option has been made mainline default. My questions:

How the hell do I override this because I have users of this driver and others that will be using old versions of gcc for years so I can't just upgrade gcc?

Where are the options such as this stored? Been using gcc for years and don't recall this coming up.

Thanks,

Nick

jurassicjordan, 2022-12-07:
Is this still being looked into? because I too am having a "gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’" error

morrownr, 2022-12-07:
Hi @jurassicjordan 

What you are seeing is likely due to a mismatch of compiler version vs the kernetl version. Can you post the output of the following?

```
gcc --version
uname -r
```

This kind of thing can happen when you upgrade to a recent mainline while the version of the compiler on your system stays the same.

ibanil, 2022-12-08:
Hi @morrownr 
I have the same problem, I think it's a version problem as you say. I have kernel 6 and the gcc version is 11.3.0, how can I update the gcc version to 12.2. ?
Thanks in advance

morrownr, 2022-12-08:
> Is this still being looked into? because I too am having a "gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’" error

https://linuxconfig.org/how-to-switch-between-multiple-gcc-and-g-compiler-versions-on-ubuntu-20-04-lts-focal-fossa

The above still works on Xubuntu 22.10, my main dev box. If you get a kernel that is simply too modern for the compiler that you have installed, you run into this kind of thing so I recommend following the guide and setting gcc to v12.

Let me know if this helps.

clayg, 2023-01-06:
i installed a more recent mainline kernel on jammy and the update-alternatives trick worked to switch gcc to gcc-12 (I couldn't figure out how to do it with env vars)

dolikemedo, 2023-07-11:
Installing gcc-12 and g++-12 worked for me too on ```5.19.17-051917-generic```,
Thanks @morrownr