morrownr/88x2bu-20210702 #20: RTW: ERROR [HALMAC][ERR]Dump efuse in suspend

mlewus, 2022-01-05:
I built a new kernel for the Raspberry pi 4 then built and  installed the 88x2bu-20210702 driver, for a TP-Link USB Wi-Fi adapter I have never used before. It seems to work fine as a wifi interface (no AP mode, no 867 MHz, no concurrent mode, USB2). But I’m getting kernel error messages in `/var/log/messages` and `dmesg` when the driver is loaded at boot.

The adapter is a TP-Link Archer 3U Plus, connected to a USB2 hub. I know USB2 is not ideal, but it would be difficult to replace it with a USB3 hub in my existing project. I am connected to a 2.4 GHz AP so throughput should not be an issue.

Thanks in advance for any help you can offer.

——- Config & Requested Info ——-
Pi 4, 4GB Ram, Pi OS 10 ‘Buster’ with upgraded kernel 5.10.89.
System was updated before kernel & driver build with
`sudo apt update && sudo apt full-upgrade `
Then kernel 5.10.89 was installed from the rpi-5.10.y branch at [https://github.com/raspberrypi/linux](https://github.com/raspberrypi/linux)
No errors in kernel or driver build. No other non-standard drivers have been installed.

Config: contents of /etc/modprobe.d/88x2bu.conf:
`options 88x2bu rtw_drv_log_level=2 rtw_led_ctrl=2 rtw_vht_enable=0 rtw_power_mgnt=0 rtw_beamform_cap=0 rtw_dfs_region_domain=1 rtw_sel_p2p_iface=1 rtw_switch_usb_mode=2`

Internal wifi disabled in /boot/config.txt with:
`dtoverlay=disable-wifi `

gcc version:
 `gcc version 7.3.0 (Raspbian 7.3.0-19)`

Here is the information requested in the README:

`$ uname -a`
`Linux magic-mirror 5.10.89-v7l+ #1 SMP Mon Jan 3 20:47:31 EST 2022 armv7l GNU/Linux`

`$ mokutil --sb-state`
`EFI variables are not supported on this system`

`$ lsusb `
` Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0556:0001 Asahi Kasei Microsystems Co., Ltd AK5370 I/F A/D Converter
Bus 001 Device 007: ID 2357:0138 TP-Link
Bus 001 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 003: ID 041e:324d Creative Technology, Ltd
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

` $ rfkill list all `
` 0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no`

` $ dkms status `
` rtl88x2bu, 5.13.1, 5.10.89-v7l+, armv7l: installed `

` $ iw dev `
` phy#0
        Interface wlx984827ea0ed6
                ifindex 3
                wdev 0x1
                addr 98:48:27:ea:0e:d6
                ssid xxxxxxxx
                type managed
                channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
                txpower 23.00 dBm `

——— ERROR Messages ———
There are numerous instances of 
[HALMAC][ERR]Dump efuse in suspend
and
[HALMAC][ALWAYS]shall R reg twice!!
in /var/log/messages at startup.  This is all of the output related to RTW:
Jan  4 11:03:42 magic-mirror kernel: [   50.211837] RTW: module init start
Jan  4 11:03:42 magic-mirror kernel: [   50.211858] RTW: rtl88x2bu v5.13.1-20-gbd7c7eb9d.20210702_COEX20210316-18317b7b
Jan  4 11:03:42 magic-mirror kernel: [   50.211874] RTW: rtl88x2bu BT-Coex version = COEX20210316-18317b7b
Jan  4 11:03:42 magic-mirror kernel: [   50.212337] RTW: [HALMAC]55772M
Jan  4 11:03:42 magic-mirror kernel: [   50.212337] HALMAC_MAJOR_VER = 1
[... 203 lines cut ...]
[   57.222835] RTW: ============ STA [98:da:c4:7c:df:e0]  ===================
[   57.222855] RTW: mac_id : 0
[   57.222870] RTW: wireless_mode : 0x0b
[   57.222886] RTW: mimo_type : 2
[   57.222901] RTW: static smps : N
[   57.222919] RTW: bw_mode : 20MHz, ra_bw_mode : 20MHz
[   57.222936] RTW: rate_id : 2
[   57.222952] RTW: rssi : -1 (%), rssi_level : 0
[   57.222968] RTW: is_support_sgi : Y, is_vht_enable : N
[   57.222985] RTW: disable_ra : N, disable_pt : N
[   57.223000] RTW: is_noisy : N
[   57.223015] RTW: txrx_state : 0
[   57.223032] RTW: curr_tx_rate : CCK_1M (L)
[   57.223047] RTW: curr_tx_bw : 20MHz
[   57.223062] RTW: curr_retry_ratio : 0
[   57.223078] RTW: ra_mask : 0x000000000fffffff

[   57.223270] RTW: recv eapol packet 1/4
[   57.225426] RTW: send eapol packet 2/4
[   57.233492] RTW: recv eapol packet 3/4
[   57.234239] RTW: send eapol packet 4/4
[   57.234619] RTW: set pairwise key camid:0, addr:98:da:c4:7c:df:e0, kid:0, type:AES
[   57.237709] RTW: set group key camid:1, addr:98:da:c4:7c:df:e0, kid:1, type:AES
[   67.024755] broken atomic modeset userspace detected, disabling atomic
[   70.699163] fuse: init (API version 7.32)

PieGuy314, 2022-01-05:
try rtw_switch_usb_mode=1 in '/etc/modprobe.d/88x2bu.conf'

PieGuy314, 2022-01-05:
The errors are consistent with the Wi-Fi adaptor failing during boot. The initial current draw on these adaptors is high. Having an unjuiced hub isn't going to help. The fact that it eventually works suggests a power supply issue. 

It's wrong to assume a USB port can power a hub with a load of gadgets hanging off it. Been there; learnt that.

morrownr, 2022-01-06:
Hi @mlewus 

> I tried that, it just breaks it. It crashes multiple times and gives up, with no wifi interface.

Ouch. Not a good idea to force USB3 when the adapter is not in a USB3 capable port.

> Or should I just disregard the error messages on startup?

This. I did not see any unknown entries and I did not see anything that looked like an operational bug report. Realtek's logging with this driver is not going to win them any prizes. I suggest:

 rtw_drv_log_level=0

...and press on. This is actually a pretty good driver for a Realtek out-of-kernel driver but that **** it spews in the log is just noise. I ignore it.

PieGuy314, 2022-01-06:
> try rtw_switch_usb_mode=1 in '/etc/modprobe.d/88x2bu.conf'

I think I meant to write 'rtw_switch_usb_mode=0'

Apologies.

mlewus, 2022-01-06:
i’ll try mode 0 but failing that, as long as it keeps working I’ll just ignore it.

morrownr, 2022-01-06:
Hey @PieGuy314 

Long time, no see. How is it going? I'm still waiting for you to rewrite the logging code in this driver... he he

mlewus, 2022-01-06:
Changing the USB mode from 2 to 0, no effect. But other than startup messages the adapter is reliable, which is way more than I can say for the internal Wi-Fi chip on the pi 4. That one was crashing on a daily basis, requiring me to rmmod/modprobe the driver to get it back up. This is way better. Thanks for all the work will you guys put into this!

closed 2022-01-06 by mlewus