morrownr/88x2bu-20210702 #22: Mode monitor airodump-ng wlan1 fixed channel 1

Appleuy, 2022-01-10:
How to fix Mode monitor airodump-ng wlan1 fixed channel 1

morrownr, 2022-01-10:
Please reply with the exact steps you are taking so that I can try to duplicate the problem.

Appleuy, 2022-01-11:
[image]
Wlan1 fixed channel 1

morrownr, 2022-01-11:
Sir, helping you is a challenge. You did not answer the question that I asked. Therefore my reply is not going to be a direct answer either:

https://www.aircrack-ng.org/doku.php?id=airodump-ng

Your answer is located at the above site. You are going to have to read to find the answer.

Another point that I want to make is that this site is for the support of this driver, The issue you reported is not an issue with this driver. The issue is that you failed to read and follow the documentation of the application you are using.

ivanovborislav, 2022-01-11:
The same driver, no fixed channel 1 issues.
`https://github.com/ivanovborislav/rtl88x2bu`

morrownr, 2022-01-11:
@ivanovborislav 

I'm sure there are no fixed channel issues with your driver. There are no fixed channel issues with this driver either.

The OP is not killing or stopping interfering processes before running airodump-ng. It has been recommended to him in another repo here to do so but he feels that it is not necessary. This is not a driver issue, it is an issue that the OP should take up in the Aircrack-ng community. I understand that inexperienced people need help and I do try to help but when they do not listen, it is very hard to help.

Appleuy, 2022-01-12:
I install this driver https://github.com/ivanovborislav/rtl88x2bu it have not wlan1 fixed channel 1 but driver for morrownr and rincat are  wlan1 fixed channel 1

Appleuy, 2022-01-12:
> Vui lòng trả lời cùng với các bước chính xác mà bạn đang thực hiện để tôi có thể tìm cách khắc phục sự cố.

I switch wlan1 to monitor mode then i type command airodump-ng wlan1 -c 10 It shows wlan1 fixed channel 1

morrownr, 2022-01-12:
Unable to duplicate. Works here. This appears to be a problem with not following the airodump-ng documentation. Closing.

closed 2022-01-12 by morrownr

PieGuy314, 2022-01-13:
MayB U provyd skriptz to hlp? <smiley face>

This is not an offer. I've dun the Goggle. I now has answrs.