morrownr/88x2bu-20210702 #24: (Information) Driver works on Ubuntu 18.04, Kernel 5.4 LTS and 5.7, but errors on 5.10 and 5.15

TbilisiHeat, 2022-01-19:
Nick,

I'm a relative Linux newbie, but wanted to mention that your driver and cilynx's won't install/build on my Ubuntu 18.04 system if I am running Kernels 5.10 OR 5.15.

Researched on github and forums for solutions and none worked.  Had EXACTLY the same problem as this gentleman:
https://github.com/morrownr/88x2bu-20210702/issues/11

Was using GCC version 11.2 and also fully updated headers/modules for kernels 5.10 and 5.15.  

Since I do not have the required technical skill to troubleshoot further, I cleaned my DKMS, uninstalled any failed driver installs, and rolled back my kernel to 5.4 LTS.  Et Voila, reinstallation worked perfectly.

I also wanted to take a moment to thank you for your efforts and for your investment in this driver.  It's true that Realtek devices are going to be more and more problematic for Linux users in the future, but your driver has saved me the expense of buying new hardware.  Appreciate you!

morrownr, 2022-01-19:
Hi @TbilisiHeat 

I'm pretty sure the issue is with gcc 11.2. Let me take a look at my current arrangement here to see how best to go ahead and test with gcc 11.2.

The problem is not with the kernels as there are many people running kernels all the way to and including 5.16. My main dev box has 5.13.

You are welcome. I don't know if you have taken a look around but I have several repos up:

https://github.com/morrownr?tab=repositories

I've used Linux since the mid-90's and was well aware of the struggles Linux users have had with USB WiFi so when COVID hit, it was the perfect opportunity to help Linux users and keep myself away from the virus. I am old enough that the virus could be a problem for me.

Stay tuned.

morrownr, 2022-01-19:
Well, that took less time that anticipated. I have to say that I was wrong, it is not gcc 11.2. The story:

I went to my little lab to see if I had any test systems that used gcc 11.2. I found it on a clean install of Ubuntu 21.10 amd64. I went about following the installation instructions in the README after plugging the 8812bu adapter in. I fully expect the compilation to blow up but it did not. It was a clean compile and I could find nothing wrong. After thinking about it, if there was a problem with Ubuntu 21.10, this and the other repos would be inundated with reports as this site gets thousands of hits per week... which is why I try to be very careful.

So, a clean compile with kernel 5.13 and gcc 11.2. Let's talk...

I noticed you said Ubuntu 18.04 and gcc 11.2 and then several LTS kernels. What that tells me is that you have a Frankenstein OS. You have parts from here and parts from there. I agree that it is cool to play and we can learn a lot from playing around but that makes it almost impossible for me to actually duplicate the problem.

Here is my recommendation: Save the important files from your current 18.04 installation and do a clean (format c:) type of installation of Ubuntu 21.10.

Fair enough?

Regards

TbilisiHeat, 2022-01-21:
Hi Nick,

I apologize but I made a small error in my original post.  I just ran gcc --version on my Ubuntu 18.04 machine and **_I was/am not running GCC 11.2_**.  I am running GCC 7.5.0.

I'm unable to gut my 18.04 LTS machine currently as its a production machine and I have some old "Frankensteined" toys on it that I can't afford to break.  Therefore, I pulled out my personal laptop, which has a relatively fresh Manjaro install, and upgraded the kernel to 5.15.  I also ran gcc --version and it's on 11.1.0.

I updated the Manjaro install's headers and tried to install your driver.  It built without issue, and then I plugged in the old Realtek/Edimax USB dongle from my Ubuntu 18.04 machine to test it out.

_**It worked perfectly and at full speed!**_

So my takeaway from this is now:
- Your driver does work on Kernel 5.15 (specifically, 5.15.12-1) (after I installed the correct headers)
- My Ubuntu 18.04 install's GCC version is super old.
- Your driver worked on my Manjaro machine with GCC 11.1.0.

So, it looks more and more likely that your suggestion is the explanation:  that my Ubuntu machine's old and/or "Frankenstein" status might be the cause of this driver being unable to be built on newer kernels.  Perhaps also, in order to build this on a newer kernel than 5.4 and 5.7, I will need to upgrade my GCC version on the Ubuntu machine.

morrownr, 2022-01-21:
> So, it looks more and more likely that your suggestion is the explanation: that my Ubuntu machine's old and/or "Frankenstein" status might be the cause of this driver being unable to be built on [...]

Let me tell you something that you know. If you have a production machine that needs  to remain stable and functional, you need to use another system for experimentation. The stock kernel for Ubuntu 18.04 is 5.4 I think so using 5.4 is a good idea.

Maybe you can find another system for experimentation. I know it is hard not to play around but as you found out in this case, it can lead to situations where it can be very difficult to figure out what the problem is.

Something that I recommend to those that like to expand their knowledge by upgrading specific parts of distros or by using non-mainstream distros, especially if using non-x86/amd64 processors, is to keep a usb wifi adapter around that uses in-kernel drivers. The in-kernel drivers can't be blown up like you saw in the case here. A good place to find usb wifi adapters that use in-kernel drivers is ...

https://github.com/morrownr/USB-WiFi

Regards

TbilisiHeat, 2022-01-21:
Nick,

Thanks for the advice.  I'm thinking the same way you are.  I already have a Panda wireless dongle that I use for emergencies (it's fully plug and play via the Linux kernel drivers), but the reason why I rely on you and your driver, is that my old Edimax dongle is 5ghz compatible, while my Panda is only on 2.4.  So - I need my Edimax (Realtek) working on my main machine (the Ubuntu 18.04 rig).  Thankfully, on Kernel 5.4 LTS, it does, and works great.

On top of that, as mentioned earlier, I have a personal laptop with fully locked and loaded Manjaro on it, which is my main machine for Linux gaming and playing around.

Given your advice and reminders, I will definitely stick with Kernel 5.4 LTS on Ubuntu 18.04 until I'm able to do a full wipe and install to the next Ubuntu (coming in soon in April?).  Also, will limit the tinkering to my Manjaro device.

Once again, thank you for your work here and for your support of Linux novices like myself!

TbilisiHeat, 2022-02-12:
Final note - nuked my Ubuntu 18.04 system finally and replaced with Manjaro Gnome on Kernel 5.15.  Also updated my laptop with Xubuntu 21.10, also on Kernel 5.15.  Installed your driver on both (after installing 5.15 kernel headers), install worked perfectly without incident and my dongle works at full speed with zero drops.

Thanks again for your work on this driver!

closed 2022-02-12 by TbilisiHeat

morrownr, 2022-02-12:
Hi @TbilisiHeat 

Glad to hear that things are looking good. It takes time to be fully comfortable with a new operating system. All I ask in return is that you take a little time to help novices once you have more experience.

Regards