morrownr/88x2bu-20210702 #27: Device is quite sensitive to orientation

callegar, 2022-01-24:
Hi, testing a BrosTrend AC1200 adapter that should have an RTL8812BU chipset.

The item appears to work with this driver, both as a USB2 and as a USB3 device. My issue is that it seems extremely sensitive to the antenna orientation. You get the orientation wrong (even by a few degrees) and immediately you get extremely erratic ping times while for another orientation the ping times are very repeatable. By erratic I mean that pinging a remote site periodically goes from the expected 100ms up to 1-2s. for 3-4 packets every 20-40 packets and then back to normal, which basically makes the connection unusable. When this happens, at times the connection breaks completely (network unreachable).

There are two weird things about this:
- when this happens there is no actual connection drop from the access point (i.e., you get network unreachable but the adapter stays connected to the router);
- when you get the erratic ping times, there is no sign of connection degradation in terms of "Signal strenght" reported by the adapter.

In the past I had a router doing something similar. It was meant to have a super-smart "adaptive" beamforming and noise filtering technology but the adaptive algorithm was periodically running into some crisis. Eventually, I found that it was possible to disable the adaptive behavior and with that the router behaved regularly.

Has anything like that been experienced by others with this chipset?
Are there any params controlling the beamforming or the adaptive behavior of the device?

morrownr, 2022-01-24:
> Are there any params controlling the beamforming or the adaptive behavior of the device?

Yes

sudo nano /etc/modprobe.d/88x2bu.conf

or from the driver directory:

sudo ./edit-options

The lines that matters is the one that starts with `option`. On that line:

Find: rtw_beamform_cap=10

Change it to: rtw_beamform_cap=0

The 88x2bu.conf file is where the documentation is located. Scroll down.

Regards

morrownr, 2022-02-16:
Has this issue been solved?

callegar, 2022-02-16:
Cannot say, I have returned the device and got a Mediatek based one. This works fine.

morrownr, 2022-02-17:
No problem about forgetting. I do like to clean things up once in a while so I thought I would check. 

You must have bought a CF-WU785AC. I have one. That cable is thick. Overall, I like the adapter.

closed 2022-03-14 by morrownr