morrownr/88x2bu-20210702 #29: Centos 8. tp-link AC1300 Archer T4U Plus

jenarogallegogit, 2022-01-29:
Hi all,
TP-Link AC1300 Archer t4U Plus up and running using this driver but taking into account this https://github.com/gordboy/rtl8812au-5.9.3.2/pull/11.
If additional info or questions just let me know.
Thanks in advance to everyone.
Info attached.
```
Linux jason 4.18.0-358.el8.x86_64 #1 SMP Mon Jan 10 13:11:20 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
SecureBoot disabled
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 004: ID 0b05:19af ASUSTek Computer, Inc. 
Bus 001 Device 003: ID 046d:c31d Logitech, Inc. Media Keyboard K200
Bus 001 Device 006: ID 2357:0115 TP-Link Archer T4U ver.3
Bus 001 Device 002: ID 04b3:3107 IBM Corp. ThinkPad 800dpi Optical Travel Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
8812au/5.6.4.2_35491.20191025, 4.18.0-358.el8.x86_64, x86_64: installed (WARNING! Diff between built and installed module!)
rtl88x2bu/5.13.1, 4.18.0-358.el8.x86_64, x86_64: installed (original_module exists)
phy#0
	Interface wlp0s20f0u10
		ifindex 3
		wdev 0x1
		addr 60:a4:b7:53:a7:7a
		ssid MiFibra-F95E
		type managed
		channel 108 (5540 MHz), width: 80 MHz, center1: 5530 MHz
		txpower 18.00 dBm
```

MidlifeRenaissanceMan, 2022-01-29:
Hey, I have the same USB dongle thingy 

I am testing with Rocky 8.5, the current downstream CentOS replacement for Big Blue Hat Enterprise Linux, a middle finger, to the middle finger Big Blue Hat gave us CentOS people a little over a year ago. 

Headed up by CentOS founder Greg Kurtzer, and named after the late early CentOS fellow founder Rocky McGaugh, it appears to be the dominant replacement for classic CentOS.....but I digress...

After purchasing one of these adapters, and being promised _Linux support_, I am currently getting a "missing argument to some function thingy" error

```
`/root/RTL8812BU/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/root/RTL8812BU/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:469:3: error: too few arguments to function ‘cfg80211_ch_switch_started_notify’
   cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /root/RTL8812BU/88x2bu-20210702/include/osdep_service_linux.h:98,
                 from /root/RTL8812BU/88x2bu-20210702/include/osdep_service.h:54,
                 from /root/RTL8812BU/88x2bu-20210702/include/drv_types.h:27,
                 from /root/RTL8812BU/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:7673:6: note: declared here
 void cfg80211_ch_switch_started_notify(struct net_device *dev,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:315: /root/RTL8812BU/88x2bu-20210702/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1571: _module_/root/RTL8812BU/88x2bu-20210702] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-348.7.1.el8_5.x86_64'
make: *** [Makefile:2510: modules] Error 2

```
Stuff you'll probably want to know....
```
[root@svr 88x2bu-20210702]# uname -a
Linux svr.esclub.local 4.18.0-348.7.1.el8_5.x86_64 #1 SMP Tue Dec 21 19:02:23 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

[root@svr 88x2bu-20210702]# mokutil --sb-state
EFI variables are not supported on this system

[root@svr 88x2bu-20210702]# lsusb
Bus 002 Device 002: ID 2357:0115 TP-Link Archer T4U ver.3
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 003: ID 04f2:b1ac Chicony Electronics Co., Ltd HP Laptop Integrated Webcam [2 MP Fixed]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 002: ID 148f:1000 Ralink Technology, Corp. Motorola BC4 Bluetooth 3.0+HS Adapter
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

[root@svr 88x2bu-20210702]# rfkill list all
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
2: hp-wifi: Wireless LAN
	Soft blocked: no
	Hard blocked: no
3: hp-bluetooth: Bluetooth
	Soft blocked: no
	Hard blocked: no

[root@svr 88x2bu-20210702]# dkms status

[root@svr 88x2bu-20210702]# iw dev
phy#0
	Interface wls1
		ifindex 3
		wdev 0x1
		addr 70:f3:95:e4:2f:6a
		ssid cubbyspace
		type managed
		channel 13 (2472 MHz), width: 40 MHz, center1: 2462 MHz
		txpower 36.00 dBm

[root@svr 88x2bu-20210702]# 
...

So any help getting this running would be superb.  

Any pointers, to what I should be looking for or things to try would be supurb.

BTW, the box is an old Compaq 620, 8Gb RAM, Core 2 Duo T6570 that I am testing on

Cheers
Chris

MidlifeRenaissanceMan, 2022-01-30:
Just checked with the current Rocky Kernel 4.18.0-348.12.2.el8_5.x86_64

Same issue.

MidlifeRenaissanceMan, 2022-01-30:
Dug in a little further. Had a look at ioctl_cfg80211.c line 469
Changed the compiler directive from looking for 5.11.0 to read this and it com 
...
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 11, 0))
...

Now to reboot and see if its worked 

Fingers crossed

MidlifeRenaissanceMan, 2022-01-30:
Seems to work. Connected on both 5GHz and 2,5GHz. 
I'll use it and see if I can break it. Report back with what I find 
Thank you for your courage

morrownr, 2022-01-31:
> Dug in a little further. Had a look at ioctl_cfg80211.c line 469 Changed the compiler directive from looking for 5.11.0 to read this and it com ... #if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 11, 0)) ...

Hi @MidlifeRenaissanceMan 

A couple of months back a user submitted a pull request to support RHEL8. I studied it a little and then tested it to make sure it would not break non-RHEL systems. My take on it is that it appears that RH (or Big Blue Hat if you prefer) has been selectively backporting certain things which is creating a hybrid kernel. That is a nightmare for those of us trying to maintain drivers such as this one. I'll keep my additional thoughts to myself and just offer this suggestion:

Most Linux users would be happier using USB WiFi adapters that use in-kernel drivers. This is especially true if using hardware that is not that common or if the distro is one that likes to tinker in ways that break things. There is a repo that has a lot of information and links to usb wifi adapters that use in-kernel drivers:

https://github.com/morrownr/USB-WiFi

MidlifeRenaissanceMan, 2022-01-31:
Cheers @morrownr 

As for your additional thoughts (thoughts++ ? No wait, that's incremental...ignore me) I am always open to stuff I may not agree with, because, usually people have come to these conclusions based on experience I have not yet had, and I am more intersted from learning from the wisdom, or what is usually, the prior mistakes of others, rather than having to learn myself the hard way, so feel free to share :-)

Yeah Big Blue Hat, as they are now, pretty much sit on a Kernel for 10 years, so their 4.18 Kernel is not so much the GNU Linux original Kernel 4.18, but something closer to a 5 Kernel that looks and feels like a 4 Kernel.....kind of how MATE desktop feels like Gnome 2, but is Gnome 3 under the hood.....but heaps more complicated....

The give away that you're in RHEL territory is the el8_5 in the 4.18.0-348.7.1.el8_5.x86_64 kernel ID, so a compiler directive that looked for that may be a simple answer for a _Buyer Beware_ fix. In future, I will be taking your advice and searching for devices with those chipsets. Much easier in kernel stuff

And thank you for the link. I did have a read through the list, however I had already made the purchase based on the "dude behind the counter" saying "Sure it has Linux support"......._maybe the Halo tee shirt was the give away_

Most of my commercial Linux work is on CentOS and now Rocky....because it doesn't change much...that and 1 key press Anaconda installs and the relative ease of creating and maintaining local repositories I have yet to figure out how to RPM something from sauce, but its on my list of skills to acquire this year......along with ditching the 10 Covid Kilos and improving my singing 

Nick, thanks for all your effort in making this thing work, and taking time out to answer the ramblings of some guy who would currently give RMS a solid run for his money on looking homeless, _but with none of the noncery_

Cheers, 73s and thank you for your courage

closed 2022-03-14 by morrownr