morrownr/88x2bu-20210702 #31: Kali Linux ARM64 Raspberry pi 4 2021.4 Re4son kernel - Addition Request

gdraper-cs, 2022-02-02:
Hi,

I am trying to use this driver suite with Kali Linux ARM64 Raspberry pi 4 2021.4 which unfortunately for me and any users in similar situation - has changed to the Re4son Kernel. 

What would need to be done to assist with the modification of this driver suite build / install scripts to support the Re4son kernel as part of kali-linux-2021.4-rpi-arm64.img?

For more information I am trying to setup Archer T4U Plus which may be suitable for this driver suite if the new addition can be made / assisted.

Thanks

gdraper-cs, 2022-02-02:
`DKMS make.log for rtl88x2bu-5.13.1 for kernel 5.4.83-Re4son-v8l+ (aarch64)
Wed Feb  2 02:46:55 AM UTC 2022
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/5.4.83-Re4son-v8l+/build M=/var/lib/dkms/rtl88x2bu/5.13.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.83-Re4son-v8l+'
Makefile:671: arch/aarch64/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/aarch64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.83-Re4son-v8l+'
make: *** [Makefile:2510: modules] Error 2
`

morrownr, 2022-02-02:
Hi @gdraper-cs 

I can only assist right now as my to-do list is far too long to dig in. If you are up for it, try this first:

Run from the driver directory:

```
sudo ./remove-driver.sh
sudo reboot
```
```
make
```

What we doing above is cleaning up and removing dkms from the picture. Let's see what happens when you run make.

gdraper-cs, 2022-02-02:
totally understand, i'll do my best to assist given i don't dive into kernels often yet and don't know your suite well.

The Makefile log shows
```
make[1]: Entering directory '/usr/src/linux-headers-5.4.83-Re4son-v8l+'
Makefile:671: arch/aarch64/Makefile: No such file or directory
```

the directory listing of /arch shown below

```
┌──(root💀wintermute)-[/usr/src/linux-headers-5.4.83-Re4son-v8l+/arch]
└─# ls -la
total 140
drwxr-xr-x 27 root root  4096 Dec  4 01:08 .
drwxr-xr-x 23 root root  4096 Dec  4 01:08 ..
drwxr-xr-x  7 root root  4096 Dec  4 01:08 alpha
drwxr-xr-x 11 root root  4096 Dec  4 01:08 arc
drwxr-xr-x 94 root root  4096 Dec  4 01:08 arm
drwxr-xr-x 10 root root  4096 Dec  4 01:08 arm64
drwxr-xr-x  6 root root  4096 Dec  4 01:08 c6x
drwxr-xr-x  7 root root  4096 Dec  4 01:08 csky
drwxr-xr-x  5 root root  4096 Dec  4 01:08 h8300
drwxr-xr-x  5 root root  4096 Dec  4 01:08 hexagon
drwxr-xr-x  9 root root  4096 Dec  4 01:08 ia64
-rw-r--r--  1 root root 29144 May 15  2021 Kconfig
drwxr-xr-x 23 root root  4096 Dec  4 01:08 m68k
drwxr-xr-x  7 root root  4096 Dec  4 01:08 microblaze
drwxr-xr-x 49 root root  4096 Dec  4 01:08 mips
drwxr-xr-x  6 root root  4096 Dec  4 01:08 nds32
drwxr-xr-x  6 root root  4096 Dec  4 01:08 nios2
drwxr-xr-x  5 root root  4096 Dec  4 01:08 openrisc
drwxr-xr-x  7 root root  4096 Dec  4 01:08 parisc
drwxr-xr-x 15 root root  4096 Dec  4 01:08 powerpc
drwxr-xr-x  6 root root  4096 Dec  4 01:08 riscv
drwxr-xr-x 16 root root  4096 Dec  4 01:08 s390
drwxr-xr-x 12 root root  4096 Dec  4 01:08 sh
drwxr-xr-x 12 root root  4096 Dec  4 01:08 sparc
drwxr-xr-x  6 root root  4096 Dec  4 01:08 um
drwxr-xr-x  5 root root  4096 Dec  4 01:08 unicore32
drwxr-xr-x 24 root root  4096 Dec  4 01:08 x86
drwxr-xr-x  7 root root  4096 Dec  4 01:08 xtensa
```
I've done some debugging and it appears to occur during dkms build command and in particular its looking for an 

`arch/aarch64/Makefile`

Digging through the file system shows it should be `arch/arm64/Makefile`

```
Running install-driver.sh version 20220108
Operating System: Kali GNU/Linux Rolling
5.4.83-Re4son-v8l+
aarch64
Starting installation...
Copying source files to: /usr/src/rtl88x2bu-5.13.1
Copying 88x2bu.conf to: /etc/modprobe.d
before dkms add
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1
after dkms add - before error occurred
before dkms build

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.4.83-Re4son-v8l+ KSRC=/lib/modules/5.4.83-Re4son-v8l+/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.83-Re4son-v8l+ (aarch64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
after dkms build before error occurred
An error occurred. dkms build error = 10
Please report this error.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

What should i be changing from aarch64 to amd64? the `/usr/src/linux-headers-5.4.83-Re4son-v8l+/arch/amd64` directory? if so will this break something in the kernel or is this just source code of the kernel leading to mapping directly into the kernel? Should I be changing something leading into dkms build in the  `./install-driver.sh`?

morrownr, 2022-02-02:
> I've done some debugging and it appears to occur during dkms build command 

Yes but when we started using `make` we are not using dkms to keep things simple. So disregard dkms for now.

Anytime you want to try a compile, use:

```
make clean
make
```
> What should i be changing from aarch64 to amd64?

Hold on. Slow down for a minute. What I would do at this point is investigate if the kernel headers are installed.

gdraper-cs, 2022-02-02:
actually i miswrote - it should be "what should i change from aarch64 to arm64"
ok:
1) make clean complete
2) ./install-driver.sh - same outcome
3) the kernel headers appear to be installed as I have confirmed `apt-get install kalipi-kernel-headers` which is the re4son-kernel package name for the kernel headers package from the re4son-kernel repository

morrownr, 2022-02-02:
> ./install-driver.sh - same outcome

Recommend you hold off on using the installation script until such time as the problem is discovered.

With that in mind, let's clean up and stay with a manual installation:

1. sudo ./remove-driver.sh
2. make clean

Remember to always run `make clean` before running `make`.

`make` starts the manual compile process. It is highly likely that once we fix the manual compile process, it will work with the installation script but we do not need or want the added complexity of dkms for now.

Given what I have seen so far, I would be investigating settings in the `Makefile`. Find the following two sections:

```
###################### Platform Related #######################
CONFIG_PLATFORM_I386_PC = n
CONFIG_PLATFORM_ARM_RPI = n
CONFIG_PLATFORM_ARM64_RPI = y
```

The above section in your `Makefile` should mirror what you see above. If not, show me what you have.

Another section of the `Makefile` to look at is:

```
### START RASPBERRY PI OS SUPPORT

ifeq ($(CONFIG_PLATFORM_ARM_RPI), y)
ifeq ($(CONFIG_CONCURRENT_MODE), y)
EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE
endif
EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
#EXTRA_CFLAGS += -DPLATFORM_LINUX
ARCH ?= arm
CROSS_COMPILE ?=
KVER ?= $(shell uname -r)
KSRC := /lib/modules/$(KVER)/build
MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/
INSTALL_PREFIX :=
endif

ifeq ($(CONFIG_PLATFORM_ARM64_RPI), y)
ifeq ($(CONFIG_CONCURRENT_MODE), y)
EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE
endif
EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
#EXTRA_CFLAGS += -DPLATFORM_LINUX
EXTRA_CFLAGS += -fno-stack-protector
ARCH ?= arm64
CROSS_COMPILE ?=
KVER ?= $(shell uname -r)
KSRC := /lib/modules/$(KVER)/build
MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/
INSTALL_PREFIX :=
endif

### END RASPBERRY PI OS SUPPORT
```

The second group in the above section is the one that should be used. It should be investigated. It has been a while since I loaded the 64 bit beta of the RasPiOS but this was working when last I checked. I suspect it is still working because this repo is VERY heavily used so I would expect reports from RasPiOS users if it is not working.

Let me know how far you get.

gdraper-cs, 2022-02-03:
Good news, I changed the config to this

```###################### Platform Related #######################
CONFIG_PLATFORM_I386_PC = n
CONFIG_PLATFORM_ARM_RPI = n
CONFIG_PLATFORM_ARM64_RPI = y
```
and then confirmed that the second code block in the make file was also what you pasted for the RASPBERRY PI OS SUPPORT  section where CONFIG_PLATFORM_ARM64_RPI = y

I was successfully able to 1) `make clean` and then 2) `make` 

er so now what? So that `make` built the files but do i need to symlink / install them anywhere?

morrownr, 2022-02-03:
> I was successfully able to 1) make clean and then 2) make

> er so now what?

make clean
sudo ./install-driver.sh

Let me know what happens.

Side note: If the above results in a good installation, please advice me on how to rewrite Step 9 of the Installation.

Regards

gdraper-cs, 2022-02-03:
ah genius it works!

calls itself: TPLink Archer T4U v3  in the NetworkManager dropdown (is sold as Archer T4U Plus) and was able to connect to a 5GHz ssid.

So: for future search - TPLink Archer T4U Plus works on Kali Rpi ARM64 Re4son kernel.

Looks like I'd add the line into Installation Step 9 to include the name `Kali 2021.4 ARM64 RPi 4 Re4son Kernel` or similar because it looks like this could have been avoided if i just ran the ./raspiOS-64.sh

or alternatively create a `./kali-rpi-ARM.sh` and `./kali-rpi-ARM64.sh`

I think i skimmed past ./raspi0s-64.sh and discounted it because it looked like it was using Raspberry Pi OS (ie new version of Raspbian) which is a different kernel than kali 2021.4 Re4son

but thank you very much for your help

Just checking for offensive capabilities:
As an aside: does this driver package support VIF (Virtual Interface)? Or is it a limitation of the chipset?

can close this :-)

morrownr, 2022-02-07:
> Looks like I'd add the line into Installation Step 9 to include the name Kali 2021.4 ARM64 RPi 4 Re4son Kernel or similar because it looks like this could have been avoided if i just ran the ./raspiOS-64.sh

I'm going to do some editing on step 9.

> As an aside: does this driver package support VIF (Virtual Interface)? Or is it a limitation of the chipset?

It is a limitation of this driver and something Realtek would have to address. For adapters that support VIF:

https://github.com/morrownr/USB-WiFi

> can close this :-)

How about I mark it solved and leave it available for now.

kalmufti, 2022-12-02:
I have the same issue with Alpine Linux, any help would be appreciated. Very easy to replicate my setup, I'm running a vm with standard Alpine Linux iso, then use ```setup-alpine``` to install pretty much everything with default settings then choose  ```sys``` hdd setup. 

```bash
localhost:~/88x2bu-20210702# ./install-driver.sh
Running install-driver.sh version 20221126
5.15.80-0-lts
x86_64
Starting installation...
Copying rtw88_8822bu.conf to: /etc/modprobe.d
Copying 88x2bu.conf to: /etc/modprobe.d
The non-dkms installation routines are in use.
#make -C /lib/modules/5.15.80-0-lts/build M=/root/88x2bu-20210702 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.80-0-lts/build M=/root/88x2bu-20210702  modules
make[1]: Entering directory '/lib/modules/5.15.80-0-lts/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/5.15.80-0-lts/build'
make: *** [Makefile:2503: modules] Error 2
An error occurred. Error = 2
Please report this error.
Please copy all screen output and paste it into the report.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

```bash
localhost:~/88x2bu-20210702# uname -a
Linux localhost 5.15.80-0-lts #1-Alpine SMP Tue, 29 Nov 2022 08:08:44 +0000 x86_64 Linux
localhost:~/88x2bu-20210702#
```

DRIgnazGortngschirl, 2023-06-01:
Hiii

Old issue but came this far with all the solutions provided. 

```diff
...
DKMS: add completed.
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
./dkms-make.sh....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.103-Re4son-v7+ (armv7l)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
+ root@MayerGangen:~/src/88x2bu-20210702# cat /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log
DKMS make.log for rtl88x2bu-5.13.1 for kernel 5.10.103-Re4son-v7+ (armv7l)
Thu  1 Jun 22:01:19 BST 2023
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.10.103-Re4son-v7+/build M=/var/lib/dkms/rtl88x2bu/5.13.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.103-Re4son-v7+'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_security.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:280: /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_cmd.o] Error 1
make[2]: *** Waiting for unfinished jobs....
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:280: /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_security.o] Error 1
make[1]: *** [Makefile:1825: /var/lib/dkms/rtl88x2bu/5.13.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.103-Re4son-v7+'
make: *** [Makefile:2552: modules] Error 2
```

In my Makefile initially, there were no config options with 

```
CONFIG_PLATFORM_ARM_RPI = n
CONFIG_PLATFORM_ARM64_RPI = y
```

Also the huge section with "### START RASPBERRY PI OS SUPPORT" not.

I just added it😬...

morrownr, 2023-06-02:
Hi @DRIgnazGortngschirl 

It appears that you are using an old version of the driver.

Do you see the following line in Makefile?

CONFIG_PLATFORM_AUTODETECT = y

You might want to run remove-driver.sh and delete the driver you have then run the install steps again.

I don't have much time right now but can explain when able.

DRIgnazGortngschirl, 2023-06-02:
Hello @morrownr 

I do indeed have this line in the Make file. But removing the driver with the script and trying to install it again did not have any effect on the outcome.

DRIgnazGortngschirl, 2023-06-02:
Sadly I can't really as I am using an image to be able to use https://pwnagotchi.ai/ all other options I am simply too bad to make it work with my little knowledge. But no worries, maybe I will find another way / buy just a different antenna to make it work more easily. 
Thanks !

morrownr, 2023-06-03:
You can figure this out.

Okay, so it seems you need to stick with Kali 64 for RasPi4b. Is a version available that is up to kernel 6.3 yet? If so, you don't need this driver, there is a driver in kernel 6.3.

morrownr, 2023-06-03:
If you are interested in getting another adapter, let me point you to the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

Menu item 2 is a long list of adapters that use in-kernel drivers. Take a look and if you want to describe to me what you need the adapter to do, I can probably make a recommendation or two.

morrownr, 2023-06-03:
Can I get you to run and post the result of the following command:

$ uname -m

I noticed that you are running a 32 bit kernel yet it is being detected as 64 bit. That is key to the problem.

DRIgnazGortngschirl, 2023-06-03:
Thanks for pointing this repo out!

> Take a look and if you want to describe to me what you need the adapter to do, I can probably make a recommendation or two.

I just need an adapter that is better than the onboard on of a Raspberry Pi 3b and also has the possibility to go into monitoring mode. 

```
root@pwnagotchi:/home/pi# uname -m
armv7l
root@pwnagotchi:/home/pi# uname -a
Linux MayerGangen 4.19.127-Re4son-v7+ #1 SMP Sun Feb 21 14:14:43 AEDT 2021 armv7l GNU/Linux
```

BR Mario

morrownr, 2023-06-03:
NR Mario

> I just need an adapter that is better than the onboard on of a Raspberry Pi 3b and also has the possibility to go into monitoring mode.

The adapter that I recommend as the single best adapter for monitor mode is the:

ALFA AWUS036ACHM

You can read about it in the Plug and Play list. Menu item 2. It has the best range of any modern dual band adapter that I have tested.

In general, all of the adapters in the Plug and Play list support monitor mode but since you seem to be locked into a specific kernel, we need to check exactly what that kernel supports. It should support the ALFA adapter as shown above but we should look to make sure the driver and firmware is there. I can provide instructions to do that if you need it. I can also provide other recommendations if the ALFA adapter is not available or if your budget won't cover it.

Regards

DRIgnazGortngschirl, 2023-06-03:
Hiii!

Thanks so much, yeah that's a bit high budget, looking more like an antenna with direct USB attachment to somewhat be comfortable to transport. 

To stop annoying @gdraper-cs and others want to chat on Telegram t.me/MarioMichel or somewhere else ? 

Best regards Mario

morrownr, 2023-06-03:
This might work better if you tell me what is available to you. Online source? I can look and recommend. Generally speaking, an adapter with a mt7610u chipset would likely work and some are available at lower prices.

DRIgnazGortngschirl, 2023-06-07:
@morrownr 

Sorry for long time no response.

I have avaliable Amazon and all the hardware shops in Austria like conrad.at and mediamarkt.at.

morrownr, 2023-06-15:
Sorry for the delay as I am very busy. Which Amazon url do you use?

DRIgnazGortngschirl, 2023-06-15:
> Sorry for the delay as I am very busy. Which Amazon url do you use?

amazon.de

morrownr, 2023-06-15:
amazon.de

My favorite adapter with my RasPi4B is the following:

https://www.amazon.de/-/en/AWUS036ACM-802-11ac-MiMo-Wi-Fi-Adapter-black/dp/B08DV22HB2/ref=sr_1_53?crid=2QKW6UL04WAUA&keywords=usb+wifi+adapters+alfa+panda&qid=1686864073&sprefix=usb+wifi+adapters+alfa+panda%2Caps%2C186&sr=8-53

That is probably more than you need with a RasPi3B. You are looking for monitor mode but do you want dual band or just single band?

DRIgnazGortngschirl, 2023-06-15:
Dual Band is fine but maybe over budged for this will see, maybe I will give it a try, thank you so much for all the help !!!

morrownr, 2023-06-17:
Here is a link to a low cost adapter that is plug and play and supports monitor mode:

https://www.aliexpress.us/item/3256804283254522.html?gatewayAdapt=glo2usa

Read the reviews. This adapter is WiFi 6 and WiFi 6e on Linux as long as your distro is running kernel 5.19 or later.

DRIgnazGortngschirl, 2023-06-20:
@morrownr 

Tried the AWUS036ACM adapter. 

And was not able to get it to work. On Windows it instantaneously lights up the blue light on the RPi happens, nothing is just being detected. 
```
root@MayerGangen:~# lsusb
Bus 001 Device 006: ID 0e8d:7612 MediaTek Inc.
```

DRIgnazGortngschirl, 2023-06-20:
```
root@MayerGangen:~# iw dev
phy#0
        Interface mon0
                ifindex 4
                wdev 0x2
                addr b8:27:eb:cc:e9:13
                type monitor
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr b8:27:eb:cc:e9:13
                type managed
root@MayerGangen:~# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.100.4  netmask 255.255.255.0  broadcast 192.168.100.255
        inet6 fe80::ba27:ebff:fe99:bc46  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:99:bc:46  txqueuelen 1000  (Ethernet)
        RX packets 249  bytes 42626 (41.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 187  bytes 28067 (27.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 3492  bytes 9135135 (8.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3492  bytes 9135135 (8.7 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

mon0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        unspec B8-27-EB-CC-E9-13-30-30-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 2081  bytes 334858 (327.0 KiB)
        RX errors 0  dropped 34  overruns 0  frame 0
        TX packets 520  bytes 187791 (183.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 10.0.0.2  netmask 255.255.255.0  broadcast 10.0.0.255
        ether 76:d1:53:6c:e7:48  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

The mon interface is wlan0 of the RPi3

morrownr, 2023-06-20:
> The mon interface is wlan0 of the RPi3

Are you saying that the mon0 interface is the RPi3 internal radio?

> Interface wlan0

What is the this?

DRIgnazGortngschirl, 2023-06-21:
```
root@MayerGangen:~#  lsmod | grep mt7
mt76x2u                28672  0
mt76_usb               28672  1 mt76x2u
mt76x2_common          45056  1 mt76x2u
mt76                   45056  3 mt76_usb,mt76x2u,mt76x2_common
mac80211              790528  4 mt76_usb,mt76,mt76x2u,mt76x2_common
cfg80211              643072  5 mt76,mac80211,mt76x2u,mt76x2_common,brcmfmac

```

morrownr, 2023-06-21:
I have a diagnosis: The adapter is working.

The below two sections tell me the adapter is working and you have an interface to use.

```
root@MayerGangen:~#  lsmod | grep mt7
mt76x2u                28672  0
mt76_usb               28672  1 mt76x2u
mt76x2_common          45056  1 mt76x2u
mt76                   45056  3 mt76_usb,mt76x2u,mt76x2_common
mac80211              790528  4 mt76_usb,mt76,mt76x2u,mt76x2_common
cfg80211              643072  5 mt76,mac80211,mt76x2u,mt76x2_common,brcmfmac

        Interface wlan0
                ifindex 3
                wdev 0x1
                addr b8:27:eb:cc:e9:13
                type managed
```

You mentioned that the blue light is not working. Not sure if it is a lack of API support at this point or other issue but the mt7610u, mt7612u and mt7921au drivers do not currently support driver based LED light support. On the mt7612u based adapters like your ACM, there is a way to turn it on:

https://github.com/morrownr/7612u

Go to Known Issue 1.

FYI: I have several repos here with a lot of info.

DRIgnazGortngschirl, 2023-06-21:
Sadly not as I have a better cap web UI and the onboard interface can continue on finding / searching the SSID even tho I disconnected the adapter so wlan0 and mon0 are not the adapters interface.

DRIgnazGortngschirl, 2023-06-21:
At my Debian System, it gets detected differently.

```
root@mm-vie-srv-01:~# lsusb
Bus 003 Device 002: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter
root@mm-vie-srv-01:~# uname -a
Linux mm-vie-srv-01 5.15.0-69-generic #76-Ubuntu SMP Fri Mar 17 17:19:29 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
root@mm-vie-srv-01:~#  lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.2 LTS
Release:        22.04
Codename:       jammy
```

morrownr, 2023-06-21:
Let me make a suggestion:

Grab another sd card and burn the Raspberry Pi OS 64 bit 2023-05-03. Set it up and plug the adapter it. It should work with no issues. This would give you the ability to learn what you should be seeing.