morrownr/88x2bu-20210702 #35: [Compile] Debian arm64 feat Raspberry Pi 4

DoM1niC, 2022-02-16:
Anyidea about this error another Drivers have the same problem here...

### Running install-driver.sh version 20220108
**Operating System:** Debian GNU/Linux bookworm/sid
**Kernel:** 5.15.23-v8+
**arch:** aarch64 (arm64)
**Device:** Raspberry Pi 4 

```
 MODPOST /var/lib/dkms/rtl88x2bu/5.13.1/build/Module.symvers
ERROR: modpost: "__const_udelay" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "__arch_copy_from_user" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "_raw_spin_unlock" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "__rcu_read_lock" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "__rcu_read_unlock" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "cpu_hwcap_keys" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "csum_ipv6_magic" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "__warn_printk" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "arm64_const_caps_ready" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
ERROR: modpost: "__udelay" [/var/lib/dkms/rtl88x2bu/5.13.1/build/88x2bu.ko] undefined!
WARNING: modpost: suppressed 2 unresolved symbol warnings because there were too many)
make[2]: *** [scripts/Makefile.modpost:134: /var/lib/dkms/rtl88x2bu/5.13.1/build/Module.symvers] Error 1
make[2]: *** Deleting file '/var/lib/dkms/rtl88x2bu/5.13.1/build/Module.symvers'
make[1]: *** [Makefile:1783: modules] Error 2
make[1]: Leaving directory '/root/linux-bd14681a067cd3d91f7ca664c6858bcbac806adb'
make: *** [Makefile:2510: modules] Error 2
```

morrownr, 2022-02-16:
Hi @DoM1niC 

I'll try to help.

> Operating System: Debian GNU/Linux bookworm/sid
> Kernel: 5.15.23-v8+ (latest rpi-update / rpi-source Kernel Headers)

The above is causing some confusion in my mind. On the one hand you are indicating that you are using the ARM64 version of Debian and on the other hand you are saying that the kernel and headers are from rpi. 

The Raspberry Pi OS and Debian ARM64 are two different operating systems and have differences. Please clarify which operating system you are using.

Regards

DoM1niC, 2022-02-16:
> Hi @DoM1niC
>
> I'll try to help.
>
>
> The above is causing some confusion in my mind. On the one hand you are indicating that you are using the ARM64 version of Debian and on the other hand you are [...]

I use a own Debian based on sid (Bookworm) a clean arm64 Bootstrapped Image with the original Kernel from Raspbian and the Modules forced with x64 = arch64

morrownr, 2022-02-16:
Okay, so a Frankenstein OS. So how did you handle installation Steps 3 and 9?

Should I ask why you are combining parts of 2 OS's?

DoM1niC, 2022-02-17:
> Okay, so a Frankenstein OS. So how did you handle installation Steps 3 and 9?
> 
> Should I ask why you are combining parts of 2 OS's?

I reinstall my Pi now with native Raspbian bootstrapped and all works with Bookworm now .... I have no idea whats wrong but ;) not all seems to work fine with my RPi 4 ... but THX anyway

closed 2022-02-17 by DoM1niC