morrownr/88x2bu-20210702 #36: Problem Reporting Guidelines - Please Read.

morrownr, 2022-02-16:
When reporting problems, remember that the people that will attempt to help you do not know what Linux distro or computer you have. Painting a picture that includes the computer you are using, the distro you are using and a brief description of the problem is important. The information provided by the below command should also be provided with the problem report: Please run the following command and paste all output in your problem report:

```
sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
```

To reiterate: remember to include:

- A brief description of the problem.
- The name and version of the Linux distro you are using.
- The name and model of the computer you are using.
- The information obtained with the above command.

Thank you

kurahaupo, 2022-09-22:
It might be helpful to incorporate this guidance into the repo's [issue template](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository).

kurahaupo, 2022-09-22:
For those of us on systems without `mokutil` or `dkms`, are there any equivalents that you'd like us to run instead?

morrownr, 2022-09-23:
> For those of us on systems without `mokutil` or `dkms`, are there any equivalents that you'd like us to run instead?

Not really. We'll just handle it.