morrownr/88x2bu-20210702 #39: new commit will let aircrack-ng can‘t switch

233lol, 2022-03-02:


morrownr, 2022-03-02:
Hi @233lol 

I need more information.

What version of Kali are you using?

Show me how you are trying to switch to monitor mode.

Regards

233lol, 2022-03-03:
> Hi @233lol
> 
> I need more information.
> 
> What version of Kali are you using?
> 
> Show me how you are trying to switch to monitor mode.
> 
> Regards

Linux kali 5.16.0-kali1-amd64 #1 SMP PREEMPT Debian 5.16.7-2kali1 (2022-02-10) x86_64 GNU/Linux

`sudo airmon-ng check kill `
`sudo airmon-ng start wlan0`

morrownr, 2022-03-03:
I am sending you the most up to date version of start-mon.sh below:

[start-mon.sh.tar.gz](https://github.com/morrownr/88x2bu-20210702/files/8174013/start-mon.sh.tar.gz)

Save it to your home folder and unzip it.

To run it:

$ sudo ./start-mon.sh <wlan0>

This script should take your intended interface into monitor mode and let you run apps as desired.

I recommend you do not use the above lines that you posted as Realtek drivers are not Linux standards compliant and I can't fix that so what I have done is work on this script to allow people easily enter monitor mode and go about their business.

The best answer for those that make heavy use of monitor mode is to get an adapter that is standards compliant:

https://github.com/morrownr/USB-WiFi

My personal favorite adapter for monitor mode is the ALFA ACHM.

Let me know if you have problems with the script.

Regards

233lol, 2022-03-03:
> 

I 

> I am sending you the most up to date version of start-mon.sh below:
>
> [start-mon.sh.tar.gz](https://github.com/morrownr/88x2bu-20210702/files/8174013/start-mon.sh.tar.gz)
>
> Save it to your home folder [...]

I also used this script, but still can't switch. Only older versions can.

morrownr, 2022-03-03:
I keep with Kali available for testing but I need to finish a couple of things and reconfigure the test machine that has Kali on it so I will test this as soon as possible.

I'm not seeing this issue on the systems that were immediately available.

Regards

emil10150, 2022-03-04:
#Fix:
cd 88x2bu-20210702;
sed -i 's/CONFIG_WIFI_MONITOR = n/CONFIG_WIFI_MONITOR = y/g' Makefile;
sudo ./install-driver.sh;

morrownr, 2022-03-04:
Hi @emil10150 @233lol 

At about the same time Emil10150 made his/her post above, I had discovered the problem. I maintain multiple copies of the driver locally for different projects and some how I seem to have pushed some information from the wrong repo. It should be fixed now.

Thanks for the help.

Regards