morrownr/88x2bu-20210702 #41: Unable to connect to WiFi

rundnc, 2022-03-20:
I followed the instructions that you provided for Ubuntu. The driver seemed to install correctly. The wifi adapter LED is now blinking and it is able see available wifi connections. The problem is that I am unable to connect to the wifi. It repeatedly asks for the wifi password and when I enter in the correct password, it asks again. 

### -I am using Ubuntu 21.10 Impish Indri 

**INFO FROM REQUESTED COMMAND**
Linux nick-Z390-GAMING-X 5.13.0-35-generic #40-Ubuntu SMP Mon Mar 7 08:03:10 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
SecureBoot disabled
Platform is in Setup Mode
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 1058:264d Western Digital Technologies, Inc. easystore 264D
Bus 001 Device 005: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 004: ID 6557:1621 Emtec USB DISK 2.0
Bus 001 Device 008: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Bus 001 Device 007: ID 1b1c:0c22 Corsair iCUE H150i RGB PRO XT Liquid CPU Cooler
Bus 001 Device 009: ID 046d:c315 Logitech, Inc. Classic Keyboard 200
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
1: phy1: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl88x2bu, 5.13.1, 5.13.0-35-generic, x86_64: installed
phy#1
	Interface wlx1cbfce64a99c
		ifindex 4
		wdev 0x100000001
		addr 1c:bf:ce:64:a9:9c
		type managed
		txpower -100.00 dBm

rundnc, 2022-03-20:
Here is a link to the product that I bought just over a year ago. It does not seem to be available for purchase anymore.

https://www.walmart.ca/en/ip/BraceTek-WiFi-Adapter-for-PC-Gaming-1300Mbps-with-USB-Wireless-Adapter-Dual-Band-5GHz-802-11-AC-WiFi-Dongle-5dBi/PRD40QXPDGEZIR7

morrownr, 2022-03-20:
Hi @rundnc 

We probably don't need additional info about the adapter. Let me through out a few things you can check:

- Check the security settings in your wifi router/ap. Make sure the setting is WPA2, not WPA3 orWPA3-WPA2 mixed.

- Check to see if `802.11w Management Frame Protection` is turned on. This is also in the wifi router/ap. If on, turn it off.

- Did you reboot the system after installing the driver? If not, do so.

Regards

rundnc, 2022-03-20:
So, I'm not sure if this is possible or not but are there different passwords for different wifi settings?

I checked the back of the router, entered in the WPA2 wifi password and now it is working. What seems weird about this is that I've never used that password for my phone or even on this machine.

Either way, thank you for your help!

morrownr, 2022-03-20:
Glad you are going. Wrong password would certainly cause what you described. I did not think to mention that.

You probably connected your phone with WPS. 

One thing that sometimes causes problems is when both 2.4 GHz and 5 GHz use the same SSID. If your router does this, you might want to think about changing the name of the one on 5 GHz to add a `-5` to the name so you can tell which one is which. It helps with troubleshooting.

Cheers