morrownr/88x2bu-20210702 #42: disconnect on openwrt

Boos4721, 2022-03-26:
OPENWRT LOG:
```
[   32.376478] usb 5-1: new high-speed USB device number 26 using xhci-hcd
[   32.811283] usb 5-1: USB disconnect, device number 26

```
UNAME -A:
`Linux OpenWrt 5.10.108 #0 SMP PREEMPT Sat Mar 26 10:20:17 2022 aarch64 GNU/Linux`

IF CONNECT: lsusb -t
```
/sys/bus/usb/devices/5-1/configuration: No such file or directory
/sys/bus/usb/devices/5-1/devnum: No such file or directory
/sys/bus/usb/devices/5-1/idProduct: No such file or directory
/sys/bus/usb/devices/5-1/idVendor: No such file or directory
/sys/bus/usb/devices/5-1/maxchild: No such file or directory
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 0, If 0, Class=, Driver=rtl88x2bu, M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
```
DISCONNET: lsusb -t
```
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
```
MY DEVICE:
`rockchip friendlyarm nanopi R4S`

[image]

MY OPENWRT PACKAGE:
https://github.com/Boos4721/openwrt/tree/master/package/kernel/rtl88x2bu

thanks for your job !

morrownr, 2022-03-27:
Hi @Boos4721 

I took a look at the modifications that you made. I'm curious as to what exactly was causing the problem? Were you able to determine the root cause?

Regards

Boos4721, 2022-03-28:
> Hi @Boos4721
> 
> I took a look at the modifications that you made. I'm curious as to what exactly was causing the problem? Were you able to determine the root cause?
> 
> Regards

im not sure.

Boos4721, 2022-03-28:
> Hi @Boos4721
> 
> I took a look at the modifications that you made. I'm curious as to what exactly was causing the problem? Were you able to determine the root cause?
> 
> Regards

i guess abt lastest mac80211, i use 5.15.8

Boos4721, 2022-03-28:
> > Hi @Boos4721
> > I took a look at the modifications that you made. I'm curious as to what exactly was causing the problem? Were you able to determine the root cause?
> > Regards
> 
> i guess abt lastest mac80211, i use 5.15.8

Boos4721, 2022-03-28:
I tried Linux 5.4-y with mac80211 5.10 **WORKS**
Linux 5.10 with mac80211 5.15 **DISCONNET**

Boos4721, 2022-03-28:
[image]

morrownr, 2022-03-28:
Hi @Boos4721 
Hi @nospam2k

So, what it appears that you are telling me is that compiling this driver works with openWRT 21.02 but nothing later at this point. That is the current stable so things appear good.

I added @nospam2k as he asked me if I knew anything about compiling the 8814au driver for openWRT. I use openWRT but am not a dev on openWRT so maybe you can help him head the right direction?

Nick

nospam2k, 2022-03-28:
Thanks @morrownr !

Yes, @Boos4721 as mentioned. I'm interested in compiling the 8814au for openWRT. Any hints will be a great help. I'm just running a normal compile for rPi 4 B right now just to get familiar.

Boos4721, 2022-03-29:
> 

https://github.com/Boos4721/openwrt/tree/master/package/kernel/rtl8812au-ct

Boos4721, 2022-03-29:
> Thanks @morrownr !
> 
> Yes, @Boos4721 as mentioned. I'm interested in compiling the 8814au for openWRT. Any hints will be a great help. I'm just running a normal compile for rPi 4 B right now just to get familiar.

https://github.com/Boos4721/openwrt/tree/master/package/kernel/rtl8812au-ac

nospam2k, 2022-03-29:
No, neither of those bring up wlan0.

nospam2k, 2022-03-29:
I do see this in dmesg:

[    7.382948] usbcore: registered new interface driver rtl88XXau
[    7.399005] usbcore: registered new interface driver brcmfmac

Boos4721, 2022-03-29:
> 

it looks like the module is loaded

nospam2k, 2022-03-29:
I had similar issues when trying to use the rtl8812au driver just compiling in rPi. I compiled morrownr's 8814au driver and it came up right away. You have any troubleshooting ideas for getting a wlan0?

nospam2k, 2022-03-29:
On a side note on the drivers loading, I think they load because I used * and not M. I've disabled the internal wifi and I think that is the brcmfmac, right?

Boos4721, 2022-03-29:
> I had similar issues when trying to use the rtl8812au driver just compiling in rPi. I compiled morrownr's 8814au driver and it came up right away. You have any troubleshooting ideas for getting a wlan0?
[image]
my rPI works.

nospam2k, 2022-03-29:
I'm using a different usb-nic
Bus 001 Device 003: ID 0bda:8813 Realtek 802.11ac NIC

nospam2k, 2022-03-29:
Just to confirm this is an rtl8814au chip not rtl8812.

Boos4721, 2022-03-29:
> Just to confirm this is an rtl8814au chip not rtl8812.

ok i dont abt rtl8814au, sorry

nospam2k, 2022-03-29:
ahh ok. I thought that was why morrownr included me in this. ok thx.

nospam2k, 2022-03-29:
Just digging through comparing makefiles and saw this in the rtl8812au makefile:

CONFIG_RTL8812A = y
CONFIG_RTL8821A = y
CONFIG_RTL8814A = n

Is this anything? If so, since it is downloaded, how can I change the package makefile to enable it?

Boos4721, 2022-03-29:
> Just digging through comparing makefiles and saw this in the rtl8812au makefile:
> 
> CONFIG_RTL8812A = y CONFIG_RTL8821A = y CONFIG_RTL8814A = n
> 
> Is this anything? If so, since it is downloaded, how can I change the package makefile to enable it?

u may edit ur config.

closed 2022-03-29 by Boos4721

nospam2k, 2022-04-01:
@Boos4721 - I was able to compile your driver for RTL8814! It shows as wlan0. I'm stuggling with other configuration issues. I can get a scan to work so I'm getting there. I don't know how to configure as 2 radios. I'd like to email you if you wouldn't mind and just ping you as I hit roadblocks. I'm also having issues with connecting as an AP as well so I may have more huddles to jump. I had to edit a few hal things as well. I really don't know what I'm doing technically. I just compared them to the 8812 hal files and took my best guess. I'm glad to send you the patches if you are interested.

I'm also interested in getting the 8814au driver to compile but I'm getting errors that I see others have run into on the internet having to do with redefinitions. I'd like you thoughts on why this can't be fixed. It seems to be what is being said but no technical explanation. I'm committed to doing this and I have the time. I may not do it if I can get the 8812 driver to fully work with my 8814 adapter.

I'm sorry to bug you but I can tell it will be a very long process trying to use OPENWRT forum. I really appreciate the help with OPENWRT side of things. I can debug the compiling things but I don't understand enough of the OPENWRT build system. Thx!

Boos4721, 2022-04-01:
> @Boos4721 - I was able to compile the driver for RTL8814! It shows as wlan0. I'm stuggling with other configuration issues. I can get a scan to work so I'm getting there. I don't know how to configure as 2 radios. I'd like to email you if you wouldn't mind and just ping you as I hit roadblocks. I'm also having issues with connecting as an AP as well so I may have more huddles to jump. I had to edit a few hal things as well. I really don't know what I'm doing technically. I just compared them to the 8812 hal files and took my best guess. I'm glad to send you the patches if you are interested.
> 
> I'm also interested in getting the 8814au driver to compile but I'm getting errors that I see others have run into on the internet having to do with redefinitions. I'd like you thoughts on why this can't be fixed. It seems to be what is being said but no technical explanation. I'm committed to doing this and I have the time. I may not do it if I can get the 8812 driver to fully work with my 8814 adapter.
> 
> I'm sorry to bug you but I can tell it will be a very long process trying to use OPENWRT forum. I really appreciate the help with OPENWRT side of things. I can debug the compiling things but I don't understand enough of the OPENWRT build system. Thx!

cool! i dont mind sending email, so any AP MODE logs?

nospam2k, 2022-04-01:
Cool! - you can email me at mar312021 at daford dot com. I'll get stuff to you then.

ThiloteE, 2022-04-01:
Would be cool, if you could post your final solution, here (or in the corresponding driver repository), when you are done. I kindly ask you to include the guides you have used to come this far. I have found some guides about building an OpenWRT from scratch [here](https://openwrt.org/docs/guide-developer/start#using_the_toolchain). Are these the ones you are using?

Boos4721, 2022-04-01:
> Would be cool, if you could post your final solution, here (or in the corresponding driver repository), when you are done. I kindly ask you to include the guides you have used to come this far. I have found some guides about building OpenWRT from scratch [here](https://openwrt.org/docs/guide-developer/start#using_the_toolchain). Are these the ones you are using?

humm, i compile toolchain to build package

nospam2k, 2022-04-03:
Yes. I've been keeping track of what I've done. I am using those guides. I'm not sure where I should actually post my findings because I already feel like I'm corrupting this post. I've kinda lost momentum on this because I think I've found this whole process is a dead end as it seems that USB device aren't dual physical chip and so can't provide simultaneous 2.4 and 5 on different SSID's. For now, I'm just going to upload a zip of my "package". Just drop it in the packages/kernel directory and then make menuco
nfig and select the entry for rtl8814au and it should compile following the guides. Please note this is UNTESTED. I never really tried to get it to work as an AP. I was able to get wlan0 and scan, but I've not done any more. Also, this is really a kludge. I have several patches that were best guess and the last few hal patches should be combined.

[rtl8814au.tar.gz](https://github.com/morrownr/88x2bu-20210702/files/8403712/rtl8814au.tar.gz)

Boos4721, 2022-04-04:
> Yes. I've been keeping track of what I've done. I am using those guides. I'm not sure where I should actually post my findings because I already feel like I'm corrupting this post. I've kinda lost [...]

GB