morrownr/88x2bu-20210702 #43: Raspberry Pi 4 using 64bit  - Compiles, installs but no wlan0

nospam2k, 2022-03-28:
5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux

dmesg doesn't show anything re registering of rtl88x2bu.

Let me know what info you need.

morrownr, 2022-03-28:
Show me the results of:

`lsusb`

nospam2k, 2022-03-28:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

nospam2k, 2022-03-28:
dmesg | grep rtl88
pi@webprojector:~ $ 

Nothing in dmesg about the driver

nospam2k, 2022-03-28:
pi@webprojector:~ $ lsmod | grep 88
pi@webprojector:~ $ 

Module no loaded either

nospam2k, 2022-03-28:
Different things I've tried (in order):

pi@webprojector:~ $ sudo modprobe 88x2bu
pi@webprojector:~ $ 

pi@webprojector:~ $ lsmod | grep 88
88x2bu               3174400  0
cfg80211              860160  1 88x2bu

pi@webprojector:~ $ sudo ip link set wlan0 up
Cannot find device "wlan0"

morrownr, 2022-03-28:
> ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU

You are installing the wrong driver. See:

https://github.com/morrownr/8814au

nospam2k, 2022-03-28:
Yep. Duh. Sorry. I was following a tutorial that linked back to the wrong driver. Working with correct driver (of course!) https://tutorialforlinux.com/2021/02/05/realtek-rtl8814au-wifi-driver-raspberry-pi-installation/

closed 2022-03-28 by nospam2k