morrownr/88x2bu-20210702 #45: Is it possible to install this driver in nethunter?

EDLLT, 2022-04-14:
Is it possible to install the driver on a phone?

I have an S20 FE 5G but whenever I plug in the wifi adapter to my phone the wrong driver shows up, 88XXau instead of 88XXBu

I tested this driver on my computer and it worked perfectly, but I don't know how to get it working on my phone

morrownr, 2022-04-14:
Hi @EDLLT 

> Is it possible to install the driver on a phone?

Doing so would be a big challenge.

If you would outline what you are trying to do, I might be able to offer some advice.

Regards

EDLLT, 2022-04-15:
> Hi @EDLLT
> 
> > Is it possible to install the driver on a phone?
> 
> Doing so would be a big challenge.
> 
> If you would outline what you are trying to do, I might be able to offer some advice.
> 
> Regards

Alright so, I have an S20 FE 5G and I installed this custom kernel on my phone, along with a full chroot installation
https://forum.xda-developers.com/t/kernel-nethunter-for-galaxy-s20-fe-5g-r8q-snapdragon.4205881/

The wireless adapter is an Archer T4U Plus V3, it has the 8822bu chipset

running `uname -m` on the phone gives me
`aarch64`

then `./install-driver.sh`
```
`Running install-driver.sh version 20220108
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
4.19.113-Nethunter_WirusMOD_r8q_v2.3
aarch64
Starting installation...
Copying source files to: /usr/src/rtl88x2bu-5.13.1
Copying 88x2bu.conf to: /etc/modprobe.d
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j6 KVER=4.19.113-Nethunter_WirusMOD_r8q_v2.3 KSRC=/lib/modules/4.19.113-Nethunter_WirusMOD_r8q_v2.3/build....................
cleaning build area...

88x2bu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.113-Nethunter_WirusMOD_r8q_v2.3/kernel/drivers/net/wireless/

depmod...
The driver was installed successfully.
Do you want to edit the driver options file now? [y/N] y
Do you want to reboot now? (recommended) [y/N] N`
```

But the driver didn't install correctly and the wifi adapter won't work
so I tried the manual build instructions

Running `make clean`
```
#make -C /lib/modules/4.19.113-Nethunter_WirusMOD_r8q_v2.3/build M=/root/src/88x2bu-20210702 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
```

[... 176 lines cut ...]
  CC [M]  /root/src/88x2bu-20210702/hal/phydm/rtl8822b/phydm_regconfig8822b.o
  CC [M]  /root/src/88x2bu-20210702/hal/phydm/rtl8822b/phydm_rtl8822b.o
  CC [M]  /root/src/88x2bu-20210702/hal/phydm/txbf/haltxbf8822b.o
  CC [M]  /root/src/88x2bu-20210702/hal/btc/halbtc8822bwifionly.o
  CC [M]  /root/src/88x2bu-20210702/hal/btc/halbtc8822b1ant.o
  CC [M]  /root/src/88x2bu-20210702/hal/btc/halbtc8822b2ant.o
  CC [M]  /root/src/88x2bu-20210702/platform/platform_ops.o
  LD [M]  /root/src/88x2bu-20210702/88x2bu.o
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
  MODPOST /root/src/88x2bu-20210702/Module.symvers
  CC [M]  /root/src/88x2bu-20210702/88x2bu.mod.o
  LD [M]  /root/src/88x2bu-20210702/88x2bu.ko
  BTF [M] /root/src/88x2bu-20210702/88x2bu.ko
Skipping BTF generation for /root/src/88x2bu-20210702/88x2bu.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.16.0-kali7-arm64'
```

Running `make install`

```
install -p -m 644 88x2bu.ko  /lib/modules/4.19.113-Nethunter_WirusMOD_r8q_v2.3/kernel/drivers/net/wireless/
install: target '/lib/modules/4.19.113-Nethunter_WirusMOD_r8q_v2.3/kernel/drivers/net/wireless/' is not a directory: No such file or directory
make: *** [Makefile:2503: install] Error 1
```

EDLLT, 2022-04-15:
Running this command `sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev` gives me
(I covered some info that I thought were sensitive like mac addresses or SSIDs

```
Linux kali 4.19.113-Nethunter_WirusMOD_r8q_v2.3 #1 SMP PREEMPT Mon May 24 10:10:48 CEST 2021 aarch64 GNU/Linux
EFI variables are not supported on this system
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2357:0115 TP-Link Archer T4U ver.3
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: bt_power: Bluetooth
        Soft blocked: yes
        Hard blocked: no
1: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
2: phy1: Wireless LAN
        Soft blocked: no
        Hard blocked: no
realtek-rtl8814au/5.8.5.1~git20210730.824d1d2, 4.19.113-Nethunter_WirusMOD_r8q_v2.3, aarch64: installed
rtl88x2bu/git, 4.19.113-Nethunter_WirusMOD_r8q_v2.3, aarch64: built
phy#1
        Interface wlan1
                ifindex 37
                wdev 0x100000001
                addr 00:e0:4c:8b:29:b8
                type managed
                txpower 20.00 dBm
phy#0
        Interface wifi-aware0
                ifindex 33
                wdev 0x4
                addr ##:##:##:##:##:##
                type NAN
        Interface p2p0
                ifindex 32
                wdev 0x3
                addr ##:##:##:##:##:##
                type managed
        Interface swlan0
                ifindex 31
                wdev 0x2
                addr ##:##:##:##:##:##
                ssid ############
                type AP
                txpower 40.00 dBm
        Interface wlan0
                ifindex 30
                wdev 0x1
                addr ##:##:##:##:##:##
                ssid ##########
                type managed
                txpower 0.00 dBm
```

morrownr, 2022-04-15:
Hi @EDLLT 

There is a lot to unpack here. I will say that you are closer than I expected.

Let me spell out what I think the details are at this point: (And you correct what is wrong)

You have a Samsung phone with an ARM64 (aarch64) processor.

You are installing what looks to be a version of Kali Linux that has been modified to work on this phone. Further explanation of what Nethunter is would be helpful.

A look at the details:

> 4.19.113-Nethunter_WirusMOD_r8q_v2.3

My script is identifying the above as the kernel in use.

> make[1]: Entering directory '/usr/src/linux-headers-5.16.0-kali7-arm64'
> /bin/sh: 1: /usr/src/linux-headers-5.16.0-kali7-common/scripts/pahole-flags.sh: not found
> /bin/sh: 1: [...]

However, make is using the above kernel headers. Kernel headers need to match the kernel so where are the kernel headers for kernel 4.19?

> install: target '/lib/modules/4.19.113-Nethunter_WirusMOD_r8q_v2.3/kernel/drivers/net/wireless/' is not a directory: No such file or directory

Well, we are trying to install a module to a directory that does not exist. This means you have a non-standard setup. We need to find where the directory for wireless modules is located. You are going to have to search.

You might be able to create the missing directory but there is still the issue of having the wrong headers installed. You might want to ask the makers of the os you installed about that.

One other point if you address these issues to the makers, if they would just compile the usb drivers in `mt76`, then folks wanting to do what you are doing could just grab an adapter with mt7612u or mt7610u chipsets and all would be plug and play. For more information on said chipsets and links to adapters:

https://github.com/morrownr/USB-WiFi

Almost forgot: To clean things up, you need to run the below scripts:

remove-driver.sh
remove-driver-no-dkms.sh

Regards

EDLLT, 2022-04-15:
As for this `4.19.113-Nethunter_WirusMOD_r8q_v2.3`, it's a symlink pointing to the headers `/usr/src/linux-headers-5.16.0-kali7-arm64`, im the one who made the symlink as I figured that the script was probably looking for it
I believe that these are the correct headers for my system as they were automatically provided once I installed the kernel

As for the directories, each directory it couldn't find, I made them then restarted the installation process

Finally I tried to run `make install`, however it still didn't work
```
install -p -m 644 88x2bu.ko  /lib/modules/4.19.113-Nethunter_WirusMOD_r8q_v2.3/kernel/drivers/net/wireless/
/sbin/depmod -a 4.19.113-Nethunter_WirusMOD_r8q_v2.3
```

I found a directory on my phone in `/vendor/firmware_mnt/image` and it had a bunch of files which I think are chipset names, most of them ended with .bin
[image]

At this point I tried to contact the OS maker, hopefully they will respond. Other than that I don't know what to do as everything that I tried failed

morrownr, 2022-04-17:
Hi again @EDLLT 

> As for this `4.19.113-Nethunter_WirusMOD_r8q_v2.3`, it's a symlink pointing to the headers `/usr/src/linux-headers-5.16.0-kali7-arm64`, im the one who made the symlink

Okay, that explains a lot. I was wondering how the results I looked at came about.

> as I figured that the script was probably looking for it I believe that these are the correct headers for my system as they were automatically provided once I installed the kernel

The headers have to match the kernel. If you are running the 4.19 kernel, the headers have to be v4.19. This is an issue you will have to sort out with the folks that wrote the os you are using.

> I found a directory on my phone in `/vendor/firmware_mnt/image` and it had a bunch of files which I think are chipset names, most of them ended with .bin [image]

Yes, these are firmware files for the in-kernel drivers. Out-of-kernel drivers do not work the same way. Those firmware files are not in their normal locations. If you have an amd64 installation of Debian or Kali, I can point out to you where the normal location of the firmware files is and also the location of the in-kernel drivers.
 
> At this point I tried to contact the OS maker, hopefully they will respond. Other than that I don't know what to do as 
everything that I tried failed

I looked up Nethunter. It is interesting. A little pen testing I see. Based on everything I have seen in this thread and what I know from previous exeperience, I know what direction I would go with Nethunter:

I would use the rtl8812bu based adapter for something else and get an adapter with an in-kernel driver as the compatibility for monitor mode projects is far better. The specific adapter I would use in this case is the ALFA AWUS036ACHM as it has fantastic range and uses less power that many of the alternatives. Here is a link:

https://store.rokland.com/collections/wi-fi-usb-adapters/products/alfa-awus036achm-802-11ac-dual-band-high-power-ac1200-mediatek-wifi-usb-adapter

If you go to the following site and go down to the `chipset - Mediatek mt7610u` section you can see additional links and other mt7610u based adapters:

https://github.com/morrownr/USB-WiFi

For some reason, the mt7610u firmware was not included in Debian and Kali until about 1.5 years ago while the driver was included. That meant users had to download and install the firmware until that point. I suspect the mt7610u driver is included in your Nethunter but things seem to be in different places than the standard amd64 Debian and Kali distros. I'm willing to go on a hunt with you if you want. First I will explain where the driver and firmware should be and then you can search for where they really are.

Regards

EDLLT, 2022-04-22:
Thank you very much for the help

Thanks to sVirusX[the nethunter kernel dev] I was able to get it working on my phone, he added support to your driver in the kernel

also, apparently from what I understood there is a chroot environment and the actual kernel environment
I was trying to install it in the chroot which is one of the reasons it had been failing

So, thank you once again for the driver, I really appreciate it!

closed 2022-04-22 by EDLLT

morrownr, 2022-04-22:
Glad it worked out for you.

Regards