morrownr/88x2bu-20210702 #47: Can't check the link speed in Debian

sugo512, 2022-04-27:
I have this installed on OMV 5 Debian. It connects but the link is slow. I don't have access to the router I connect to (shared) and I can't see the link speed in Debian CLI

Any ideas?

morrownr, 2022-04-28:
Try this:

$ sudo apt install wavemon
$ wavemon

Tell me what the numbers show.

Regards

sugo512, 2022-04-28:
[image]

sugo512, 2022-04-28:
Linux omv 5.10.0-0.bpo.12-amd64 #1 SMP Debian 5.10.103-1~bpo10+1 (2022-03-08) x86_64 GNU/Linux
SecureBoot disabled
Bus 002 Device 002: ID 0b05:184c ASUSTek Computer, Inc.
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

rtl88x2bu, 5.13.1, 5.10.0-0.bpo.12-amd64, x86_64: installed

morrownr, 2022-04-28:
Thanks for posting the screen output of wavemon.

The first thing that catches my attention is `signal level -67 dBm` . That tells me that the wifi signal is poor. This is caused by distance and/or obstructions like walls. Questions:

Is this adapter plugged into a desktop or laptop?

Does the adapter have a single or dual antenna design?

What is wall made of?

Do you have options as far as moving the computer to try to get a better signal?

Run `$ iw dev` and post the results.

Do you have an Android phone? If so, there are some apps called `wifi analyzer` install one and see if you can tell how many systems are on channel 149 to `165`. This technology shares bandwidth. You may have a congestion problem.

Regards

sugo512, 2022-04-28:
Machine is a HP t620 Plus mini. The adapter is a mini USB Asus AC53, no antenna.  My router belongs to my landlord and I can't get access to it, but other machines in the same area connect at 520 or even 866. I'll move the machine around to see if I can get a better sig.

My phone says that there is no other APs at 149-165, my neighbors are pretty far away.

phy#0
        Interface wlxfc34972d5468
                ifindex 3
                wdev 0x1
                addr fc:34:97:2d:54:68
                ssid NETGEAR55-5G-2
                type managed
                channel 153 (5765 MHz), width: 80 MHz, center1: 5775 MHz
                txpower 15.00 dBm

sugo512, 2022-04-28:
[image]

Moved closer

morrownr, 2022-04-28:
-57 dBm is much better. You should see better speeds.

> but I still wonder what the link speed is?

I can tell you a few ways to check it but I am not familiar with OMV5. You might want to ask on the forums for the distro. I will say that the reason you don't see it in wavemon as these Realtek drivers are not fully standards compliant.

I see that is a nano adapter. I have a couple and enjoy using them it the situation is right but they are pretty much same room devices as far as signal is concerned. If you decide at some point to get an adapter with longer range, I keep the follow repo for Linux users looking for usb wifi adapters:

https://github.com/morrownr/USB-WiFi

sugo512, 2022-04-28:
It's just Debian - I've tried a bunch of commands - nothing.

Aloha!

morrownr, 2022-04-28:
Try this:

$ nmcli dev wifi

That should work if the desktop you are using uses Network Manager.

Are you located in the middle of the Pacific ocean?

sugo512, 2022-04-28:
I don't have network manager installed now, but I did try it, no luck with nmcli.  OMV doesn't use a Desktop - just a web interface.

Yes, Kaneohe.

morrownr, 2022-04-29:
Hi @sugo512 

I was reading earlier this week about the water troubles on your island. I hope there is a good solution.

Is you speed better now with the better signal?

Regards