morrownr/88x2bu-20210702 #48: ERROR (dkms apport): binary package for rtl8821au: 5.12.5.2 not found (Build fail with dkms or no_dkms)

johnnietien, 2022-05-04:
I try so to install the driver on WSL2 Ubuntu 20.04 and get fail with auto dkms or no_dksm.sh and also install by manual but still got no luck.
$lsusb  I got :

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:b82c Realtek Semiconductor Corp. 802.11ac+Bluetooth 5.0 Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

then I do
$sudo modprobe btusb 
and I got 
$dmesg | egrep -i 'blue|btusb'

[ 2821.834001] usb 1-1: Product: 802.11ac+Bluetooth 5.0 Adapter
[ 2822.005394] Bluetooth: hci0: RTL: examining hci_ver=07 hci_rev=000b lmp_ver=07 lmp_subver=8822
[ 2822.111394] Bluetooth: hci0: RTL: rom_version status=0 version=2
[ 2822.112016] Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_fw.bin
[ 2822.112029] bluetooth hci0: Direct firmware load for rtl_bt/rtl8822b_fw.bin failed with error -2
[ 2822.112655] Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8822b_fw.bin not found

after some internet check some solution I try to manage get some driver and found there is many firmware in my folder but have no idea still got error cannot found [ 2822.112655] Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8822b_fw.bin not found

$ ls /usr/lib/firmware/rtl_bt/

rtl8192ee_fw.bin  rtl8723b_fw.bin                rtl8723d_config.bin  rtl8812ae_fw.bin     rtl8821c_config.bin  rtl8822b_fw.bin       rtl8822cu_config.bin  rtl8852au_fw.bin
rtl8192eu_fw.bin  rtl8723bs_config-OBDA8723.bin  rtl8723d_fw.bin      rtl8821a_config.bin  rtl8821c_fw.bin      rtl8822cs_config.bin  rtl8822cu_fw.bin
rtl8723a_fw.bin   rtl8723bs_fw.bin               rtl8761a_fw.bin      rtl8821a_fw.bin      rtl8822b_config.bin  rtl8822cs_fw.bin      rtl8852au_config.bin

but the bluetoothctl and hcitool dev cannot work because cannot found the bluetooth adapter.
try to remove driver and install again still cannot solve the problem.

Please check a log file below and help me fix that build error
Thanks in advance.
this is my log file:

DKMS make.log for rtl8821au-5.12.5.2 for kernel 5.10.102.1-microsoft-standard-WSL2 (x86_64)
Thu May  5 05:45:49 +07 2022
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.102.1-microsoft-standard-WSL2/build M=/var/lib/dkms/rtl8821au/5.12.5.2/build  modules
make[1]: Entering directory '/home/johnnietien/WSL2-Linux-Kernel'
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/monitor/rtw_radiotap.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/wds/rtw_wds.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8821au/5.12.5.2/build/core/rtw_btcoex.o
[... 136 lines cut ...]
ERROR: modpost: "cfg80211_roamed" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "cfg80211_scan_done" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "cfg80211_ibss_joined" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "cfg80211_michael_mic_failure" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "cfg80211_disconnected" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "__cfg80211_send_event_skb" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "ieee80211_freq_khz_to_channel" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "cfg80211_get_bss" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "wiphy_new_nm" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "wiphy_free" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "cfg80211_ready_on_channel" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "ieee80211_get_channel_khz" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "cfg80211_ch_switch_started_notify" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "cfg80211_ch_switch_notify" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "wiphy_unregister" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "cfg80211_del_sta_sinfo" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "cfg80211_vendor_cmd_reply" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "cfg80211_remain_on_channel_expired" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "wiphy_register" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
ERROR: modpost: "__cfg80211_alloc_event_skb" [/var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:124: /var/lib/dkms/rtl8821au/5.12.5.2/build/Module.symvers] Error 1
make[2]: *** Deleting file '/var/lib/dkms/rtl8821au/5.12.5.2/build/Module.symvers'
make[1]: *** [Makefile:1726: modules] Error 2
make[1]: Leaving directory '/home/johnnietien/WSL2-Linux-Kernel'
make: *** [Makefile:2510: modules] Error 2

morrownr, 2022-05-05:
Hi @johnnietien 

>  kernel 5.10.102.1-microsoft-standard-WSL2 (x86_64)

There is the problem. 

I respect your right to use any operating system you want but please respect my right to refuse to work problems that have anything to do with Microsoft. That company has moral and ethical problems and they have been convicted of crimes in federal court in the US.  I have watched for 40 years and cannot in good conscience install a Microsoft OS to work a problem, even if it is based on Linux.

I can suggest numerous good Linux distros if you are interested.

Good luck

johnnietien, 2022-05-05:
@morrownr 
ok thanks for your reply soon. but FYI I 've found this issued " firmware file rtl_bt/rtl8822b_fw.bin not found " also happen with somebody using other Realtek chip in other Linux distros after upgrade or build new kernel. 

Anyway thanks for your feedback. I will try my luck in another way.

morrownr, 2022-05-06:
I read the case that you posted but that case is not about a failure to compile, it is about missing firmware. I can't recall ever seeing a case where missing firmware causes a failure to compile. That is not how it works. That is also a somewhat old case and it is with a different driver. So, that case is not relevant in my opinion.

What I suspect is causing the problem is the development environment. In the installation section of README.md:

Step 3: Install the required packages...

...I give instruction on how to setup the dev environment for various distros. I have no idea what the instructions need to be for your distro because I have no knowledge of how your distro is put together. I don't even know what package manager it uses. Is it based on a well known distro like Debian or is it a unique distro?

Regards

morrownr, 2022-05-06:
Probably the best advice I can give you is this. Go to the following repo:

https://github.com/morrownr/USB-WiFi

Click on menu item 2 and get yourself a good usb wifi adapter that uses in-kernel drivers. The amount of time you are likely to spend on this issue is likely not worth it when you could have an adapter that just works when plugged in.

Menu item 1 gives a lot of information about USB WiFi adapters and Linux. It is a good read and will help you understand a lot about using USB WiFi adapters with Linux.

Regards

closed 2022-05-23 by morrownr