morrownr/88x2bu-20210702 #49: CentOS 7 unable to install RTL8812BU

leozrucan, 2022-05-07:
Hello, I am quite new to linux, so may have some dumb errors. I am running CentOS 7 and was trying to install the driver but when I used automated script it gave me dkms error =10, I tried to run it manually through make but also without success. please find the output of the terminal in the file attached. \I would appreciate if you can advice where I should move from here. Thank you for your help and detailed instructions
[file.txt](https://github.com/morrownr/88x2bu-20210702/files/8644500/file.txt)

morrownr, 2022-05-07:
Hi @leozrucan 

Welcome to Linux.

I took a look at `file.txt`. Then I remembered that CentOS is based on RHEL. Then I saw that the kernel is v3.10.

This is not what you want to hear I'm sure but this driver is never going to work on CentOS 7. This driver does support RHEL 8.5 but @misha4gps is the person that provided the pull request to support that. I have no experience with RHEL myself and the product is a commercial product and I have more than enough to keep me busy anyway.

I'm going to make a suggestion: Linux is very cool to learn and the benefits are many. However, some Linux distros are better for those new to the OS. I've only been using Linux since 1994 but I have learned a few things. My advice is to find another distro to learn with. A good one to take a look at would be Linux Mint:

https://linuxmint.com/

If you need to Learn RHEL/CentOS, you could run it in a VM under Mint. This would allow you to use this driver with your adapter and enjoy a much more modern distro while still allowing you to learn RHEL/CentOS

Regards

leozrucan, 2022-05-08:
Thank you Nick for your quick response. I had mint installed before but
ended up changing distros due to issues webcams. I guess I will give it
another try. Thanks again for your help.

morrownr, 2022-05-10:
Hi again @leozrucan 

Mint is designed to be very stable so major updates only happen every two years which means it is possible specific minor issues may not be fixed in a timely manner. A new major update to Mint will be coming in June.

An alternative that has already seen a major update is ubuntucinnamon which can be found at the following site:

https://ubuntucinnamon.org/

This distro is very similar to Mint.

Regards

leozrucan, 2022-05-22:
Thanks, I'll check it out.

jdemars-netscout, 2023-12-10:
By "I'm sure but this driver is never going to work on CentOS 7" do you mean that the driver is dependent on kernel items that are not present in the older kernel (centos7)?

morrownr, 2023-12-10:
Hi @jdemars-netscout 

The fundamental problem is that Realtek builds these out of kernel drivers for the Linux Mainline Kernel. That is the kernel maintained by and released by a guy by the name of Linus Torvalds. Red Hat may start with a Mainline kernel but they modify it in ways that make it incompatible with the Mainline kernel. They pull future fixes and changes back into older kernels where these fixes and changes are not expected which will cause failure in driver drivers such as this that expect a Mainline kernel.

You will notice in the README that a RHEL dev has taken his time to make modifications in this driver to allow some recent RHEL released to be supported by this driver. This is the only driver of the 6 I currently keep available where a RHEL dev has done this work. I simply do not have the time to do it as it would effectively double my workload. 

That you are looking for support on a VERY old kernel makes it extremely unlikely the support will happen as it would take a LOT of work.

With that said, I know a LOT about usb wifi adapters and am will to give advise to you if you are interested in finding a solution. The first bit of advise is to take a another look at what RHEL based distro you want to use. The following are already supported:

RHEL 8.4, 8.6, 8.8, 8.9, 9.1, 9.2 and 9.3

There is also the possible solution of using adapters with in-kernel drivers but it would take some research to see is available with the RHEL kernels.

@morrownr

jdemars-netscout, 2023-12-10:
Hi @morrownr,
Thank you.

blackwingcat, 2024-08-29:
I tried 8822BU on RHEL 8.9 compatible linux.
The device driver was succeeded to load on system and show its MAC address , but WiFi LED would not on.
Is the reason it would not work correctly , RHEL 8.9 kernel version is 4.18 and over range from 4.19- ?

misha4gps, 2024-08-29:
Hi @blackwingcat,
Can you check the LED functionality of 8822BU on any modern kernel ?

blackwingcat, 2024-08-30:
Sorry, it resolve myself.
I forgot install NetworkManager-wifi

yum install NetworkManager-wifi
reboot
nmcli d wifi
nmtui <-  set wifi password on activate

and It works on kernel 4.18.