morrownr/88x2bu-20210702 #50: Manual Install issue on Ubuntu 2022.04 with 5.15 Kernel installed on Raspberry PI + FIX

grandadmiralmcb, 2022-05-16:
The manual installation of this driver on ubuntu 2022.04 with the 5.15 kernel on a raspberry pi may have an error

if you run into an issue concerning -mabi options being incorrect then the solution is to specify the architecture during the make step.

So the new steps are as follows

make clean
make ARCH="arm64"
sudo make install
reboot.

The only change is to specify that arm64 is architecture.

morrownr, 2022-05-16:
Hi @grandadmiralmcb 

Thanks for the report but I am curious about something.

Did you follow the installation instructions in Step 9?

In your case, you should have run `./ARM64_RPI.sh` before selecting a compilation option in Step 10. If you did this then I will run a test on my RasPi4B and see what is going on.

Regards

grandadmiralmcb, 2022-05-16:
I followed this step, and everything ran properly as it should on that step.

morrownr, 2022-05-16:
I'll check this.

grandadmiralmcb, 2022-05-16:
Specifically, we installed a Ubuntu 2022.04 jammy on a raspberry pi 4, then installed this driver

morrownr, 2022-05-16:
Oh. That is a good detail to know as I was off to test with RasPiOS 2022.04.04. I'll have to dig out an unused sd card and burn Ubuntu which may take some time.

grandadmiralmcb, 2022-05-16:
Absolutely

morrownr, 2022-05-16:
Forgot to ask: Desktop, server or core?

grandadmiralmcb, 2022-05-16:
Desktop

morrownr, 2022-05-18:
Back with you. I was able to test this morning.

I have good news and bad news.

The bad news is I was not able to do duplicate the issue. The good news is that the installation worked.

I should have pulled out my ssd for this test. Wow, Ubuntu is slower than honey on a cold winter day on an sd card.

> make ARCH="arm64"

In looking at the issue, my best guess is that something went wrong when `./ARM64_RPI.sh` was run because it should cause Makefile to do what you say above.

When running that little script, remember capitalization has to be exact and don't use sudo. It will tell you what it did if it runs correctly.

If this continues to be a problem, please let me know. I am going to mark this issue as `cannot duplicate` for now.

Regards

grandadmiralmcb, 2022-05-18:
Thanks for your observations, if I come across any details in the future I'll check back here.  We made an image of our solution, and it's working on multiple pi's now so great work!

morrownr, 2022-05-18:
Glad it is working. I do a little consulting on the side. Sometimes for commercial products and sometimes just for fun things. Something I need to warn of is that I normally recommend Mediatek based adapters instead of Realtek for commercial products . Reason:

You or your users may find a need to upgrade... such as for security reasons. If the upgrade requires a kernel upgrade, you may find that this driver no longer works. This is a real problem with static, out-of-kernel drivers such as the ones Realtek provides. On the other hand, Mediatek's drivers are in-kernel so if a new kernel is required, the new Mediatek driver is already in that new kernel.

I started this site to help USB WiFi adapter users and I keep a lot of information available. Here is a link to the Main Menu:

https://github.com/morrownr/USB-WiFi

Regards