morrownr/88x2bu-20210702 #55: Driver installed successfully, but wifi is not connected

opiuman, 2022-06-22:
Hi

The driver is successfully installed into the Pi 4 with 64bit Pi OS. The onboard wifi is disabled via the `/boot/config.txt` file. After the reboot, the `wlan0` shows a list of available wifi endpoints. Once a home wifi is selected, the ESSID shows up correctly in `iwconfig` result. However, after a few seconds, it is back to `unassociated`.  I tried so many time on either 2.4G or 5G wifi. It never connects. Anyone knows what is going on there? 

PS: The onboard wifi connects without any issue.
Thanks,
Wilson

morrownr, 2022-06-22:
Hi

Please paste the information obtained with:

$ sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev

`iwconfig` is no longer supported, Recommend you use `iw` and `ip`.

Please post the contents of:

/etc/wpa_supplicant/wpa_supplicant.conf

Is your AP/WiFi router using WPA2 or WPA3 for security?

Are the 2.4 and 5 GHz networks on your AP/WiFi router using the same name?

Regards

morrownr, 2022-08-19:
Closing due to no reply.

closed 2022-08-19 by morrownr