morrownr/88x2bu-20210702 #56: Slower connection speed compared to machine under Windows

michal-devel, 2022-06-23:
Hi Nick (if I may),

first of all, many thanks for this driver. This is the best and most stable driver for my card. I have no problems, but the speed is slower in comparison with the same card under Windows. For my Internet connection the download speed is 200Mb at best while under Windows is almost maximum at 300Mb. It's not a big deal however I decided to report this.

I've tested different kernels (LTS and latest in Arch) and all the ports in my computer (HP MP9 G2 Retail System). My Windows and Linux machines are close to each other and the router is nearby (about 2m). I'm happy to provide some more details if you need some.

Cheers!

```
Linux arch 5.15.49-1-lts #1 SMP Wed, 22 Jun 2022 12:52:35 +0000 x86_64 GNU/Linux
SecureBoot disabled
Bus 002 Device 002: ID 0951:16d5 Kingston Technology DataTraveler Elite G2
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 005: ID 04a9:1912 Canon, Inc. LiDE 400
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
rtl88x2bu/5.13.1, 5.15.49-1-lts, x86_64: installed
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr b4:4b:d6:c4:66:5f
                ssid my_ssid
                type managed
                channel 44 (5220 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 15.00 dBm
```

morrownr, 2022-06-30:
Hi @michal-devel 

Sorry for the delay. I appear to have skipped over this issue.

Please run the following and post the results:

$ lsusb -t

michal-devel, 2022-06-30:
No problem at all. Please have a look the results:

```
$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 2: Dev 2, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 2: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 2: Dev 2, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 4: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
    |__ Port 5: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 4, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 13: Dev 5, If 1, Class=Vendor Specific Class, Driver=, 480M
    |__ Port 13: Dev 5, If 2, Class=Vendor Specific Class, Driver=, 480M
    |__ Port 13: Dev 5, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
```

morrownr, 2022-06-30:
> Port 4: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M

There is your answer. You are running in USB2 mode. If you were in USB3 mode, you would see 5000M.

Go to your driver directory and run:

$ sudo ./edit-options.sh

On the first `options` line you will see some parameters that you can chance. The one you want to change is:

rtw_switch_usb_mode=1

Cheers

michal-devel, 2022-07-01:
Thanks for this! I will be back in two days and check my USB 3 port.

michal-devel, 2022-07-06:
OK, I have my usb wifi card plugged in to USB 3 port. But the usb wifi is usb 2.0, not 3.0.

```
❯ lsusb -D /dev/bus/usb/001/003
Device: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
```

Thanks for your help!

morrownr, 2022-07-07:
Can you run:

$ sudo ./edit-options.sh

...and post the `options` line? It is the only line that is not commented out.

Regards

michal-devel, 2022-07-07:
Sure!

``options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_switch_usb_mode=0``

Btw, when I place `rtw_switch_usb_mode=1` in options then the wifi card is not picked up by the driver.

morrownr, 2022-07-07:
> `options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_switch_usb_mode=0`
> 
> Btw, when I place `rtw_switch_usb_mode=1` in options then the wifi card is not picked up by the driver.

I would expect this if you were using a USB2 port but you seem sure that the port is capable of USB3 so...

Try the following line:

options 88x2bu rtw_drv_log_level=4 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_switch_usb_mode=1

After booting, run:

$ sudo dmesg

... and see if you can find any lines related to this driver that might give us a hint.

Separate subject: Some BIOSs have settings related to USB ports. You might take a look and see if anything of interest shows up.

I remember back in the early days of the IBM PC and the only thing that mattered was whether a computer was fully compatible with the IBM PC. Now days companies kick out components that have all kinds of incompatibilities and nothing much is said. I don't know if that is the case here but it is not obvious to me what would be causing this. This is actually a very fast driver in managed mode. I don't use windows so I don't have a comparison but something is not right and it has to do with USB mode.

michal-devel, 2022-07-08:
Dear Nick,

thanks for your detailed reply.

HP MP9 G2 Retail System is my computer and according to specification has two USB 3.0 port in front: https://support.hp.com/us-en/document/c04899269#AbT9

My WiFi dongle is plugged in to one of these ports. When it comes to BIOS, it
 has indeed some options related to USB ports, however I don't see anything relevant - no option to change port speed etc.
I will restore BIOS options to the defaults and let you know whether it helped or not.

Perhaps logs from dmesg give some clue. Please find them attached. There are some errors and warnings.

[log.txt](https://github.com/morrownr/88x2bu-20210702/files/9074982/log.txt)

morrownr, 2022-07-09:
> There are some errors and warnings.

Thanks for the log. There is some previously unknown ugliness in there. I'll have to do some research and I need to do a fresh test installation... however, I am right in the middle of badly needed maintenance in my little lab. I usually use Manjaro to test what is going on in the Arch world but that installation is getting a little old. Right now I have to finish rebuilding my main dev box and then I can look at setting up to see what I can find. Sorry for the delay. If you haven't heard back from me in around 7 days, ping me.

Regards

michal-devel, 2022-07-10:
Sure, I'll ping you. Arch has recently got installer, it's more friendly than manual installation used to be. Still it is not as friendly as other installers.

I rested BIOS to its defaults to be sure that its settings doesn't affect USB ports. It didn't help thought.

michal-devel, 2022-07-17:
Dear Nick,

This is just a friendly reminder of the issue I reported.

morrownr, 2022-07-23:
Dear Michal,

I still don't have an answer for this issue. It seems that the cause of the slower speed is inability to go into USB mode. What is causing that is still unknown as I am not finding other issues exactly like this. I can keep my eyes open.

Regards

michal-devel, 2022-07-25:
Dear Nick,

Many thanks for your help.

Cheers!

guilletrejo, 2022-08-16:
Hi! I don't know if I have to open another issue or only comment here, so let me know and I'll create another issue if it's necessary.

I have the same problem. I am dual booting windows and I have 300mbps, but on Linux (Ubuntu 22.04, kernel 5.15), initially I only managed to reach 12mbps.

To make the driver work I had to set rtl_wireless_mode to 2.4GHz 802.11b/g, but now I tried to set it to 2.4GHz 802.11b/g/n and it still works, and it increased the speed to 70mbps.

I tried to set 5GHz but it doesn't pass the authentication. I have no problem on using 2.4 but the limited speed seems weird to me. Maybe is there another config I can try? I already changed the port to USB 3.0 but there are no improvements.

Any advice is welcome, and I thank you a lot for this driver, it is the only one that worked for me. You are awesome!

Below is some information:

```
$ lsusb -t | grep 88 
    |__ Port 2: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 5000M
```
```
$ cat /etc/modeprobe.d/88x2bu.conf
options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_switch_usb_mode=1 rtw_wireless_mode=11
```
```
$ sudo dkms status
amdgpu/5.16.9.22.20-1438747~22.04, 5.15.0-46-generic, x86_64: installed
rtl88x2bu/5.13.1, 5.15.0-46-generic, x86_64: installed
```
```
$ lsmod | grep 88x2bu
88x2bu               3420160  0
cfg80211              974848  1 88x2bu
```

morrownr, 2022-08-17:
Hi @guilletrejo 

Post the results of:

$ iw reg get

guilletrejo, 2022-08-17:
Hi, thanks for replying. This is the result:

```
iw reg get
global
country AR: DFS-FCC
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 17), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)

```

morrownr, 2022-08-17:
Okay. That looks good. 

Pensando...

>  rtw_wireless_mode=11

Is there a reason you added that as a module parameter?
What happens if you remove it and reboot?

FYI: I used to live in BA.

guilletrejo, 2022-08-17:
Nice! I live in Cordoba :) 

The reason was that 5GHz doesn't work, but now that you say it, I can remove it safely because I am automatically connecting to the 2.4GHz network. I removed it and it's exactly the same, the speed is 40mbps approx. (it was the same before removing the flag and rebooting).

morrownr, 2022-08-17:
Hmmm... generally when I see 2 GHz working and 5 GHz not, it is the country code not set but your country is set and correct. I guess it is time to start at the beginning. Can you post the results of:

$ sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev

Also, please confirm your distro and version.

I remember Cordoba. Nice city. I have fond memories of AR.

guilletrejo, 2022-08-17:
Ok here we go:
```
Linux gtrejo-ubuntu 5.15.0-46-generic #49-Ubuntu SMP Thu Aug 4 18:03:25 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
SecureBoot disabled
Platform is in Setup Mode
Bus 004 Device 004: ID 05e3:0620 Genesys Logic, Inc. USB3.1 Hub
Bus 004 Device 005: ID 17e9:6006 DisplayLink Dell Universal Dock D6000
Bus 004 Device 003: ID 05e3:0620 Genesys Logic, Inc. USB3.1 Hub
Bus 004 Device 002: ID 2357:0115 TP-Link Archer T4U ver.3
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 019: ID 046d:c092 Logitech, Inc. G203 LIGHTSYNC Gaming Mouse
Bus 003 Device 018: ID 0c45:7603 Microdia USB Keyboard
Bus 003 Device 016: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 017: ID 06c4:c411 Bizlink International Corp. D6000 Controller
Bus 003 Device 014: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 020: ID 046d:0a8f Logitech, Inc. H390 headset with microphone
Bus 003 Device 015: ID 1235:8006 Focusrite-Novation Focusrite Scarlett 2i2
Bus 003 Device 013: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 003 Device 012: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
amdgpu/5.16.9.22.20-1438747~22.04, 5.15.0-46-generic, x86_64: installed
rtl88x2bu/5.13.1, 5.15.0-46-generic, x86_64: installed
phy#0
        Interface wlx6c5ab0c922ed
                ifindex 4
                wdev 0x1
                addr 6c:5a:b0:c9:22:ed
                ssid Fibertel WiFi113
                type managed
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
                txpower 16.00 dBm
```

Distro is Ubuntu 22.04 LTS, and I'm using KDE, if that means anything.

morrownr, 2022-08-17:
I see one thing that might be an issue or it might not be:

> ID 05e3:0620 Genesys Logic, Inc. USB3.1 Hub

Is the adapter plugged into this hub? If so, try the adapter in a USB2 or USB3 port on the system.

> were you born here or just visited the country?

Neither, I worked in the Palermo district of BA for a few years.

guilletrejo, 2022-08-17:
I see, that's great.

The adapter is not connected to that hub. It is alone in another one, also in the back of the case.

morrownr, 2022-08-17:
I am scratching my head. I'm not sure what to try next but will keep an eye open.

morrownr, 2022-08-21:
@guilletrejo 

Still pondering this. Have you figured anything out?

ThiloteE, 2022-08-21:
1. `width: 20 MHz` is at the minimum value. I am not sure if this hardware supports higher widths e.g. 40, but if so, you could try that. This can increase speed, but can potentially decrease range.
2. lowering or increasing `tx power` can increase/decrease speeds. The phenomenon called "Interference" (e.g. see [SNR](https://documentation.meraki.com/MR/WiFi_Basics_and_Best_Practices/Signal-to-Noise_Ratio_(SNR)_and_Wireless_Signal_Strength)) unfortunately makes it impossible to come up with accurate predictions, if higher or rather lower values will lead to higher throughput and better latency. Keep in mind, there are regulatory limits, which depend on the country you live in.

guilletrejo, 2022-08-23:
Hi. @morrownr .

I couldn't try anything. I'm currently with 50Mbps according to fast.com. I can't think of anything to try to improve things.

I tried to swap channels (11, 10, 9, 8, 1) but it is the same.

@ThiloteE I don't know how to force the use of 40MHz for the channel width. I have set up in the router the option to use (20/40MHz) but there is no option to only set 40Mhz.

The same goes for the tx power, I have no idea how to tweak that.

If you tell me how to force 40Mhz I can try to see if it improves anything.

Thanks!

ThiloteE, 2022-08-23:
Sorry, I don't know, but once you find out, It would be wise to follow upper limits as defined in the [Linux wireless-regulatory database](https://wireless.wiki.kernel.org/en/developers/regulatory/wireless-regdb) for your country. If you go over the limit, you could break devices in a wide area around you. There usually are high fines for this illegal activity. Better to go lower. There could be great benefits if you live in a congested area: [Example](https://forum.openwrt.org/t/netgear-wax202-wifi-6-30-at-amazon/130029/106). 40MHz is probably not advised if you live in a neighborhood with high congestion. Most efficient would be for you and all your neighbours to talk to each other and limit yourself to 20MHz so as to avoid noise. Cities are hell. You can compare it to a room full of people shouting. Nobody understands each other, because everybody is shouting. Now, if people were to just come to an understanding with each other to "talk quietly", people would be able to understand each other more clearly, as their noise will not reach the other side of the room. Of course, it is not that easy. Some kind of tx power is still required to be heard...

ThiloteE, 2022-08-23:
```
iw reg get
global
country AR: DFS-FCC
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
```

This would suggest the upper limit for width would be 40 MHz in 2.4G band
and probably (I don't know?) 20 dbm for tx power, right?

guilletrejo, 2022-08-23:
I understand, thanks for the explanation. Nevertheless, there's nothing I can do about it because I don't know how to change the MHz anyway, so I will just leave the router config in 20/40 MHz and automatic channel.

Regarding the connection speed for linux, I'll keep thinking of things to try to improve speed. (I'm seriously considering of buying a PCI wifi card lol, but in my city I only find TP-link and I don't know if that's a good brand).

augreed, 2023-03-19:
I just bought two comfast NICs, one with the rtl8821cu chip and the other with the rtl8812bu chip. the 8821cu one works fine at 5GHz, but the 8812bu one has very slow transfer speeds at 5GHz. Both cards use your driver.

ThiloteE, 2023-03-19:
@augreed I would suggest to open a separate issue. If it is relevant here, it can be linked to this issue. To be honest, @guilletrejo fell into the same trap :D
Michal-devel has to endure a lot :D

ThiloteE, 2023-03-19:
> I tried to set 5GHz but it doesn't pass the authentication.

@guilletrejo I completely missed that.

1. maybe if you activate different log level, it mentions WHY authentication was not passed?
2. Depending on the channels used by the device you are trying to connect to, it could be that you tried to connect to [DFS channels](https://en.wikipedia.org/wiki/Dynamic_frequency_selection). The default option of this driver is to disable DFS channels. So, you could either try to activate DFS channels in the `88x2bu.conf` file, or alternatively to switch to a non-DFS channel on your main router/AP.
      
      ```
      # DFS Options ( rtw_dfs_region_domain )
      #
      # 0 = NONE (default)
      # 1 = FCC
      # 2 = MKK
      # 3 = ETSI
      #
      # Notes:
      # - Activates DFS channels in AP mode.
      # - DFS FCC 80 MHz channels for hostapd: 52(58), 100(106), 116(122) and 132(138)
      # - For more information: https://en.wikipedia.org/wiki/List_of_WLAN_channels
      #
      # Note: An AP needs to listen on a DFS channel for a period of 60 seconds
      # before transmitting on the channel. If any radar pulses are detected,
      # the AP cannot use that channel and will have to try a different channel.
      ```

augreed, 2023-03-20:
> @augreed I would suggest to open a separate issue. If it is relevant here, it can be linked to this issue. To be honest, @guilletrejo fell into the same trap :D Michal-devel has to endure a lot :D

Aren't we talking about the 88x2bu being very slow at 5G frequencies? I don't know how different the 8821cu's driver is from the 88x2bu's driver, but the 8821cu works fine. Just an additional note.

morrownr, 2023-03-20:
> The default option of this driver is to disable DFS channels.

This is true but only for AP mode.

> Aren't we talking about the 88x2bu being very slow at 5G frequencies?

Are you in USB3 mode?

$ lsusb -t

augreed, 2023-03-20:
> 

> Are you in USB3 mode?
> 
> $ lsusb -t

I've tried both modes (rtw_switch_usb_mode=0/1), and in either mode, the speed measurement is an order of magnitude lower as long as it's running at 5G. But I have returned the 8812bu so there is no way to continue testing.

I am going to buy the USB network adapter that uses the mt7612u chip.

Thanks.