morrownr/88x2bu-20210702 #57: Extremely slow internet connection on Kernel 5.18.*

DashCruft, 2022-06-26:
Ever since the kernel 5.18 has been released, I've been in a misery of trying to fix my Wifi adapter. For some specific reason, the internet speed drops to 20mb per second from 300mb per second. Therefore I have reverted back to 5.17 kernel for the meantime.

I personally think this issue exists due to the fact that linux 5.18 introduced new wifi adapters to the kernel, making the latest drivers for 88x2bu to be broken possibly?

Regards

DashCruft, 2022-06-26:
**New Update**

Actually, I've noticed that rtl88x2bu dkms module doesn't get installed during the update process to kernel 5.18. The install gets stuck when trying to do `dkms install --no-depmod rtl88x2bu/5.8.7.1 -k 5.18.7-zen1-1-zen`

morrownr, 2022-06-26:
Hi @DashCruft 

Hmmm...

> dkms install --no-depmod rtl88x2bu/5.8.7.1 -k 5.18.7-zen1-1-zen

No wonder it gets stuck. v5.8.7.1 is an old version of the 88x2bu driver that does not exist here anymore. If that version is on your system, you need to uninstall the driver, delete the driver directory and start with a fresh download:

$ ./remove-driver.sh

Delete the driver directory however you see fit. Reboot and start the installation from the beginning with the current version here which is 5.13.1.

I have no way to update drivers that are sitting on your harddrive so anytime you decide to move up to a new kernel version, you need to check here to see what level of kernel is supported and if you are covered. There is no way that old driver is going to compile on 5.18.

For what is is worth, since you use a usb wifi adapter and might be interested, I have a Main Menu that provides a lot of information that might be of interest to you:

https://github.com/morrownr/USB-WiFi

Regards

DashCruft, 2022-06-27:
Thank you very much for a very in depth response! I have reviewed your [USB-WiFi](https://github.com/morrownr/USB-WiFi) Main menu and I gotta say, very great detail and information in there. Things have became more clarified for me.

And for the driver, I completely removed any DKMS module regarding 88x2bu. Re installed the whole thing but now a very weird problem occurs. 
[image]

Basically, once the adapter connects to the network, it works fine but then after a few seconds, it completely shuts down. The speed drops to 0 and then eventually it disconnects. This issue happens every single time on startup and when I restart NetworkManager.

DashCruft, 2022-06-27:
Output of `sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev`
```
Linux Porya-PC 5.18.7-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Sat, 25 Jun 2022 20:22:03 +0000 x86_64 GNU/Linux
SecureBoot disabled
Platform is in Setup Mode
Bus 004 Device 002: ID 21c4:0cd1 Lexar USB Flash Drive
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 0c76:161e JMTek, LLC. USB PnP Audio Device
Bus 003 Device 003: ID 0c45:5004 Microdia Redragon Mitra RGB Keyboard
Bus 003 Device 002: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 05ac:1392 Apple, Inc. Apple Watch charger
Bus 001 Device 007: ID 1b1c:0c1a Corsair CORSAIR Lighting Node CORE
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 004: ID 0b05:1939 ASUSTek Computer, Inc. AURA LED Controller
Bus 001 Device 003: ID 0b05:17cb ASUSTek Computer, Inc. Broadcom BCM20702A0 Bluetooth
Bus 001 Device 002: ID 258a:2022 Glorious Model O Wireless
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
broadcom-wl/6.30.223.271, 5.18.7-arch1-1, x86_64: installed
broadcom-wl/6.30.223.271, 5.18.7-zen1-1-zen, x86_64: installed
rtl88x2bu/5.13.1, 5.18.7-zen1-1-zen, x86_64: installed
phy#0
	Interface wlp11s0f3u2
		ifindex 3
		wdev 0x1
		addr 1c:bf:ce:1c:01:4d
		type managed
		txpower 30.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

DashCruft, 2022-06-27:
@morrownr Sorry to ping but... I have finally found a fix which you might want to add to the README.md

Basically, Starting from Linux 5.18, some distributions have added experimental RTW88 USB support. It is not yet stable but if it works well on the system, then you will no longer need this driver. However if it doesn't work or is unstable, you need to manually blacklist it due to the fact that it has a higher loading priority than this external driver we are loading.

running `echo "blacklist rtw88_8822bu" > /etc/modprobe.d/rtw8822bu.conf` with a following reboot after, fixes everything!

morrownr, 2022-06-29:
That is a surprise. I monitor the linux-wireless list so was aware that work to include usb support was underway but had no idea that any distros were jumping the gun.

I'm going to borrow your statement for the README if that is okay with you.

When able, give us a report about the in-kernel driver.

DashCruft, 2022-06-29:
> I'm going to borrow your statement for the README if that is okay with you.

Absolutely fine with me. Go ahead.

> When able, give us a report about the in-kernel driver.

In the current kernel that I have, the path of the `rtw88` drivers by default are located at:
```
/lib/modules/5.18.7-zen1-1-zen/drivers/net/wireless/realtek/rtw88 
```

The contents includes many drivers including `rtw88_8822bu`
[image]

The content of `rtw88_8822bu.ko` includes a single file called `rtw88_8822bu.ko`

morrownr, 2022-07-03:
Hey @DashCruft 

Can you give us a performance report? How is that in-kernel driver performing?

DashCruft, 2022-07-03:
On both of my 88x2bu bluetooth adapters, it performs awfully... as explained, it keeps disconnecting after a short due ration and it's extremely unstable. The internet speed is also drastically low compared to your driver.

morrownr, 2022-07-03:
That stinks. I hope it improves.

morrownr, 2022-08-21:
@DashCruft 

I was pondering whether I should add code to the installer to automatically blacklist the in-kernel driver. The removal script could unblacklist when run. What do you think?

DashCruft, 2022-08-21:
That sounds pretty good. I think that should happen because all the future kernels until further notice are going to include this driver. Only kernel 5.17 or lower will not have this issue.

morrownr, 2022-08-21:
> That sounds pretty good.

Okay. I'll work on a patch. Should not take long. Are you willing to test?

What distro are you using that patches the driver in? I ask because I'd like to test also. This is a very heavily used driver so when I make a mistake, issues lights up.

>  I'll check once in a while to state the current status of the in kernel driver if you'd like.

Yes, please do. I have my hands full with everything at this site. You probably know but here is the site Main Menu:

https://github.com/morrownr/USB-WiFi

morrownr, 2022-08-21:
Patch has been applied. Let me know if you see any problems.

I patched the dkms install and remove scripts as well as the non-dkms install and remove scripts. I hope I did not make any errirs but they need testing.

DashCruft, 2022-08-21:
> Okay. I'll work on a patch. Should not take long. Are you willing to test?

For sure. I'll def test it.

> What distro are you using that patches the driver in? 

I use Arch Linux and as of now, I'm using the `5.19.2-zen1-1-zen` kernel.

DashCruft, 2022-08-21:
Just tested the new `.sh` scripts. First I made sure I don't have it blacklisted in `/etc/modprobe.d/rtw8822bu.conf` and tried out the `uninstall-driver.sh` script first and then `install-driver.sh` script. Upon a restart, everything seems to be working well!

Also a quick update for the in kernel driver currently: it's very unstable as expected. It still has the bugs mentioned in the past. Maybe in Kernel `6.0` we might see an improvement. Until then, that is all.

morrownr, 2022-08-21:
> Upon a restart, everything seems to be working well!

Thanks for testing. Hopefully this will save a lot of headaches.

Now I need to repeat the process with the 8821cu driver I maintain as it should be supported with the same rtw88 driver.

> Also a quick update for the in kernel driver currently: it's very unstable as expected. It still has the bugs mentioned in the past. Maybe in Kernel 6.0 we might see an improvement.

I haven't seen anything coming through linux-wireless lately regarding this subject so I don't know if a review is going on... or what.

firxworx, 2024-01-22:
I found this issue's discussion helpful when troubleshooting performance issue and will share my resolution... which was to bump my kernel from 5.15 to 6.5 and use the latest in-kernel drivers instead.

While earlier in-kernel realtek drivers might suck, I am happy to report the drivers that ship with kernel 6.5+ resolved my issues.

**System & Hardware**

6th gen Intel i5 running 5.15.0-91-generic kernel on Ubuntu 22.04 LTS.

USB wifi adapter is a generic Realtek: [Newfast 1300mbps adapter](https://www.amazon.ca/dp/B0CL9CQ8SY) which has supported device ID `0bda:b812`.

**Performance Issue w/ 88x2bu Driver**

The install script for 88x2bu worked great. I really appreciate the effort put into how smooth it was!
My adapter started working and I was able to join a wireless network with zero issues.

However performance was a major issue: the speed was _extremely slow_ to the point of being barely usable for any practical purposes. The adapter seemed to work as intended for a short span of time and then slow to a crawl.

My troubleshooting steps included activating USB 3 from USB 2 in the config (`/etc/modprobe.d/88x2bu.conf`) since it wasn't enabled by default (I was a bit surprised to see that).

Nothing I tried seemed to resolve the speed issue. 

**v6.5 Kernel Drivers**

Ubuntu very recently released the 6.5 kernel as an official apt package for Ubuntu 22.04 LTS: 
`linux-image-generic-hwe-22.04`

The in-kernel drivers for Realtek include 8822BU: https://www.kernelconfig.io/config_rtw88_8822bu.

I'm happy to report these drivers work great. Whatever made the earlier versions in the 5.x kernel terrible seems to have been fixed. 

Even features like the MATE "Netspeed" desktop GUI widget which did not work with 88x2bu (no readings) work great with the rtw88_8822bu drivers.

**Instructions for Ubuntu 22.04 LTS: Upgrade Kernel & Run In-Kernel Drivers**

First make sure you aren't running anything that's important to you that isn't supported with the 6.5 kernel.

- `apt update && apt upgrade` (reboot if necessary)
- `apt install linux-image-generic-hwe-22.04`
- edit `/etc/modprobe.d/88x2bu.conf` and comment out the line `blacklist rtw88_8822bu` which disables the in-kernel driver
- reboot

The in-kernel driver should take priority over 88x2bu.

Confirm that the new drivers are in play by inspecting `lsmod` output. Here's mine:

```
lsmod | grep rtw
rtw88_8822bu           12288  0
rtw88_usb              24576  1 rtw88_8822bu
rtw88_8822b           229376  1 rtw88_8822bu
rtw88_core            356352  2 rtw88_usb,rtw88_8822b
mac80211             1720320  2 rtw88_core,rtw88_usb
cfg80211             1314816  3 88x2bu,rtw88_core,mac80211
```

For reference here's the output with the 5.15 kernel and 88x2bu from before I performed the above steps:

```
lsmod | grep 88
88x2bu               3321856  0
cfg80211              974848  1 88x2bu
```

If you're happy with the results you may want to run the uninstall script shipped by this project.

If you want to go back to these drivers simply uncomment the blacklist line in `88x2bu.conf` and reboot.

morrownr, 2024-01-22:
Hi @firxworx 

> I am happy to report the drivers that ship with kernel 6.5+ resolved my issues.

This is good to hear.

> The install script for 88x2bu worked great. I really appreciate the effort put into how smooth it was!

You are welcome.

> However performance was a major issue: the speed was extremely slow to the point of being barely usable for any practical purposes. The adapter seemed to work as intended for a short span of time and then slow to a crawl.

The driver here in this repo is faster than the in-kernel driver in 6.5 per my testing. There is a problem somewhere in your installation if that is not what you are seeing. I actually prefer users to switch to the in-kernel driver if it meets their needs as I would like to shut this repo down at some point over the next year so I can concentrate on other work.

> Whatever made the earlier versions in the 5.x kernel terrible seems to have been fixed.

The USB support for RTW88 was first introduced in kernel 6.2.

> edit /etc/modprobe.d/88x2bu.conf and comment out the line blacklist rtw88_8822bu which disables the in-kernel driver

No please don't do it that way. The `remove-driver.sh` script will delete the `88x2bu.conf` file as it is no longer necessary if you want to move to the in-kernel driver... and just as important, `remove-driver.sh` will uninstall the driver so that it does not conflict with the in-kernel driver.

> If you want to go back to these drivers simply uncomment the blacklist line in 88x2bu.conf and reboot.

No. That does not work. If you do that way then you are going to have both drivers trying to load. Please use `install-driver.sh` and `remove-driver.sh`. Let the scripts do the work.

Regards

@morrownr

p0ryae, 2024-01-23:
> The driver here in this repo is faster than the in-kernel driver in 6.5 per my testing. There is a problem somewhere in your installation if that is not what you are seeing. I actually prefer users [...]

I just want to add that I can confirm this. While the in-kernel driver operates perfectly fine and stable, the speed is definitely unmatched. The driver in this repo is x4 faster in my case when compared to in-kernel driver.