morrownr/88x2bu-20210702 #58: After reboot I need to always switch wifi usb adapter from usb 1 to usb 2

Predrag, 2022-07-02:
Hello, 

I have ASUSTek Computer, Inc. ASUS USB-AC58 USB Wireless adapter. I have successfully installed driver on Fedora 36 with kernel 5.18.7-200.fc36.x86_64. Everything works as expected I have internet connection, signal is great but every time when I reboot or turn on my PC I have to switch from usb 1 to usb 2 (after another reboot from usb 2 to usb 1). Because if I dont do this ritual I am connected to my wifi network but I dont have a internet connection. When I switch adapter from one usb port to another one after that I have internet connection. I have tried to install and reinstall driver many times but it wont help.

morrownr, 2022-07-03:
Hi @Predrag 

Interesting. The 5.18 kernel gets my attention. It recently came to my attention that some distros starting with kernel 5.18 have started including the usb support for the rtw88 driver. A little background:

https://github.com/morrownr/USB-WiFi

Go to `Issues` and read issue 77.

I don't know that this issue is the case here but we can test. Try the following with the adapter in a usb port:

$ sudo ./remove-driver.sh
$ sudo reboot

Post the results of the following:

$ lsmod
$ iw dev

Predrag, 2022-07-03:
Hello,

I am sending you in attachment results from lsmod the iw dev returned nothing.

[lsmod.txt](https://github.com/morrownr/88x2bu-20210702/files/9034238/lsmod.txt)
[iw_dev.txt](https://github.com/morrownr/88x2bu-20210702/files/9034239/iw_dev.txt)

morrownr, 2022-07-05:
That was not the problem. 

Do you also have windows on the computer?

Predrag, 2022-07-05:
Yes I do

morrownr, 2022-07-05:
I don't run windows so this may be worth what you are paying for it...

It seems like I have read about similar problems on dual boot (windows plus linux) if Fastboot is active. Where you turn off Fastboot is information I may get wrong here but there should be a setting in the bios or in windows or both. Maybe you can search for how to turn Fastboot off.

If you have Secure Mode on, you might try turning it off to see if that is an issue.

Only do one at a time so you know if one of the options helps.

FYI: My main dev box runs kernel 5.19 rc4. I loaded it a few days ago to check on kernel 5.19. I'm still working on a few things but it is mostly cosmetic things like the docs and flags to keep compilers happy as I have not had to make any code changes for 5.19. I did have to make a couple of minor code changes for 5.18 but both it and 5.19 are working well in my testing here.

My main dev box runs Kubuntu 22.04. I also test on Ubuntu, Kali, RasPiOS and Manjaro. I've been meaning to install Fedora but never seem to get the time.

Predrag, 2022-07-25:
You are right, if you are dual boot this wifi is not working correctly. If you have only linux installed everything works as expected.

closed 2024-04-10 by Predrag