morrownr/88x2bu-20210702 #59: I have to enter the password again on every system startup, otherwise it won't connect

ibrag474, 2022-07-15:
I use Gembird WNP-UA1300-02, it's a RTL8812BU chipset USB 3.0 wifi adapter. On every startup the connection to wifi network takes up to 10 minutes and fails at the end, so I have to forget the network, enter the password again and then it connects in a few seconds.
Here is my 88x2bu.conf:
`options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_switch_usb_mode=0`
I set rtw_power_mgn to 0, it improved connection stability and now I am able to just unplug the adapter and plug it to another usb port, with power saving adapter was not starting up after replug to different port.
Info about my system Linux 5.13.0-40-generic #45~20.04.1-Ubuntu SMP Mon Apr 4 09:38:31 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

morrownr, 2022-07-15:
Please post the results of:

sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev

ibrag474, 2022-07-20:
`sudo uname -a` 

> Linux ZEN-PC 5.15.0-41-generic #44~20.04.1-Ubuntu SMP Fri Jun 24 13:27:29 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

`mokutil --sb-state`

> SecureBoot disabled
> Platform is in Setup Mode

`lsusb`

> Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 003 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
> Bus 003 Device 002: ID 18d1:4ee7 Google Inc. Nexus 5X
> Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 001 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
> Bus 001 Device 003: ID 1b1c:1b20 Corsair Corsair STRAFE RGB Gaming Keyboard
> Bus 001 Device 002: ID 0bda:b812 Realtek Semiconductor Corp. USB3.0 802.11ac 1200M Adapter
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

`rfkill list all`
> 0: hci0: Bluetooth
> 	Soft blocked: no
> 	Hard blocked: no
> 1: phy0: Wireless LAN
> 	Soft blocked: no
> 	Hard blocked: no

`dkms status`
> nvidia, 510.73.05, 5.15.0-41-generic, x86_64: installed
> rtl88x2bu, 5.13.1, 5.15.0-41-generic, x86_64: installed

`iw dev`
> phy#0
> 	Interface wlx1cbfce22ce98
> 		ifindex 4
> 		wdev 0x1
> 		addr 1c:bf:ce:22:ce:98
> 		ssid NETGEAR23
> 		type managed
> 		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
> 		txpower 16.00 dBm

morrownr, 2022-07-20:
@ibrag474 

I've been doing a little research. What I am seeing is various issues that seem to correlate with the following situation:

Upgrading Ubuntu 20.04 to kernel 5.15. Your Linux Mint is based on Ubuntu 20.04. Ubuntu 20.04 shipped with kernel 5.4. Ubuntu 20.04 is a LTS version but when upgrading with HWE or mainline kernels that are much newer, some things may not work. Why is this? A vast amount of work goes into stabilizing LTS releases. It is impossible to fully test newer kernels with a LTS base and the Linux kernel is under constant development.

I test LTS releases of several distros. I test Ubuntu 20.04  with kernel 5.4 and I test Ubuntu 22.04 with kernel 5.15 as they are the stable releases. This driver works well with those releases. I do not have the capability to test on the what amounts to the carious hybrid configuration.

From my perspective, there are two options that lead to very stable function of usb wifi adapters in Linux:

Option 1: Stick with the kernel that the LTS release used. It is fun to upgrade kernels but unless there is a need to upgrade, don't. If you want to upgrade to kernel 5.15, wait until the new release of Mint happens and upgrade the entire release as it will be based on Ubuntu 22.04 and will have kernel 5.15.

Option 2: Use an adapter with an in-kernel driver. The drivers for these adapter are updated by the kernel devs and stay current with kernel updates and the changes come into your system when a new kernel is installed. I maintain a lot of information and links to adapters that use in-kernel drivers. The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

ibrag474, 2022-07-21:
Downgrading to kernel 5.4 and reinstalling the driver did not solved the problem

morrownr, 2022-07-22:
I took the time to carefully reread this issue from the beginning. Your last message seems to provide an additional hint. If it were me having the same issues, I would start by taking a look at my wifi router settings. The README has a section called...

Recommended WiFi Router/ Access Point Settings

Give it a look and see if it helps.

closed 2022-10-09 by morrownr