morrownr/88x2bu-20210702 #61: error when install

provinace, 2022-07-16:
Error:
```
Running install-driver.sh version 20220705
5.18.0-kali5-amd64
x86_64
Starting installation...
Copying source files to: /usr/src/rtl88x2bu-5.13.1
Copying 88x2bu.conf to: /etc/modprobe.d
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1

Building module:
cleaning build area...
'make' -j4 KVER=5.18.0-kali5-amd64 KSRC=/lib/modules/5.18.0-kali5-amd64/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.18.0-kali5-amd64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

Result for: sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev

```
┌──(kali㉿kali)-[~/src/88x2bu-20210702]
└─$ sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
Linux kali 5.18.0-kali5-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.5-1kali5 (2022-07-04) x86_64 GNU/Linux
EFI variables are not supported on this system
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 0e0f:0008 VMware, Inc. Virtual Bluetooth Adapter
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
0: hci0: Bluetooth
        Soft blocked: yes
        Hard blocked: no
rtl88x2bu/5.13.1: added
```

morrownr, 2022-07-16:
Hi @provinace 

I'm pretty sure this issue is due to VMware. I don't recall what the issue is but will take a look around. I probably need to put the solution in the FAQ.

Are you locked into using VMware or would you be willing to give VirtualBox a try? The reason I ask is that I do have experience with VirtualBox.

Regards

provinace, 2022-07-16:
I think I'll stick to VMware

closed 2022-08-19 by morrownr