morrownr/88x2bu-20210702 #63: dkms issue?

kanatos92, 2022-07-19:
First of all, thank you for maintaining this repo. It is the only driver that works for me. Without it, I would have to bear with a long cable running across the entire house.

Short story, `install-driver.sh` succeeds but the usb adapter won't work (blinks for a few seconds then off). On the other hand, `install-driver-no-dkms.sh` works.

Here's the longer story. I have an old Dell precision T1500, it is running Ubuntu 20.04. I've been using the 88x2bu-5.8.7.4 driver for the past year and it worked fine until about two weeks ago when suddenly the system cannot recognize the usb wifi adapter. I removed the driver and tried to `install-driver.sh` again, but make failed with `fatal error: net/ipx.h: No such file or directory` (due to changes in kernel 5.15?). Then I found this new repo 88x2bu-20210702. The `install-driver.sh` script runs successfully, dkms confirms that the driver is installed. The usb adapter blinks for a couple of seconds, giving a list of available wifi networks, then the light turns off. Connecting to the networks failed with the message `activation of network connection failed` (light on usb adapter never turns on again). After multiple unsuccessful attempts with different combinations of options, I decided to give `install-driver-no-dkms.sh` a try, and voila, it works! I have no clue where (or which log) to look for potential causes of the issue but thought posting it here may be helpful.

More details:
Desktop: Dell precision T1500, only has usb 2.0. The usb adapter says usb 3.0 on it but it has been working fine with a decent speed (60/10 Mbps comparing to 75/11 Mbps on my tablet which is on the same network).
uname: `5.15.0-41-generic #44~20.04.1-Ubuntu SMP Fri Jun 24 13:27:29 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
(I don't have mokutil...)
lsusb: `Bus 001 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. `
rfkill: `0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no`
(dkms only has driver for nvidia graphics card, when using `install-driver.sh`, this driver does show up.)

morrownr, 2022-07-19:
Hi @kanatos92 

My first guess is that you have two drivers trying to run the adapter.

> it worked fine until about two weeks ago when suddenly the system cannot recognize the usb wifi adapter.

You upgraded to kernel 5.15 at the same time as the adapter stopped working?

Please the results of:

$ lsusb

$ dkms status

morrownr, 2022-07-19:
Opps...

I meant `lsmod` instead of `lsusb`.

kanatos92, 2022-07-19:
Hi @morrownr 
I was traveling and returned a couple days back when I noticed wifi stopped working. According to the upgrade log, the kernel was installed through an unattended-upgrade on 7/14. I assume it caused the adapter issue.

Following is the current output of dkms. 88x2bu would also show up if `install-driver.sh` is used. The script succeeds in both kernels 5.13 and 5.15 (separately, with `remove-driver.sh` in between), but the adapter doesn't work in either.
```
nvidia, 470.129.06, 5.13.0-52-generic, x86_64: installed
nvidia, 470.129.06, 5.15.0-41-generic, x86_64: installed
```
and lsmod
```
Module                  Size  Used by
nls_iso8859_1          16384  1
snd_hda_codec_realtek   147456  1
snd_hda_codec_generic    98304  1 snd_hda_codec_realtek
ledtrig_audio          16384  1 snd_hda_codec_generic
snd_hda_codec_hdmi     73728  1
snd_hda_intel          53248  6
snd_intel_dspcfg       28672  1 snd_hda_intel
snd_intel_sdw_acpi     20480  1 snd_intel_dspcfg
snd_hda_codec         155648  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
snd_hda_core          106496  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
88x2bu               3391488  0
intel_powerclamp       20480  0
dcdbas                 20480  0
snd_hwdep              16384  1 snd_hda_codec
snd_pcm               135168  4 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_core
coretemp               24576  0
dell_smm_hwmon         24576  0
nvidia_uvm           1171456  0
snd_seq_midi           20480  0
cfg80211              954368  1 88x2bu
snd_seq_midi_event     16384  1 snd_seq_midi
kvm_intel             372736  0
kvm                   991232  1 kvm_intel
snd_rawmidi            45056  1 snd_seq_midi
joydev                 32768  0
snd_seq                73728  2 snd_seq_midi,snd_seq_midi_event
nvidia_drm             65536  8
input_leds             16384  0
nvidia_modeset       1200128  11 nvidia_drm
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
intel_cstate           20480  0
nvidia              35352576  439 nvidia_uvm,nvidia_modeset
mei_me                 40960  0
snd_timer              40960  2 snd_seq,snd_pcm
drm_kms_helper        303104  1 nvidia_drm
snd                   102400  23 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_pcm,snd_rawmidi
cec                    61440  1 drm_kms_helper
mei                   135168  1 mei_me
rc_core                61440  1 cec
soundcore              16384  1 snd
i7core_edac            36864  0
fb_sys_fops            16384  1 drm_kms_helper
syscopyarea            16384  1 drm_kms_helper
sysfillrect            20480  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
serio_raw              20480  0
mac_hid                16384  0
sch_fq_codel           20480  6
ipmi_devintf           20480  0
ipmi_msghandler       122880  1 ipmi_devintf
msr                    16384  0
parport_pc             53248  0
ppdev                  24576  0
lp                     28672  0
parport                69632  3 parport_pc,lp,ppdev
drm                   602112  12 drm_kms_helper,nvidia,nvidia_drm
ip_tables              32768  0
x_tables               53248  1 ip_tables
autofs4                49152  2
hid_generic            16384  0
usbhid                 65536  0
hid                   147456  2 usbhid,hid_generic
gpio_ich               16384  0
broadcom               24576  1
bcm_phy_lib            24576  1 broadcom
psmouse               172032  0
i2c_i801               36864  0
i2c_smbus              20480  1 i2c_i801
ahci                   45056  2
lpc_ich                28672  0
tg3                   188416  0
libahci                40960  1 ahci
```

morrownr, 2022-07-19:
In the driver directory, please post the contents of:

dkms.conf

kanatos92, 2022-07-19:
Maybe I misunderstood but I didn't modify dkms.conf, it is the same as in the current repo head
```
PACKAGE_NAME="rtl88x2bu"
PACKAGE_VERSION="5.13.1"
MAKE[0]="'make' -j$(nproc) KVER=${kernelver} KSRC=/lib/modules/${kernelver}/build"
CLEAN="'make' clean"
BUILT_MODULE_NAME[0]="88x2bu"
DEST_MODULE_LOCATION[0]="/kernel/drivers/net/wireless"
AUTOINSTALL="yes"

```

morrownr, 2022-07-19:
> Maybe I misunderstood but I didn't modify dkms.conf

I did modify it a recently and this was my way of seeing if you have a current version of the driver. This driver has to have regular updates so I have to check if users have a very recent download. This causes problems at times as some users think of this driver in the way drivers are done for windows and it doesn't work that way with Linux.

Let me point something out: An autoinstallation that includes bumping the kernel from 5.13 to 5.15 is not how I would handle things. In the docs I recommend running the removal script prior to installing a new version kernel. 5.13.45 to 5.13.56 is okay. 5.13 to 5.15 is not okay. Realtek does not do us any favors with how these out-of-kernel are provided. 

Back on topic: This is a very heavily used driver and I am not seeing other reports similar to this so it must be something that is unique or nearly unique to your system. We are pretty sure it has something to do with dkms. I did a little searching and nothing jumped out at me.

Let me throw some options at you:

Option 1: go over the installation steps to make sure we are where we should be and try again

$ sudo apt update && sudo apt upgrade
$ sudo apt install -y build-essential dkms git
$ cd ~/src/88x2bu-20210702
$ git pull

Run the non-dkms removal script
$ sudo ./remove-driver-no-dkms.sh

$ sudo ./install-driver.sh
Post all of the screen output so I can see the error

Option 2: run with the non-dkms script since it works on your setup

Option 3: upgrade Ubuntu to 22.04

Option 4:  go to:

https://github.com/morrownr/USB-WiFi

That is the Main Menu for the site here. Menu items 1 and 2 provide some information (1) and links (2) to adapters that use in-kernel drivers. I throw this out because Linux users usually find that adapters that use the in-kernel drivers are most reliable. We are now seeing adapters using the in-kernel mt7921u driver shipping. There should be many more available in the coming weeks and months. These new adapters are WiFi 6.

kanatos92, 2022-07-20:
Thanks for the suggestions. I've tried option 1 multiple times, the install script finishes without any error but the adapter refuses to work. I'll stick with option 2 for now and consider option 3&4 in the future if other issues pop out.

closed 2022-07-20 by kanatos92

morrownr, 2022-07-20:
Option 2 is likely your best option for now. Here is the deal from my perspective. If your base install is Ubuntu 20.04 and you are running a kernel more modern than 5.4, then you are probably installing a HWE kernel. That is fine if everything works but you have to understand that you are running a base that was tested with specific parts and if you install a mainline or HWE kernel that is a lot newer than the kernel than the distro came with, a lot of things can stop working right. Ubuntu 20.04 is supported for many years but if the support is important in an installation, those wanting the support know that what is supported in 20.04 is kernel 5.4, not the mainlines and HWEs.

I test and support all 5 of the Realtek drivers I work on. I test on many supported distros. The supported version and the version I test when it comes to Ubuntu 20.04 is kernel 5.4. I cannot test on everything as I would need several full time employees to even get started on it. Kernel 5.15 is the release kernel in Ubuntu 22.04. Ubuntu 22.04 has a seriously newer version of several components such as gcc and dkms.than does Ubuntu 20.04. Your best bet, if you want to not run into problems such as this, is to use an adapter that uses in-kernels drivers.

Good luck

kanatos92, 2022-07-20:
Totally agree. Thanks for your patient explanations.