morrownr/88x2bu-20210702 #64: 88x2bu & slackware64

BusinessKid, 2022-07-19:
Hi Nick,
Slackware's main site is slackware.com, and there are many mirrors. Slackware.uk is a fast one, and Alien Bob hangs his stuff on slackware.nl. Slackware packages are everything installed by 'make install' - programs, libs, shares, docs & includes. There's no dependency tracking, but there is ldd:). You can upgrade, downgrade, install non_standard, etc. I have a /tmp/pack directory always. The packages are made by 'make DESTDIR=/tmp/pack install; cd /tmp/pack; makepkg ../my_package.txz; rm -rf /tmp/pack/*
Slackware has no systemd, uses /lib for 32bit and /lib64 for 64bit libs, prefers old stuff over new and expects you to know a little. Scripts are usually bash scripts. There is also significant external repositories, arm ports even a risc-V with slarm64. Alien Bob (A slackware dev) supplies useful stuff: 64+32 bit versions of wine, glibc & gcc(the latter 2 needed for multilib); his 32bit compatibility libs(or multilib); an ungoogled version of Chromium; piles of stuff you wanted but couldn't get - like restricted codecs in vlc for the Excited States and unrestricted for the rest of the world. Multilib is needed for running 32bit windows applications (which is most of them) on a 64bit OS.

My 88x2bu is not going. The module appears sensitive to multithreaded builds.  I have two git clones here(well organised, amn't I?) and two modules, and diff shows them as differing :-o. The infuriating thing for me was that I had the dongle working in my shiny new pc, but updated the kernel trying to get kernel support for the RTL8811CU, and lost my module. I could never get it going again. This thread refers [https://www.linuxquestions.org/questions/linux-hardware-18/hardware-problem-with-wifi-dongle-4175714726/](url) and saves a lot of typing. I installed my OS on this new pc by installing the basics, then restoring a backup from my recently retired laptop. I still have the laptop. The same dongle works in that! It has the same crash in dmesg when I insert the dongle, 'ifconfig wlan2 up' only takes 2-3 seconds, I get good upload & download, but that has it's own highly sensitive built-in Atheros wifi hardware as well, which I have to disable to test the dongle. 

I can't get debug type feedback from ifconfig or iw while they are sitting quietly for 10 seconds - I'd love a suggestion on that score. I know strace & gdb  are included with slackware, but I've never used them. Ditto the insertion crash, which happens every time.

So as you see, I've kinda fingered my shiny new pc as the problem, although it worked there before. And the dongle is obviously a POS.  (usb id: 0bda:b812). Any ideas welcome

morrownr, 2022-07-19:
Hi @BusinessKid 

Thanks for the update on Slackware. I will add it to my to-do list. I stay busy so it will have to happen when I have time.

You did throw out enough info for me to make a suggestion. Try an adapter that uses in-kernel drivers. The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

Menu item 1 provides information and menu item 2 gives information and links to specific adapters that use in-kernel drivers. The drivers should be in the kernel Slackware uses as they are part of the core kernel. You might have to add the firmware...does Slackware install the firmware for in-kernel adapters? If not, I can show you how quickly. I don't know if Slackware includes usb-modeswitch so staying with a single-state adapter might be a good idea. usb_modeswitch can be installed if need be. I can point you toward a specific adapter if you tell me what your use case is.

Regards

BusinessKid, 2022-07-20:
OK. This issue is sorted. It's the dongle, I think, and here's why.

I uncovered on my box a dmesg.log  extract where I inserted it initially to get the usb.id. No kernel crash.  Now every time I insert it, there is a kernel process crash. So I have sidelined the dongle. I can't return it while it's working in so many things, but I can't use it. I ordered an Atheros PCIE card (which WILL get returned pronto if it doesn't work). So I'm closing this issue.

FWIW, Slackware forked their kernel firmware from the main kernel tree. About 2015(?) someone did a cleanout of the kernel firmware to remove "All that old sh** that nobody uses." But people WERE using it, (mainly old wifi stuff) so skackware forked & added it in to their kernel-firmware-<date> package. I'm pretty sure they have all the firmware released by Realtek.

closed 2022-07-20 by BusinessKid