morrownr/88x2bu-20210702 #65: loosing AP access on distance

hakaishi, 2022-07-22:
Hi everyone!

I've got a very strange case here.   
My setup first. 
I'm using a raspberry as a wifi to wifi router. The raspberry pi 4B is connected to a wifi modem/router that doesn't support DNS settings nor DynDNS etc. As I can't access it directly, I'm using my raspberry pi and connect it to it. A USB WiFi in ap mode is connected to the raspberry pi and other devices connect to this AP. On the raspberry pi I run a pi-hole with a custom DNS setting. A iptables rule forwards the requests from the wifi ap ip address to the wifi module that is connected to the modem/router.

Now the problem:   
If I'm near the hotspot, I can connect and access to the internet without any trouble. But if  I'm about 10m away (WiFi strength is still good with about 80%), then I can't connect to the hotspot. It always stops at obtaining the IP address and fails there.

OS is Ubuntu Server 22.04.   
My USB WiFi is EDUP EP-AC1686. It uses the RTL8812BU driver.
Driver settings:   
```
options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_switch_usb_mode=2
```

Hostapd settings:
```
interface=wlxe84e069cc6e0
driver=nl80211
ssid=myaccesspoint
hw_mode=g
channel=7
wmm_enabled=0
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=mypassword
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
```

Any ideas what could be the cause?

morrownr, 2022-07-23:
> Any ideas what could be the cause?

Not really. I'm really good with RasPi4Bs, drivers settings and hostapd settings but there are two things that take me out of the know here:

- Ubuntu server. I've never used it. I've used Ubuntu desktop and various flavors of Ubuntu but not the server. I run and know Raspberry Pi OS very well.
- I don't have any visibility on what your basic setup is trying to accomplish and what guide you used to set it up.

If you go to the Main Menu of this site and select menu item 9, you can see I know my way around setting up a wireless access point with a RasPi4b, but my guide is not what you are doing. If you were up for burning a sd with RasPiOS 64 bit and giving it a try, I might be able to help.

Regards

hakaishi, 2022-07-23:
I'm actually wanting to use a nextcloud instance from the internet and with the same address from the local network. I think that this is only possible by using a local DNS. (I can't ask everyone to use some VPN or hosts file on their phone). I can't access the modem/router physically, so only a wifi-to-wifi bridge is possible.

I use pi-hole and nextcloud docker containers. The AP is setup by hostap, the DHCP is isc-dhcp-server and the DNS is setup by the pi-hole.

If there is a better way, I'd welcome any ideas.

Back to the actual problem. I got a hint that the wifi signal strength is only one way. In other words, my phone can't reach the raspberry pi, but the other direction works.

I tried changing the (physical) location and right now no more problems. 😊

> If you go to the Main Menu of this site and select menu item 9

Sorry, could you provide more details or a link? - I can't find it 😅

Best regards, Hakaishi

PS: I'll close this ticket later, or you could do it if you want.

morrownr, 2022-07-24:
> Sorry, could you provide more details or a link? - I can't find it sweat_smile

https://github.com/morrownr/USB-WiFi

closed 2022-07-24 by hakaishi