morrownr/88x2bu-20210702 #66: Make errors for RTL88x2BU driver

marcoevang, 2022-07-26:
My system is a raspberry pi 3b+ equipped with piCorePlayer v8.2 64 bit and a Tp-Link Archer T3U Plus wifi dongle.
The wifi dongle is not recognized out of the box, thus I ran the procedure for generating a dedicated driver for RTL8812BU chipset.
The selected kernel was 5.15.35-pcpCore-v8. Therefore I launched the following command:
   make KVER=5.15.35-pcpCore-v8 -j 1

The processs ended with errors, as follows:
  CC [M]  /mnt/mmcblk0p2/tce/realtek-driversrc/platform/platform_ops.o
  LD [M]  /mnt/mmcblk0p2/tce/realtek-driversrc/88x2bu.o
  MODPOST /mnt/mmcblk0p2/tce/realtek-driversrc/Module.symvers
ERROR: modpost: "__const_udelay" [/mnt/mmcblk0p2/tce/realtek-driversrc/88x2bu.ko] undefined!
ERROR: modpost: "__arch_copy_from_user" [/mnt/mmcblk0p2/tce/realtek-driversrc/88x2bu.ko] undefined!
ERROR: modpost: "gic_nonsecure_priorities" [/mnt/mmcblk0p2/tce/realtek-driversrc/88x2bu.ko] undefined!
ERROR: modpost: "cpu_hwcap_keys" [/mnt/mmcblk0p2/tce/realtek-driversrc/88x2bu.ko] undefined!
ERROR: modpost: "csum_ipv6_magic" [/mnt/mmcblk0p2/tce/realtek-driversrc/88x2bu.ko] undefined!
ERROR: modpost: "__warn_printk" [/mnt/mmcblk0p2/tce/realtek-driversrc/88x2bu.ko] undefined!
ERROR: modpost: "queued_spin_lock_slowpath" [/mnt/mmcblk0p2/tce/realtek-driversrc/88x2bu.ko] undefined!
ERROR: modpost: "arm64_const_caps_ready" [/mnt/mmcblk0p2/tce/realtek-driversrc/88x2bu.ko] undefined!
ERROR: modpost: "vabits_actual" [/mnt/mmcblk0p2/tce/realtek-driversrc/88x2bu.ko] undefined!
ERROR: modpost: "__udelay" [/mnt/mmcblk0p2/tce/realtek-driversrc/88x2bu.ko] undefined!
WARNING: modpost: suppressed 1 unresolved symbol warnings because there were too many)
make[2]: *** [scripts/Makefile.modpost:134: /mnt/mmcblk0p2/tce/realtek-driversrc/Module.symvers] Error 1
make[2]: *** Deleting file '/mnt/mmcblk0p2/tce/realtek-driversrc/Module.symvers'
make[1]: *** [Makefile:1783: modules] Error 2
make[1]: Leaving directory '/mnt/mmcblk0p2/tce/kernelsrc'
make: *** [Makefile:2499: modules] Error 2

Despite the errors, I ran the script to recompile the kernel (adding the driver for RTL8812BU). Predictably that script ended with errors as follows:

morrownr, 2022-07-26:
Hi @marcoevang 

From the pCP website:

```
Significant changes 
Kernel 5.15.35
Removed out of tree Realtek wifi drivers from Kernel. If you see a warning on the insitu update, you should not proceed until you build your driver—see pCP Kernels.
RPi Firmware Apr 26, 2022 (Does not affect RPi4 bootloader)
Squeezelite 1.9.9-1392
Image includes Mediatek wifi firmware.
Various bug fixes and web interface updates
```

Two things of interest in the above. That the devs removed the Realtek wifi drivers tells me they figured out just how badly the Realtek drivers don't play well given the development model of Linux. That they are including the Mediatek firmware is a big hint. Mediatek's in-kernel drivers are good and are the best solution for this situation.

I have too much on my to-do list to chase down this rabbit hole but I will post information on the various Mediatek based adapters and links to where you can get one:

https://github.com/morrownr/USB-WiFi

That is the Main Menu for this site. Menu item 1 gives some background and Menu item 2 gives information and links to many adapters that use in-kernel drivers. An adapter that is single-state and uses an mt7612u chipset would probably work well for you.

Regards

marcoevang, 2022-07-28:
I built a new driver for rtl8812bu on a clean install of the latest 32 bit piCorePlayer. The new extension was generated with no errors. Unfortunately, when I added it to the bootloader configuration file, my rPi 3+ wouldn't start up again after a reboot. I gave up on it. 
You were right  when you advised against buying a wifi dongle based on Realtek chip for use with the raspberry pi.
I ordered a new dongle based on Mediatek m7612u.
Thank you for your help.

morrownr, 2022-07-29:
You are welcome and please let me know how it turns out.

Nick

morrownr, 2022-10-09:
Did you get that new adapter yet?

marcoevang, 2022-10-10:
Yes, and it is natively supported by piCorePlayer and it works as expected.