morrownr/88x2bu-20210702 #67: Using wront 802.11 frequency and bit rate.

Deses, 2022-07-30:
Hello, I have a generic [Amazon 8812bu dongle](https://www.amazon.es/gp/product/B07PMS4BFC?psc=1) and I'm having some issues.

I installed this driver without any issue. I selected Concurrent_Mode during install because I want to have both a 5Ghz and a 2.4Ghz AP.

I'm using Mint 20.3 and I used the GUI to set my two wireless AP as following:

[image]
[image]

But as far as I see, the access points are working in 802.11n mode only, not ac. And the bit-rate is way too low, 144.4 Mbps is far from the advertised numbers (`5GHz (867Mbps) / 2.4GHz (300Mbps)`). I verified this by connecting to the networks with my phone and seeing that the AP is "WiFi 4", not 5.

This is the output from iwconfig:
```
$ iwconfig
wlx1ebfcee632b8  IEEE 802.11bgn  ESSID:"Home"  Nickname:"WIFI@RTL88X2BU"
          Mode:Master  Frequency:2.412 GHz  Access Point: 1E:BF:CE:E6:32:B8
          Bit Rate:144.4 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wlx1cbfcee632b8  IEEE 802.11bgn  ESSID:"Home_5G"  Nickname:"WIFI@RTL88X2BU"
          Mode:Master  Frequency:2.412 GHz  Access Point: 1C:BF:CE:E6:32:B8
          Bit Rate:144.4 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

See? both AP are using `IEEE 802.11bgn`, `Frequency:2.412 GHz` and `Bit Rate:144.4 Mb/s`. 🤔

My dongle is connected to a USB3.0 port in the back on the computer, directly to the motherboard.

I also tried to manually edit `/etc/modprobe.d/88x2bu.conf` to define my country code and the wireless mode, but this didn't fix the issue.

`options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_switch_usb_mode=1 rtw_country_code=ES rtw_wireless_mode=95`

My goal is to have a 2.4ghz and a 5ghz network to extend Wi-Fi coverage to my garage where I have some smartplugs that need 2.4ghz Wi-Fi but depending on the day they don't have a good enough connection. The 5ghz network would be used so phones and other mobile devices don't drop connection when they are brought to the garage.

The computer I'm using for this task is an old Intel i5 4670K based computer that I'm using as a "small" server and it's connected to the network using ethernet and it's getting a stable Gigabit connection from a nearby switch.

Deses, 2022-07-30:
By the way, why does my USB dongle has 2 MAC addresses? I understand one is for 5Ghz and the other for 2.4Ghz.
Which one should I use for each AP? I tried with both of them and I get the same results.

Deses, 2022-07-30:
Some more data:

```
$ iw list
Wiphy phy0
        max # scan SSIDs: 9
        max scan IEs length: 2304 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CMAC (00-0f-ac:6)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CCMP-256 (00-0f-ac:10)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
                * CMAC-256 (00-0f-ac:13)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * monitor
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x1963
                        RX LDPC
                        HT20/HT40
                        Static SM Power Save
                        RX HT20 SGI
                        RX HT40 SGI
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 16 usec (0x07)
                HT Max RX data rate: 300 Mbps
                HT TX/RX MCS rate indexes supported: 0-15
                Bitrates (non-HT):
                        * 1.0 Mbps
                        * 2.0 Mbps
                        * 5.5 Mbps
                        * 11.0 Mbps
                        * 6.0 Mbps
                        * 9.0 Mbps
                        * 12.0 Mbps
                        * 18.0 Mbps
                        * 24.0 Mbps
                        * 36.0 Mbps
                        * 48.0 Mbps
                        * 54.0 Mbps
                Frequencies:
                        * 2412 MHz [1] (16.0 dBm)
                        * 2417 MHz [2] (16.0 dBm)
                        * 2422 MHz [3] (16.0 dBm)
                        * 2427 MHz [4] (16.0 dBm)
                        * 2432 MHz [5] (16.0 dBm)
                        * 2437 MHz [6] (16.0 dBm)
                        * 2442 MHz [7] (16.0 dBm)
[... 105 lines cut ...]
                 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
        Supported RX frame types:
                 * IBSS: 0xd0
                 * managed: 0x40 0xb0 0xd0
                 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
        WoWLAN support:
                 * wake up on anything (device continues operating normally)
        software interface modes (can always be added):
                 * monitor
        valid interface combinations:
                 * #{ managed, P2P-client } <= 2, #{ AP, P2P-GO } <= 1,
                   total <= 2, #channels <= 1
        Device supports SAE with AUTHENTICATE command
        Device supports scan flush.
        Supported extended features:
                * [ RRM ]: RRM
```

morrownr, 2022-07-30:
Hi @Deses 

I need to slow you down just a minute. We all have to learn and ...

> I'm using Mint 20.3 and I used the GUI to set my two wireless AP as following:

A single usb wifi adapter cannot support dual band because it only has one radio. Dual band Wifi routers/access points have two internal radios.

Deses, 2022-07-30:
Ah, bummer. When I had a read at [Concurrent Mode](https://github.com/morrownr/88x2bu-20210702/blob/main/docs/Concurrent_Mode.md) my understanding was that I could use both radios at the same time. :P

closed 2022-07-30 by Deses