morrownr/88x2bu-20210702 #68: Compile pukes on 5.18.x kernel

BusinessKid, 2022-08-01:
To deal with a graphics issue, I upgraded to kernel (& headers) 5.18.14. The '-mharden-sls=all' option has gone in as a security measure by default for 5.18.x. The compile bales out thusly: `bash-5.1$ make -j6
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.18.14/build M=/home/dec/Downloads/88x2bu-20210702  modules
make[1]: Entering directory '/usr/src/linux-5.18.14'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 12.1.0
  You are using:           gcc (GCC) 11.2.0
  CC [M]  /home/dec/Downloads/88x2bu-20210702/core/rtw_cmd.o
  CC [M]  /home/dec/Downloads/88x2bu-20210702/core/rtw_security.o
  CC [M]  /home/dec/Downloads/88x2bu-20210702/core/rtw_debug.o
  CC [M]  /home/dec/Downloads/88x2bu-20210702/core/rtw_io.o
  CC [M]  /home/dec/Downloads/88x2bu-20210702/core/rtw_ioctl_query.o
  CC [M]  /home/dec/Downloads/88x2bu-20210702/core/rtw_ioctl_set.o
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
make[2]: *** [scripts/Makefile.build:289: /home/dec/Downloads/88x2bu-20210702/core/rtw_io.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [scripts/Makefile.build:289: /home/dec/Downloads/88x2bu-20210702/core/rtw_ioctl_query.o] Error 1
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
make[2]: *** [scripts/Makefile.build:289: /home/dec/Downloads/88x2bu-20210702/core/rtw_cmd.o] Error 1
make[2]: *** [scripts/Makefile.build:289: /home/dec/Downloads/88x2bu-20210702/core/rtw_ioctl_set.o] Error 1
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
make[2]: *** [scripts/Makefile.build:289: /home/dec/Downloads/88x2bu-20210702/core/rtw_security.o] Error 1
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
make[2]: *** [scripts/Makefile.build:289: /home/dec/Downloads/88x2bu-20210702/core/rtw_debug.o] Error 1
make[1]: *** [Makefile:1838: /home/dec/Downloads/88x2bu-20210702] Error 2
make[1]: Leaving directory '/usr/src/linux-5.18.14'
make: *** [Makefile:2499: modules] Error 2
bash-5.1$ `

I still have your module compiled for the 5.15.27 kernel, so this isn't a disasater, just a heads up.

morrownr, 2022-08-01:
Hey @BusinessKid 

What a mess you have there. I'm not seeing any problems with kernels 5.18 or 5.19 here and I test on multiple distros. Help me figure out some of the details of your system by posting the results of:

sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev

BusinessKid, 2022-08-03:
bash-5.1$ sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
Password: 
Linux Ebony.chateau.dec 5.18.14 #1 SMP PREEMPT_DYNAMIC Sun Jul 24 12:38:27 CDT 2022 x86_64 AMD Ryzen 5 5600 6-Core Processor AuthenticAMD GNU/Linux
SecureBoot disabled
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0c45:64a1 Microdia USB 2.0 Camera
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1462:7c95 Micro Star International MYSTIC LIGHT 
Bus 001 Device 005: ID 25a7:fa67 Areson Technology Corp 2.4G Receiver
Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 04e8:6031 Samsung Electronics Co., Ltd USB device
Bus 001 Device 006: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
phy#0
	Interface wlan4
		ifindex 3
		wdev 0x1
		addr b4:ee:b4:35:01:79
		ssid Moriarty
		type managed
		channel 9 (2452 MHz), width: 20 MHz, center1: 2452 MHz
		txpower 16.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0

Hi Nick,

Information as requested. wlan4 is a pcie card with Qualcomm Atheros AR9462 and I'm online ATM. As you're after the compile problem, I'm in 5.18.14.The getting online issue is sorted somewhat. My init scripts are not getting me online, but networkmanager is. I've reinstalled wpa_supplicant & dhcpcd, but no change. If I change to kernel 5.15.27, I have a module and my RTL8812BU appears as wlan2. Interestingly, output stays at -100dBm, i.e. OFF. I'm putting this business down to too many upgrades with no clean reinstall, so I'm going to do that at the next big kerfuffle with that box.

I tried getting online just to prove the concept, but getting the pcie card out is a major deal as I'm disabled.

BusinessKid, 2022-08-03:
You missed it. The piece of information you wanted was the o/p of "gcc --version."

My gcc was 11.2, which isn't enough for the 5.18.x kernel headers. So I went to gcc-12.1. But that complains about libc.so.6 from glibc-2.34, and wants glibc-2.35. Compiling your module has gone on the back burner as I feel a reinstall coming on. Reinstalls are quick with slackware, but configuration can be a real pain, as well as making the box your own. You can really do that in Slackware, rather than the other "one size fits all" distros. But it's often slow.

closed 2022-08-03 by BusinessKid