morrownr/88x2bu-20210702 #69: wlan0: CTRL-EVENT-SCAN-FAILED ret=-1

MephresGH, 2022-08-01:
Greetings. I have been facing odd issues as of late with my TP-LINK Archer T3U; WiFi would randomly cut off, stutter throughout the day, and behave weird as a whole. While looking in confusion into the systemd logs, i found that wpa_supplicant spams the error message mentioned in the title. NetworkManager says the following: 

`platform-linux: do-change-link[3]: failure changing link: failure 1 (Operation not permitted)`
`device (wlan0): set-hw-addr: failed to set MAC address to ((blank)) (scanning) (NME_UNSPEC)`
`device (wlan0): supplicant ninterface state: disconnected - inactive`

I installed the driver without modifying any settings later on or while having installed said driver. Is there a fix for this issue? Thank you.

morrownr, 2022-08-01:
Hi @MephresGH 

I need you to run and post the results of the following:

sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev

MephresGH, 2022-08-01:
> Hi @MephresGH
> 
> I need you to run and post the results of the following:
> 
> sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev

Linux 5.18.15-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 29 Jul 2022 22:52:39 +0000 x86_64 GNU/Linux
zsh: command not found: mokutil
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 046d:c33c Logitech, Inc. G513 RGB MECHANICAL GAMING KEYBOARD
Bus 003 Device 002: ID 2516:0141 Cooler Master Co., Ltd. MM720 Gaming Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0909:001c Audio-Technica Corp. AT2020USB+
Bus 001 Device 002: ID 2357:0138 TP-Link 802.11ac NIC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl88x2bu/5.13.1, 5.18.15-arch1-1, x86_64: installedError! Could not locate dkms.conf file.
File: /var/lib/dkms/rtl88x2bu/5.13.1.r163.9983008/source/dkms.conf does not exist.

zsh: command not found: iw

morrownr, 2022-08-01:
Hmmm...

Confirm for me what distro you are using? Arch, Manjaro or ?

Is it a rolling distro?

I see some things that are of concern but it is not clearn what yet.

MephresGH, 2022-08-01:
The distribution of choice is vanilla Arch Linux. I have manually installed packages that are required for the compilation and installation of kernel drivers, too. The system is up-to-date and the 88x2bu.ko driver was installed via DKMS installation script. The configuration file was not modified.

morrownr, 2022-08-01:
Okay... let's try some things.

> zsh: command not found: iw

If Arch does not have `iw` installed, that is a mistake by the developers. It should be there. Please install it.

When did you download this repo? The reason I ask is because the download on your hd remains static. i can't fix it and almost every Linux kernel release requires some updating so anytime you upgrade kernel, you first need to uninstall and delete the previous version and come get a new download.

MephresGH, 2022-08-01:
iw does not work because I deactivated it manually; iwd and wpa_supplicant are known to conflict with each other at times. It is still installed on my system.
As for when I downloaded the repo: I have the newest version of it right now installed and usually purge all traces of the previous installation on my system before installing a new driver.
On another note: I am currently able to use my Wi-Fi on 2.4GHz,

MephresGH, 2022-08-01:
...Actually, nevermind. I understood iw as part of iwd, my mistake. This is what iw spit out:

```
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr ***************
		ssid ************
		type managed
		channel 6 (2437 MHz), width: 40 MHz, center1: 2427 MHz
		txpower 16.00 dBm
```

morrownr, 2022-08-01:
Okay. Now we see a wifi interface so that tells us the driver is installed and at least working somewhat. Still not sure why `iw` in my script did not work. Did you install `iw`? If you did, that is a utility that should be in all Linux distros unless wifi is not supported.

> zsh: command not found: mokutil

I also see that mokutil must not be installed. I can't see if secure mode is enabled without it. It also means that if you are in secure mode, dkms can't key the driver resulting in a driver that will not load.

Are you sure you have your country code set properly? When I see 2.4 GHz working and 5 GHz not working, that is generally the problem.

> rtl88x2bu/5.13.1, 5.18.15-arch1-1, x86_64: installedError! Could not locate dkms.conf file.
File: /var/lib/dkms/rtl88x2bu/5.13.1.r163.9983008/source/dkms.conf does not exist.

Something is messed up. dkms is bitching.  Would you be willing to use the remove script and then try a manual installation so we can have better vis on what is going on?

MephresGH, 2022-08-01:
I installed mokutil.

```SecureBoot disabled; Platform is in Setup Mode```

I also do not know why dkms is bitching, as I installed it the way it should be. And I don't know about setting a country code, since I am from Germany and saw no anomalies in this regard 1-2 months ago.

morrownr, 2022-08-01:
> And I don't know about setting a country code,

Run and post the results of the following:

iw reg get

>  I am from Germany

Nice place. I enjoyed my time there.

> saw no anomalies in this regard 1-2 months ago.

Have you updated your system since then? Are you on the same kernel as then?

These bloody Realtek out-of-kernel drivers can be difficult to keep going. If you get pissed, relax and go to the follow link and read the first two menu items:

https://github.com/morrownr/USB-WiFi

MephresGH, 2022-08-01:
> run and post the results of the following: iw reg get

```
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (6, 20), (N/A)
	(2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

> Have you updated your system since then? Are you on the same kernel as then?

Yes, I update my system regularly. The kernel was 5.10 or something when I first went back to Linux. 5.18.15 is my current Linux kernel, as I posted above.
I also am fine, thanks for the comment though, haha.

morrownr, 2022-08-01:
> country 00: DFS-UNSET

That explains why you have no 5 GHz. See mine below:

```
 iw reg get
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)
```

What you probably need to do, since most of my experience is in the Debian world,  is go to the appropriate Arch forum and show them the result of your `iw reg get` command ask them how to set your country in Arch because it is not set. Once it is set, 5 GHz should work.

We can explore why dkms is bitching anytime you want.

Even if you don't get pissed, reading my suggested reading above might be a good idea as it contains a lot of good info for Linux users that use usb wifi adapters.

Regards

MephresGH, 2022-08-02:
I will talk in a forum later, but I'd prefer to get my hands on the dkms issue first. Are there any suspicions why the installation might have been incomplete or faulty?

morrownr, 2022-08-02:
> Are there any suspicions why the installation might have been incomplete or faulty?

I'm not saying that the installation is incomplete or faulty. In fact, you were able to show me a wifi interface and use 2.4 Ghz so that tells me the driver is working. That you can't use 5 GHz is not the driver's fault as it is doing what it should be doing if the OS doesn't provide country information.

Why dkms is bitching is unknown but I can guess: Your installation was originally using kernel 5.10 and now has 5.18 installed. I classify that as a hybrid installation in that it may haves parts from widely different time periods and changes that are not fully compatible.

The best idea I have right now if for you to use the dkms removal script:

$ sudo ./remove-driver.sh

Reboot and then we can see what a manual install does:

$ make clean
$ make
$ sudo make install
$ sudo reboot

MephresGH, 2022-08-02:
Alright, I have followed up with what you recommended, and it seems to work a bit better now. 5GHz seems to work fine, too. However, manual connecting and disconnecting still causes for stability issues and for "device could not be configured" and "secrets were required but not provided" errors, which is odd. I haven't checked for a regional fix in that regard, as that seems to not be required at all from what I have tested so far. It merely seems like either the driver is unstable, or my device is, even though my device never overheats and behaves quite normally, even being connected on a USB3.0 port while having USB2.0-USB3.0 options in the driver enabled.

morrownr, 2022-10-09:
Status?

closed 2022-10-19 by morrownr