morrownr/88x2bu-20210702 #74: ? blacklist rtw88_8822bu ?

ohquait, 2022-08-23:
i search on all what can find, but
**rtw88 brings not a module rtw_8822bu,**
lwfingers and also in kernel-tree nothing
rtw88 build only modules for pci/pcie **not for usb-devices**

so ithink:
 if a rtw_8822be conflicts with 88x2bu (or with a renamed rtl88x2bu),
then the correct entry would here:
 **blacklist rtw88_8822be**   **not  bu**

is it right ?

morrownr, 2022-08-24:
A few months ago, a gentleman submitted a series of patches to the linux-wireless mailing list in an effort to upstream usb support for the rtw88 driver. The code from this effort is not in the mainline kernel and it is not clear to me what the status is. However, some distros took the parch series and compiled it into the kernels for their distro thereby creating a conflict with the driver in this kernel.

What I merged is doing what I want it to do. The installation scripts add the blacklist and the removal scripts delete the file during removal so if folks follow the instructions here, all will be fine.

Thanks for checking my work. If you see any enhancements, please let me know or submit a PR.

Nick

ohquait, 2022-08-27:
Aha, now i understood.
the module is not placed on pclinuxos in kernel...
(not now, and i not knowing if it will be) 
Cool,
Can you give me a hint if you intend to continue your work?
**I really hope so, because I can't find anywhere such good descriptions and help as here!
THANK YOU again for that!**

morrownr, 2022-09-11:
Hi again @ohquait 

Sorry about the delay. I have been busy and assumed the answer to your question could wait.

> Can you give me a hint if you intend to continue your work?

I intend to continue but I could use some help. This repo is just a small part of what I do here. The Main Menu is located:

https://github.com/morrownr/USB-WiFi

There is a lot of information available and I hope to keep adding to it.

> I really hope so, because I can't find anywhere such good descriptions and help as here!

I've been using USB WiFi adapters for a long time. It has been clear to me for a long time that one of the problems that Linux users run into with USB WiFi adapters is the lack of correct, up to date information. That is the main thrust of this site but I learned programming many years ago so I am able to work on code when required. One thing that is happening right now is the new release of adapters with the new mt7921au chipset and the new mt7921u in-kernel driver that supports it. This chipset and driver are not just WiFi 6 capable but are WiFi 6e capable. That means 6 GHz channel support. I've been testing one of the new adapters. The results are good but I don't have anything, like a WiFi 6e wireless router, to test with so my ability to test the 6 GHz channels has been limited so far. I think we will see many new adapters with this chipset over the coming months. This is the first USB WiFi adapter that has come on the market and the Linux kernel had an in-kernel driver is place and working prior to the hardware being available. This is cool.

> THANK YOU again for that!

You are welcome.

Nick

ohquait, 2022-09-11:
For the AX1800 USB3 / 1850... adapter,
can you test the LWFinger driver or do you already have actual results with this driver?
`git clone https://github.com/lwfinger/rtl8852au -b dwa-x1850`

In any case, it can be built and installed as a dkms-rpm, with the current kernel 5.19.8, but
I don't have the appropriate hardware to test it.
If possible, it could perhaps be expanded to AX1800,
by passing the ID's to the usb_inf.c?
I do not know.
I also don't remember why I built it as a package for PCLinuxOS, but I think some user needed it and I haven't found any negative reports about the driver since then, so it seems to work, or it's ignored, unfortunately I get the feedback in the Forum also only sporadically, as the users are happy right now...

morrownr, 2022-09-11:
I have a repo for the 8852au. It is private. It is based on the same source as the driver you mentioned. Two other individuals and I had the intent to make it public but alas, it was very buggy and the decision was made to wait until a new version is available so as to look at the issue again.

I can take a look again at the current code again but one major factor against 8852/8832 is that Realtek is still trying to support their usb wifi adapters with out-of-kernel drivers and this does not work well at all. With the release of the mt7921au chipset that is supported by an in-kernel driver, I can't see a reason for Linux users to buy Realtek based adapters.

ohquait, 2022-09-12:
Aha, now i understood .
can closed please