morrownr/88x2bu-20210702 #75: CentOS Stream 9. TP-link T3U. dkms build error = 10

kpal99, 2022-08-24:
output of `uname -a; mokutils --sb-state; lsusb; rfkill list all; dkms status; iw dev` is
[command-output.txt](https://github.com/morrownr/88x2bu-20210702/files/9413052/command-output.txt)

output of sudo ./install-driver.sh
[install-output.txt](https://github.com/morrownr/88x2bu-20210702/files/9413053/install-output.txt)  

man the required make.log file is attached as below
[make.log](https://github.com/morrownr/88x2bu-20210702/files/9413063/make.log)

morrownr, 2022-08-24:
Hi @kpal99 

The RHEL 8.4 support that is currently in this driver was submitted by a RHEL user. At one point, I explored what it would take to add RHEL and CentOS support to the other drivers here on this site. What I discovered is that Red Hat selects certain things to add or backport to their kernels that make maintaining this driver on their kernels to be a real challenge.

It would take far more work than I have time to allocate but I welcome PRs from those who do have the time.

My recommendation for those using CentOS or RHEL is to get an adapter that uses in-kernel drivers. You can find info about and links to several adapters using menu item 2 at the following link:

https://github.com/morrownr/USB-WiFi

The ALFA AWUS036ACM, TEROW ROW02CD and ALFA AWUS036ACHM are favorites of mine.

Regards,

Nick

kpal99, 2022-08-25:
Hello @morrownr

Thanks for letting me know. In my country, can't get the adapters mentioned in [USB-WiFi repo](https://github.com/morrownr/USB-WiFi), but I am able to get the generic AR9271 USB-WiFi. Will that be okay in CentOS Stream 9? 
Thanks in advance.

closed 2022-08-25 by kpal99

morrownr, 2022-08-25:
Good morning @kpal99 

Everything that I know tells me that an adapter with a AR9271 chipset should work on CentOS.

Regards,

Nick