morrownr/88x2bu-20210702 #77: Does not build for kernel 5.19.4 in Fedora 36

ClodoaldoPinto, 2022-08-29:
Does not build for kernel 5.19.4 in Fedora 36
````
Running scriptlet: kernel-core-5.18.16-200.fc36.x86_64                                                                 12/15
dkms: removing: rtl88x2bu 5.13.1 (5.18.16-200.fc36.x86_64) (x86_64)
Module rtl88x2bu-5.13.1 for kernel 5.18.16-200.fc36.x86_64 (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

88x2bu.ko.xz:
- Uninstallation
- Deleting from: /lib/modules/5.18.16-200.fc36.x86_64/extra/
- Original module
- No original module was found for this module on this kernel.
- Use the dkms install command to reinstall any previous module version.
depmod...

Erasing          : kernel-core-5.18.16-200.fc36.x86_64                                                                 12/15
warning: file /lib/modules/5.18.16-200.fc36.x86_64/updates: remove failed: No such file or directory

Erasing          : kernel-devel-5.18.16-200.fc36.x86_64                                                                13/15
Cleanup          : kernel-headers-5.18.13-200.fc36.x86_64                                                              14/15
Cleanup          : kernel-devel-matched-5.18.19-200.fc36.x86_64                                                        15/15
Running scriptlet: kernel-core-5.19.4-200.fc36.x86_64                                                                  15/15
dkms: running auto installation service for kernel 5.19.4-200.fc36.x86_64
Sign command: /lib/modules/5.19.4-200.fc36.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j8 KVER=5.19.4-200.fc36.x86_64 KSRC=/lib/modules/5.19.4-200.fc36.x86_64/build......(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.19.4-200.fc36.x86_64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
Done.
dkms: running auto installation service for kernel 5.19.4-200.fc36.x86_64
Sign command: /lib/modules/5.19.4-200.fc36.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j8 KVER=5.19.4-200.fc36.x86_64 KSRC=/lib/modules/5.19.4-200.fc36.x86_64/build.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.19.4-200.fc36.x86_64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
Done.
````
[make.log](https://github.com/morrownr/88x2bu-20210702/files/9446482/make.log)

morrownr, 2022-08-29:
Hi @ClodoaldoPinto 

When did you last download this driver?

ClodoaldoPinto, 2022-08-29:
> Hi @ClodoaldoPinto
> 
> When did you last download this driver?

March of this year

morrownr, 2022-08-29:
Then I need to do some explaining.

Linux kernel development is continuous with new releases every 2.5 months or so. That is just how the dev model developed over the years. To support this model, there are guidelines that should be followed. Realtek doesn't follow those guidelines and makes their usb wifi drivers out-of-kernel. This means someone like me has to continuously patch the drivers to keep up with kernel development. If you just upgraded from kernel 5.18 to 5.19 and had the driver here in this repo from 2 weeks ago, it would not have worked. You really should have removed the driver prior to upgrading kernel.

I know this sounds like a lot of unnecessary work...and it is... which is why I strongly recommend Linux users buy USB WiFi adapters that use in-kernel adapters. The Main Menu for this site can be found at the following link and it has more info and menu item 2 has links to many adapters that use in-kernel drivers:

https://github.com/morrownr/USB-WiFi

Now, the fix to your problem:

- go to the directory where the driver is located
- $ sudo ./remove-driver.sh
- $ git pull
- $ sudo ./install-driver.sh
- The script with ask you to configure the driver options. it is a good idea to do so.
- $ sudo reboot

You should be in business.

Good luck.

ClodoaldoPinto, 2022-08-29:
@morrownr Thank you for the detailed answer. The blame obviously in on Realtek and I never had any doubt about. It is working now.

closed 2022-08-29 by ClodoaldoPinto

morrownr, 2022-08-29:
Good. Enjoy and stop by anytime.

Nick