morrownr/88x2bu-20210702 #78: Devuan 5.19.0-1 headers error: too few arguments to function ‘cfg80211_ch_switch_notify’

zastroyshchik, 2022-09-03:
rtl88x2bu Works great on 5.18.X kernels, but fails to build with 5.19.0-1:

```
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:478:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  478 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl88x2bu/5.13.1/build/include/osdep_service_linux.h:98,
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/include/osdep_service.h:54,
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:17:
/usr/src/linux-headers-5.19.0-1-common/include/net/cfg80211.h:7967:6: note: declared here
 7967 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c: At top level:
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:10409:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, unsigned int)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *)’ [-Werror=incompatible-pointer-types]
10409 |         .stop_ap = cfg80211_rtw_stop_ap,
      |                    ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:10409:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:10453:24: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, unsigned int,  struct cfg80211_chan_def *)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct wireless_dev *, struct cfg80211_chan_def *)’ [-Werror=incompatible-pointer-types]
10453 |         .get_channel = cfg80211_rtw_get_channel,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:10453:24: note: (near initialization for ‘rtw_cfg80211_ops.get_channel’)
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
/var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.c:10689:17: error: ‘struct wireless_dev’ has no member named ‘current_bss’
10689 |         if (wdev->current_bss) {
      |                 ^~
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-5.19.0-1-common/scripts/Makefile.build:254: /var/lib/dkms/rtl88x2bu/5.13.1/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-5.19.0-1-common/Makefile:1871: /var/lib/dkms/rtl88x2bu/5.13.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-1-amd64'
make: *** [Makefile:2499: modules] Error 2

```

morrownr, 2022-09-03:
Hi @zastroyshchik 

See the following section in the Installation Instructions:

Removal of the Driver

If you are upgrading to a new level of Kernel such as 5.18 to 5.19, you really need to remove the old driver and install the newest available before installing the new kernel. Try the following:

$ sudo ./remove-driver.sh
$ git pull
$ sudo ./install-driver.sh

Regards

zastroyshchik, 2022-09-05:
Hi @morrownr

Sorry to waste your time, I should have paid more attention to the doc :s

I take this opportunity to ask you some questions:

- Could you make your work merged upstream into the kernel?

- Why not triggering during the building of the new module, for a new major kernel release, or minor revision:
    1. the syncing of the source,
    2. the removal of the old module,
    3. the building and installation of the new module?

Like this, upgrading the module will be more convenient, especially with machines that are upgraded with unattended-upgrades.

Anyway, thank you for all your work!

closed 2022-09-05 by zastroyshchik