morrownr/88x2bu-20210702 #79: 88x2bu: disagrees about version of symbol module_layout

asnjudy, 2022-09-16:
[   50.159009] usb 2-1: new high-speed USB device number 2 using ehci-platform
[   50.308294] usb 2-1: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[   50.308320] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   50.308340] usb 2-1: Product: USB3.0 802.11ac 1200M Adapter
[   50.308358] usb 2-1: Manufacturer: Realtek
[   50.308375] usb 2-1: SerialNumber: 123456
**[   50.372446] 88x2bu: disagrees about version of symbol module_layout**

The installation was successful on Rock 5B(rk3588, os is debian), but the wifi did not work.
Then, I tried it out on the Raspberry PI 4B,  the installation is successful, and the wifi is available. 

rock@rock-5b:~$ uname -a
Linux rock-5b 5.10.66-20-rockchip-g0cdf37b755fe #rockchip SMP Wed Aug 24 14:12:32 CST 2022 aarch64 GNU/Linux

rock@rock-5b:~$ lsusb
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
**Bus 002 Device 002: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]**
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 003: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

morrownr, 2022-09-16:
Hi @asnjudy 

I happen to have a Rock 4SE that I am doing an assessment on right now. 

Tell me exactly which Debian you are using. Is it the one up on the Rock site? I downloaded the Beta Debian Buster 10 with kernel 5.10 and that is what I am working with right now.

> My guess is that the problem is somewhere between the driver and the os.

It is not the driver. I have not been able to get any of the Realtek driver here to compile. I have it narrowed down to an issue with the headers but I keep getting interrupted. We can probably figure it out but first I need to know the exact distro you are using so that we can see about getting on the same sheet of music.

Nick

asnjudy, 2022-09-17:
> Hi @asnjudy
>
> I happen to have a Rock 4SE that I am doing an assessment on right now.
>
> Tell me exactly which Debian you are using. Is it the one up on the Rock site? I downloaded the Beta Debian [...]

Thank you for your prompt reply.
I reburned the operating system with the latest official image, the problem disappeared.
Since the Rock5B development board is still in development and many features are not complete. New images are released every few days recently.
Obviously, my previous attempt, which failed, was a problem with the operating system.

**the development board**: Rock5B 
**the official image page**:  https://wiki.radxa.com/Rock5/downloads
**the latest image used**: https://github.com/radxa/debos-radxa/releases/download/20220915-0946/rock-5b-debian-bullseye-xfce4-arm64-20220915-1225-gpt.img.xz
**the os version**: Debian GNU/Linux 11
**the kernel version**: Linux rock-5b 5.10.66-21-rockchip-gbbe5dbfb385d

[  951.520499] usb 2-1: new high-speed USB device number 4 using ehci-platform
[  951.669741] usb 2-1: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[  951.669836] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  951.669870] usb 2-1: Product: USB3.0 802.11ac 1200M Adapter
[  951.669901] usb 2-1: Manufacturer: Realtek
[  951.669929] usb 2-1: SerialNumber: 123456
[  952.365202] rtl88x2bu 2-1:1.0 wlx90de805238a5: renamed from wlan0
[  953.459758] IPv6: ADDRCONF(NETDEV_CHANGE): **wlx90de805238a5: link becomes ready**

morrownr, 2022-10-09:
@asnjudy 

What is the status? I was busy for several days.

My own thoughts on radxa OS support is not positive at this point. As I said, I have been testing a Rock 4SE. The stable version of the Debian distro they have up is based on kernel 4.4. That kernel is not even supported anymore so I have tested the Debian Buster beta kernel 5.10 that they have up. 

The team that works on their distros simply does not know what they are doing in my opinion. They don't have the wifi firmware installed so I had to install it to get my wifi adapters that use in-kernel drivers going. I then tried to test the Realrek drivers that I maintain here and that did not end well. It appears the header files are not properly prepared. radxa REALLY needs to take a look at how they are providing OS support or they are not going to sell many boards.

What they should do is talk to the Xubuntu team and see if they can come to an agreement because they certainly cannot manage their own distros.

Nick

closed 2022-10-19 by morrownr

ahnjs1987, 2023-07-13:
I have same issue on raspberry pi 4.
I tested on 2 versions of kernel : 5.15.76-v7l+ / 5.15.92-rt57

Those versions return the same message.
[   90.875976] usb 1-1.3: USB disconnect, device number 5
[   93.983912] usb 1-1.3: new high-speed USB device number 7 using xhci_hcd
[   94.122177] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[   94.122212] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   94.122233] usb 1-1.3: Product: USB3.0 802.11ac 1200M Adapter
[   94.122251] usb 1-1.3: Manufacturer: Realtek
[   94.122267] usb 1-1.3: SerialNumber: 123456
[   94.435394] 88x2bu: disagrees about version of symbol module_layout

How do I fix it?

morrownr, 2023-07-13:
Hi @ahnjs1987 

Is this a 32 bit RasPiOS?

If so, what I would do is save anything important and burn the sd with a clean iso of the 64 bit version of RasPiOS. The RasPi devs seem to be on a roll these days as far as breaking the 32 bit version. I don't have the time fix their bugs.

ahnjs1987, 2023-07-13:
@morrownr 
Yes It is 32bit rasPiOS.
I will try with it on 64bit rasPiOS.
Thanks.