morrownr/88x2bu-20210702 #80: [not a bug] Undefined behavior warnings by UBSAN

spcharc, 2022-09-19:
The driver works fine but there are some call stack dump in my `dmesg`. So I think I should share them:

I think I did not see this kind of logs on Ubuntu 20.04. Recently I upgraded to 22.04, so perhaps these logs started to show up because of the kernel version change?
```
[   11.787959] UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/phydm_dfs.c:1286:51
[   11.787962] index 6 is out of range for type 'u8 [6]'
[   11.787965] CPU: 3 PID: 881 Comm: RTW_CMD_THREAD Tainted: P           OE     5.15.0-47-generic #51-Ubuntu
[   11.787968] Hardware name: AAEON MIX-EHLD1/MIX-EHLD1, BIOS 0006 01/07/2022
[   11.787970] Call Trace:
[   11.787973]  <TASK>
[   11.787977]  show_stack+0x52/0x5c
[   11.787983]  dump_stack_lvl+0x4a/0x63
[   11.787989]  dump_stack+0x10/0x16
[   11.787992]  ubsan_epilogue+0x9/0x49
[   11.787994]  __ubsan_handle_out_of_bounds.cold+0x44/0x49
[   11.787997]  ? rtw_hal_write_bbreg+0x17/0x25 [88x2bu2]
[   11.788093]  ? odm_set_bb_reg+0x24/0x73 [88x2bu2]
[   11.788169]  phydm_dfs_histogram_radar_distinguish+0x4d4/0x1d80 [88x2bu2]
[   11.788241]  phydm_radar_detect+0x1ce/0x249 [88x2bu2]
[   11.788311]  rtw_odm_radar_detect+0x17/0x1d [88x2bu2]
[   11.788382]  rtw_dfs_rd_hdl+0x1d0/0x3e1 [88x2bu2]
[   11.788431]  rtw_drvextra_cmd_hdl+0x202/0x580 [88x2bu2]
[   11.788478]  ? _rtw_up_sema+0xe/0x14 [88x2bu2]
[   11.788553]  ? rtw_ac_parm_cmd_hdl+0x19b/0x19b [88x2bu2]
[   11.788599]  rtw_cmd_thread+0x41c/0x76b [88x2bu2]
[   11.788645]  ? rtw_stop_cmd_thread+0x49/0x49 [88x2bu2]
[   11.788690]  kthread+0x127/0x150
[   11.788693]  ? set_kthread_struct+0x50/0x50
[   11.788696]  ret_from_fork+0x1f/0x30
[   11.788700]  </TASK>
```
```
[   11.788712] UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/phydm_dfs.c:1287:46
[   11.788714] index 6 is out of range for type 'u8 [6]'
[   11.788716] CPU: 3 PID: 881 Comm: RTW_CMD_THREAD Tainted: P           OE     5.15.0-47-generic #51-Ubuntu
[   11.788718] Hardware name: AAEON MIX-EHLD1/MIX-EHLD1, BIOS 0006 01/07/2022
[   11.788719] Call Trace:
[   11.788719]  <TASK>
[   11.788720]  show_stack+0x52/0x5c
[   11.788723]  dump_stack_lvl+0x4a/0x63
[   11.788726]  dump_stack+0x10/0x16
[   11.788728]  ubsan_epilogue+0x9/0x49
[   11.788730]  __ubsan_handle_out_of_bounds.cold+0x44/0x49
[   11.788733]  ? rtw_hal_write_bbreg+0x17/0x25 [88x2bu2]
[   11.788820]  ? odm_set_bb_reg+0x24/0x73 [88x2bu2]
[   11.788901]  phydm_dfs_histogram_radar_distinguish+0x502/0x1d80 [88x2bu2]
[   11.788984]  phydm_radar_detect+0x1ce/0x249 [88x2bu2]
[   11.789057]  rtw_odm_radar_detect+0x17/0x1d [88x2bu2]
[   11.789132]  rtw_dfs_rd_hdl+0x1d0/0x3e1 [88x2bu2]
[   11.789183]  rtw_drvextra_cmd_hdl+0x202/0x580 [88x2bu2]
[   11.789231]  ? _rtw_up_sema+0xe/0x14 [88x2bu2]
[   11.789305]  ? rtw_ac_parm_cmd_hdl+0x19b/0x19b [88x2bu2]
[   11.789352]  rtw_cmd_thread+0x41c/0x76b [88x2bu2]
[   11.789402]  ? rtw_stop_cmd_thread+0x49/0x49 [88x2bu2]
[   11.789448]  kthread+0x127/0x150
[   11.789451]  ? set_kthread_struct+0x50/0x50
[   11.789453]  ret_from_fork+0x1f/0x30
[   11.789457]  </TASK>
```
Here are some warnings similar to the previous one (call stack removed):
```
[   11.789460] UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/phydm_dfs.c:1286:24
[   11.789462] index 6 is out of range for type 'u8 [6]'
```
```
[   11.790185] UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/phydm_dfs.c:1288:53
[   11.790187] index 6 is out of range for type 'u8 [6]'
```
```
[   11.790919] UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/phydm_dfs.c:1289:47
[   11.790922] index 6 is out of range for type 'u8 [6]'
```
```
[   11.791672] UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl88x2bu/5.13.1/build/hal/phydm/phydm_dfs.c:1288:25
[   11.791674] index 6 is out of range for type 'u8 [6]
```

morrownr, 2022-09-19:
Thanks for the report. I'll add it to my list.

spcharc, 2022-09-19:
> Thanks for the report. I'll add it to my list.

tbh I am surprised that this kernel enabled UBSAN. I thought this should be some debug feature for kernel developers and should never be turned on in these linux distributions.
Also I am surprised that undefined behavior can be detected at runtime.

spcharc, 2024-09-29:
track with #233

closed 2024-09-29 by spcharc