morrownr/88x2bu-20210702 #81: TP-Link AC1300 slow performance on Raspberry Pi

lorivaro, 2022-09-21:
Hi there, I recently bought a TP-Link Archer T3U Plus USB WiFi adapter from Amazon for my Raspberry Pi 3B (non +, so no built-in dual band) so that it can serve as an extra 5GHz access point at home. I managed to get this working quite well by bridging eth0 to wlan1.

Problem is, both the up and down speeds never pass 30mbps (I even tested using iperf3)
I was able to reproduce this issue with a Raspberry Pi 4B as well. (Bridging eth0 to the built-in wlan0 WiFi adapter did produce wonderful results however, but I intend to use this Pi 4 for other projects)

If there is any more information I can provide, please let me know. Thanks!
`5.15.61-v7+ armv7l
EFI variables are not supported on this system
Bus 001 Device 004: ID 2357:0138 TP-Link 802.11ac NIC
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
        Soft blocked: yes
        Hard blocked: no
1: phy1: Wireless LAN
        Soft blocked: no
        Hard blocked: no
2: hci0: Bluetooth
        Soft blocked: yes
        Hard blocked: no
rtl88x2bu, 5.13.1, 5.15.61-v7+, armv7l: installed
phy#1
        Interface wlan1
                ifindex 5
                wdev 0x100000001
                addr 34:60:f9:54:63:69
                ssid ZiggoD4F8C6E
                type AP
                channel 36 (5180 MHz), width: 20 MHz (no HT), center1: 5180 MHz
                txpower 13.00 dBm
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr ba:27:eb:52:03:f4
                type P2P-device
        Interface wlan0
                ifindex 4
                wdev 0x1
                addr b8:27:eb:52:03:f4
                type managed
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
global
country NL: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5875 @ 80), (N/A, 13), (N/A)
        (5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)
`

morrownr, 2022-09-22:
Hi @lorivaro 

I'll be glad to help.

> If there is any more information I can provide, please let me know.

Yes, you did not mention what software you are using to run the AP.

> channel 36 (5180 MHz), width: 20 MHz (no HT), center1: 5180 MHz

Hmmm... 20 MHz (no HT). That is not going to get you a lot of speed.

Let me see. I have a guide for this type of setup. It has some Pi4B specific things but they can be ignored. See menu item 9 at the site Main Menu:

https://github.com/morrownr/USB-WiFi

Let me know how you want to preceed.

Nick

lorivaro, 2022-09-22:
Thanks for the quick response.
>Yes, you did not mention what software you are using to run the AP.

I'm running hostapd with the following configuration:
```
interface=wlan1
bridge=br0
driver=nl80211

hw_mode=a
channel=36
country_code=NL
ieee80211ac=1
ieee80211d=1
ieee80211h=1
wmm_enabled=1
vht_capab=[SHORT-GI-80]

auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

ssid=[redacted]
wpa_passphrase=[redacted]
```
Initially I followed these guides, maybe the issue can be found there:
https://forum.openmediavault.org/index.php?thread/13912-raspberry-pi-3-access-point-without-dhcp-server/
https://www.raspberrypi.com/documentation/computers/configuration.html#setup-the-network-bridge
But like I said, I was satisfied with the speeds wlan0 produced on the RPi 4B with the same configuration (~70mbps) . I'm only really aiming for about 100mbps, since that's all a RPi 3B can do and is plenty for most wireless use cases.

I have looked at the guides and I'm really impressed with the level of detail. Looking at the bridging guide, I can spot a couple of simlarities in my setup:
```
                          ((((( NAS    
                        ╱
INTERNET >>>>>>> modem/router >>>>>>> RasPi ))))) ((((( laptop/desktop
                (cable)         ╱                ╲
                         Powerline AV2            ((((( phone
```
Running a speedtest on the RPi 3B results in ~100 mbps ethernet speeds.
Thanks again,
Louk

morrownr, 2022-09-26:
Louk,

Are you ready for this issue to be marked solved?

Nick

closed 2022-10-09 by morrownr